leosac/access-control #62: Error while running CMake

holachek, 2015-05-12:
I'm following the build & install steps here (https://github.com/islog/leosac/wiki/Build-and-install) but get a few errors:

```

cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DZMQPP_BUILD_STATIC=0 -DZEROMQ_LIB_DIR=`pwd`/libzmq/.libs/ -DZEROMQ_INCLUDE_DIR=`pwd`/../libzmq/include -DZMQPP_LIBZMQ_CMAKE=1
CMake Warning at CMakeLists.txt:17 (message):
  No hardware platform is specified

  Example: -DLEOSAC_PLATFORM=RaspberryPi

-- Try C++11 flag = [-std=c++0x]
-- Performing Test CXX11_FLAG_DETECTED
-- Performing Test CXX11_FLAG_DETECTED - Success
CMake Error at /usr/share/cmake-2.8/Modules/FindBoost.cmake:1194 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.49.0

  Boost include path: /usr/include

  The following Boost libraries could not be found:

          boost_serialization

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.
Call Stack (most recent call first):
  CMakeLists.txt:30 (find_package)

-- Detected ZMQ Version - 4.2.0
-- Checking whether SOCK_CLOEXEC is supported
-- Checking whether SO_KEEPALIVE is supported
-- Checking whether TCP_KEEPCNT is supported
-- Checking whether TCP_KEEPIDLE is supported
-- Checking whether TCP_KEEPINTVL is supported
-- Checking whether TCP_KEEPALIVE is supported
-- Could NOT find AsciiDoc (missing:  ASCIIDOC_EXECUTABLE)
-- Configuring incomplete, errors occurred!

```

I'm not sure how to fix these...

xaqq, 2015-05-12:
Hello,

Are you sure that libboost is installed? 
Assuming you are on Debian / Ubuntu, can you try to `apt-get install libboost-serialization-dev`.

Note: While the Wiki is ~fine for installing, it is largely outdated wrt to configuration information and example. I recommend you generate the documentation with `doxygen` instead. Let me know if you're having trouble.

holachek, 2015-05-12:
Ah, that was the problem! Thanks xaqq!

xaqq, 2015-05-15:
You're welcome, glad that it's fixed.

closed 2015-05-15 by xaqq