leosac/access-control #66: Compilation freezes at 98% when compiling LEDBuzzerModule.cpp.o

Gbengat, 2015-07-30:
I'm trying to install leosac and compiling from the source. However when the compilation reaches that point, it doesn't finish or give me an error even after leaving it all night. Please assist me with this issue.

xaqq, 2015-07-30:
Hello,
I assume you are compiling on the Raspberry Pi directly. Correct me if I'm wrong.
I believe I've had that happen to me sometime (not necessarily when compiling this particular file) when building on the raspberry. However most of the time I'm cross compiling so I've not investigated this issue.

What happens if you simply type `make` again? The compilation should restart where it left of, and the file should get compiled. Can you try that and let me know?

Gbengat, 2015-07-30:
Yes, However I have started recompiling afresh from scratch. When it gets back to that point in about 1hr,  I will try that and let you know.

Gbengat, 2015-07-30:
One thing I forgot to mention, there are a bunch of these warnings  "/home/pi/leosac/libzmq/src/plain_client.cpp:135:5: warning: unused parameter ‘cmd_data’ [-Wunused-parameter]
 int zmq::plain_client_t::process_welcome ("  Not sure if that points to something I'm doing wrong. I followed the instructions here: https://github.com/islog/leosac/wiki/Build-and-install

xaqq, 2015-07-30:
> Not sure if that points to something I'm doing wrong

No don't worry, you can safely ignore these warnings.

Gbengat, 2015-07-30:
Oh ok. So this is where it gets stuck:
[image]

Typing "make" again simply scrolls through all the compiled files and gets stuck at the same point. I have no way of telling if it is doing something the background or it has stopped.

xaqq, 2015-07-30:
Okay, follow-up question then:
    + How long have you been waiting?
    + Can you run `top` in an other shell and check if the compiler is actually using CPU and working?

This file is actually rather heavy on the compiler because it uses boost state machine: that's heavy template meta-programming and is hard for the compiler to process.

Gbengat, 2015-07-30:
+ How long have you been waiting?
Yesterday I left it compiling all night, and this morning it was still stuck at that point.

+ Can you run top in an other shell and check if the compiler is actually using CPU and working?
top tells me that process the pi is working. Process cclplus which I'm guessing is the compiler is using close to 90% of CPU and 80% of the RAM.

Gbengat, 2015-07-30:
Oh sorry I forgot, today I've been waiting for over 1 hour at that 98%

xaqq, 2015-07-30:
Okay -- It could be that the compiler is swapping to disk, making the compilation insanely slow.
I'll start a build on a Raspberry Pi and report back to you (probably tomorrow).

Can you tell me if your raspberry pi has 256MB of RAM or 512MB ?

In the meantime (assuming you are running Raspbian) you can download the binary `*.deb` to install Leosac. Package is available [here](https://github.com/islog/leosac/releases/download/0.6.0/leosac_0.6.0-1_armhf.deb).

Gbengat, 2015-07-30:
Thank you xaqq. It is a 512MB RAM. I'd appreciate that and if there are any developments before then I'll be sure to let you know.

Gbengat, 2015-07-30:
Update. leosac has finally finished compiling. I guess those last files were very resource intensive on the Pi. Now that I've installed it, I am unable to start it because the system can't find the files hw.xml and core.xml as this page https://github.com/islog/leosac/wiki/Basic-Setup of the documentation states. Or do I need to create them?

xaqq, 2015-07-30:
Glad to know it compiled !

The wiki (except for the building-leosac part) is pretty much outdated. You can find the documentation at http://islog.github.io/leosac/doc_output/develop/html/ .

I just realized that if you copy-pasted from the install guide, you may have build the `master` branch (version 0.4) while the proper documentation targets the `develop` branch (version 0.6). Some information will be correct but some may be outdated.
I recommend you build the `develop` branch when you have some time, since the compilation is ... rather long :)

The basic feature *should* work fine with v0.4 though.

Leosac is based around one configuration file (no more `hw.xml` and `core.xml`) (However some module can use additional configuration file)

This [tutorial](http://islog.github.io/leosac/doc_output/develop/html/d5/d38/install_guide.html) explains how to setup a basic Leosac unit. However it assumes that you have a Piface Digital device. If you don't own one, you'll have to replace use the `SYSFS_GPIO` module instead of the `PIFACEDIGITAL_GPIO` module.

Gbengat, 2015-07-30:
Oh,  that explains alot.  Thank you very much :-) I will go through the
links and reinstall the packages.  Someone needs to update those pages.

Gbengat, 2015-07-31:
I have installed leosac with the .deb package in the link however when I try to run it before any configuration, It shows me this when I check if its running:
[image]

xaqq, 2015-07-31:
You should try running it in foreground first (simply running `leosac -k config_file`) and see what happens.
You could also check the log. Supposedly the config file doesn't exist (`/etc/leosac.d/kernel.xml`) which is why it fails.

Gbengat, 2015-07-31:
Ok, so I will need to create my config file first? I assumed there is a base config that comes with the package. For some reason, the kernel.xml is found in /etc/leosac.d/factory/kernel.xml

xaqq, 2015-07-31:
The file `/etc/leosac.d/factory/kernel.xml` is a backup configuration file. It should be a copy of your initial configuration file. But unless you plan to run Leosac in a master/slave configuration or plan to remotely change the configuration you shouldn't need a factory reset config file.

Yes you will need to create your configuration file. A base configuration file wouldn't really make sense, as the config file is very use-case / available-hardware specific.

What kind of hardware do you have?

Gbengat, 2015-07-31:
Oh I see. I have a weigand generic keypad rfid reader and a raspberry Pi. I intend to use it to control a couple of solenoids and magnetic doorlocks. I'm testing them

xaqq, 2015-07-31:
Ok, I believe you can successfully setup this indeed.
You should have some success following the install guide but using the `SYSFS_GPIO` module instead of the `PIFACEDIGITAL_GPIO`.

The documentation for each module configuration is locate [here](http://islog.github.io/leosac/doc_output/develop/html/pages.html)

Gbengat, 2015-07-31:
Ok, in what directory do I keep my xml config files? That part is not very clear to me from the documentation. These are the directories leosac created when I installed it.

[image]

Where do I keep my working files and config.

xaqq, 2015-07-31:
You can keep them wherever you want. Simply points to your configuration file when you start Leosac.

Leosac also supports a `--working-directory` command line switch if you ever want to change the working directory (if you have relative path in the main config file).

Gbengat, 2015-07-31:
Ok. Thank you for your help. I should be alright now.

closed 2015-07-31 by Gbengat

xaqq, 2015-07-31:
You're welcome.
Btw, you may run into #64 since I haven't updated the Debian package yet. However, unless you run in `strict` mode if should work despite the error message in the log. 

If you find any problem in the documentation / have some problem feel free to open another issue.
Good luck.