leosac/access-control #67: Help Configuring leosac

Gbengat, 2015-08-04:
I have a little confusion as to what file to edit inorder to show leosac where my config files are. I assumed that once installed on the pi it could be launched from any directory containing the configuration files.  I have tried launching leosac with the command in the documentation 'leosac -g hw.xml -f core.xml' where I named my hw and core files 'auth.xml' and 'leosac.xml' respectively. However it tells me it can't find a match for the arguement.
[image]
I would really appreciate it if I could have a small demonstration or view some example working config files. I used the documentation to create my config files. Then I am using SYSFS_GPIO instead of the  PIFACEDIGITAL_GPIO. In short I need a quick walk through from after the installation of leosac to launching the program with a simple configuration files.

xaqq, 2015-08-04:
The `hw` file doesn't exist anymore. Hardware is fully configured through modules. The only configuration file you pass to Leosac is the core configuration file. It's given with `-k`.

The auth module will find it's configuration file on its own, because the path to this additional config file is specified in the core configuration file.

Could you use pastebin to upload your auth and core config file?

Gbengat, 2015-08-04:
Oh ok, I uploaded with gist.
Here is my core file <script src="https://gist.github.com/Gbengat/684a6b19b39bc29965df.js"></script>
and auth file <script src="https://gist.github.com/Gbengat/5dee3150bd3988762886.js"></script>

xaqq, 2015-08-04:
The core configuration is mostly correct, except the part where you configure the SYSFS gpio module.  More info on the module SYSFS_GPIO: http://islog.github.io/leosac/doc_output/develop/html/d2/d1b/mod_sysfsgpio_main.html

The configuration for the auth is also incorrect.        
```
 <map>
            <user>Samuel Idowu</user>
            <WiegandCard>Sammy_Idowu</WiegandCard>
        </map>
```

`WiegandCard` should contain a `card_id` and `bits` field. See http://islog.github.io/leosac/doc_output/develop/html/d0/d25/mod_auth_file_main.html#mod_auth_cred_wiegandcard

Gbengat, 2015-08-04:
Oh ok, I'll change that now. So when I have all those fixed, I just execute leosac from the directory with my configuration file ie ./leosac -d ["Door_Locks" which is the directory where I have all my files] -k leosac.xml and it should run?

xaqq, 2015-08-04:
Yes it should.

Gbengat, 2015-08-04:
Thank you for your help. "./leosac" doesn't work so I just used "leosac" and it I'm getting the error below.
[image]

xaqq, 2015-08-04:
Hum this is weird. Can you give me the following:
    + The `md5sum` for every `.so` in `/usr/lib/leosac/` and `/usr/lib/leosac/auth`.
    + The output of `ldd /usr/lib/leosac/libsysfsgpio.so`

Gbengat, 2015-08-04:
+ The md5sum for all .so in /usr/lib/leosac/ and /usr/lib/leosac/auth.
[image]
+ The ouput of ldd /usr/lib/leosac/libsysfsgpio.so
[image]

Gbengat, 2015-08-04:
Oops disregard the second one. The ouput of ldd /usr/lib/leosac/libsysfsgpio.so is:
[image]

xaqq, 2015-08-04:
I don't really know what is happening. After comparing your Leosac output to mine it looks like you are not using the version `0.6.0`. I am saying this because the usage don't match mine.

Can you post the ouput of:
    + `leosac --version`
    + `which leosac`

Gbengat, 2015-08-04:
No its 0.4 
Version:
[image]
Which leosac
[image]
I installed this one with the .deb package

xaqq, 2015-08-04:
Well, I guess this is the explanation then.
The leosac binary at `/usr/local/bin/leosac` is actually the result of your manual compilation and installation of Leosac from the master branch.

The debian package put the binary in `/usr/bin/` but since `/usr/local/bin` has the priority in the `PATH`, the wrong binary is invoked.
You can safely `rm /usr/local/bin/leosac` and hopefully that should fix the issue.

Gbengat, 2015-08-04:
Oh I see..let me do that now

Gbengat, 2015-08-04:
Now bash is complaining that the /usr/local/bin/leosac path is missing

Gbengat, 2015-08-04:
Or let me purge leosac and do a fresh install of the .deb package and I will get back to you if it still doesn't launch

Gbengat, 2015-08-05:
So I checked my package again carefully and it turns out I accidentally installed the 0.4 version. Following the link you sent me earlier I was able to download the 0.6 version. This time it launched started it. But I'm getting this error below.
[image]

xaqq, 2015-08-05:
Oh yes that's my bad. A new `<instance_name>blah</instance_name>` tag is required in the general section of the configuration (http://islog.github.io/leosac/doc_output/develop/html/d2/d94/general_config_main.html#general_instance_name).
This is rather new -- I need to update the documentation of the getting-started guide.

Gbengat, 2015-08-05:
Oh ok...I've just added that to the config file. However there is another error when it tries to run.
[image]

xaqq, 2015-08-05:
Are you missing the various configuration option like:
```
<aliases>
<default>gpio__NO__</default>
</aliases>
<export_path>/sys/class/gpio/export</export_path>
<unexport_path>/sys/class/gpio/unexport</unexport_path>
<edge_path>/sys/class/gpio/__REPLACE_ME__/edge</edge_path>
<value_path>/sys/class/gpio/__REPLACE_ME__/value</value_path>
<direction_path>/sys/class/gpio/__REPLACE_ME__/value</direction_path> 
```

See http://islog.github.io/leosac/doc_output/develop/html/d2/d1b/mod_sysfsgpio_main.html

Gbengat, 2015-08-05:
Turns out I was. I have entered those and changed "_REPLACE_ME_" to a gpio number. This is the output. I think I am missing things in my configuration file.

[image]

Gbengat, 2015-08-05:
Ok I have used exactly as the example. So the error above has gone. But it is complaining that there is a problem with my auth.xml file. Let me check it. This is the output:
[image]

Gbengat, 2015-08-05:
Ok I am unable to figure out what is wrong with the auth.xml file

xaqq, 2015-08-05:
Can you post the content of your `auth.xml` file?

Gbengat, 2015-08-05:
Sure...Find it here:
{% gist 5dee3150bd3988762886 gist.md %}

xaqq, 2015-08-05:
Mmm, your definition for `WiegandCard` are wrong. it's supposed to be something like:
```
<WiegandCard>
<card_id>aa:bb:cc:dd</card_id>
<bits>32</bits>
</WiegandCard>
```

Gbengat, 2015-08-05:
Ok..Ive changed that but I still have the same error. The changes are here: 
{% gist 5dee3150bd3988762886 gist.md %}

xaqq, 2015-08-05:
You still have `<bits>32<bits/>` instead of `<bits>32</bits>`.

Gbengat, 2015-08-05:
Oh oh oh...my bad! That was it! It's running now :-) Thank you very much

xaqq, 2015-08-05:
I hope it works for you :).

By the way, know that all your troubles were not for nothing. We'll be working at improving error reporting so that it's easier for users to know what's wrong.

Gbengat, 2015-08-05:
Thanks :-) If any other modules give errors I'll be sure to let you know.

Gbengat, 2015-08-24:
Hi, I am having a slight issue with wiegand_data_high and wiegand_data_low. Leosac says my pins do not exist. Or possibly I am not naming the pins correctly. This is the output.
[image]

My code for my core file is here: {% gist 684a6b19b39bc29965df gist.md %}  

I've tried naming my gpio pins as eg GPIO1 , just numbers but still it tells me the pins do not exist.

xaqq, 2015-08-24:
Hello,
This should be harmless as it's, IIRC, just a message into the log (unless you run in `strict` mode).

This has been reported and fixed already: see https://github.com/islog/leosac/issues/63
Installing the new Debian package will fix the issue.

Gbengat, 2015-08-24:
Oh ok, let me install the latest package.

Gbengat, 2015-08-24:
So it turns out there are some unmet dependencies which weren't there before with 0.6.0. #62 mentions the libboost-serialization1.55.0 but when I checked, I have the latest installed. However I'm not sure what this libboost-regex1.55.0 dependency is.

xaqq, 2015-08-24:
https://github.com/islog/leosac/blob/develop/pkg/debian/control is the file that declares dependency but it didn't change for a while.
Try updating your system because the update should happen without trouble. What boost packages do you have installed? `dpkg -l | grep boost`.

Gbengat, 2015-08-24:
Hmmm, that's odd. Updating is the first thing I did, but the dependency error still comes. 
Boost packages: http://pastebin.com/2xFs5dKX

xaqq, 2015-08-24:
Hum... okay. Can you run `apt-get install -f` after running `dpkg -i leosac_0.6.1-1.deb` ? This *should* correctly setup the dependency.

Gbengat, 2015-08-24:
Ok..I've tried a couple of things but I'm still getting the dependency error.

xaqq, 2015-08-24:
This is a bit weird. Can you confirm that you have the Jessie repositories enabled in your `/etc/apt/source.list` ?

Gbengat, 2015-08-24:
Yea, Jessie is enabled
[image]

xaqq, 2015-08-24:
Okay...
Well I admit I have no idea. I did the upgrade on my own Pi and it went without trouble. 

Maybe try fully removing Leosac and reinstalling (assuming you haven't already). But this is a random guess :(

Gbengat, 2015-08-24:
Hmmm, ok. Let me try that and get back to you later

Gbengat, 2015-08-28:
HI, Ive tried abit of this and that over the past couple of days but I still get these two dependency errors 
[image]
and can't install the package. 0.6.0 installs fine but it has that bug that was fixed in 0.6.1. The one you installed on your pi, is it the same package that is uploaded here: https://github.com/islog/leosac/releases ? I suspect there could be a problem with the package. I want to do a clean install of raspian on another raspberry pi just to be sure it is not my system. My last resort would probably be to try and build from source.

xaqq, 2015-08-28:
Yes it is this package I have installed.
> I want to do a clean install of raspian on another raspberry pi just to be sure it is not my system.

That's something worth trying.

However, can't you install `libboost-regex1.55-dev` using `apt` ? Maybe you need to reinstall boost using the Jessie repository (instead of Wheezy?). 
I've checked with `dpkg -I` and indeed the package `0.6.1` requires `boost-regex-1.55` while `0.6.0` requires `1.49`. I don't know why. I didn't spot the problem because I have multiple boost installled, so it worked fine I guess.

I'll need to investigate.

Gbengat, 2015-08-28:
Oh ok, thank you. I'm currently getting a 101 network unreachable error for the distros both wheezy and jessie. Could be some temporary down time, I will try again later on.

Gbengat, 2015-09-07:
Hi, quick update on the issue I had several days ago. I was finally able to update libboost and leosac 0.6.1 was able to install. Now that I've got it running. I can't seem to get my reader to read any cards or the pin code to work either. Here is my auth.xml file http://pastebin.com/mmCfXjSk I'm pretty sure I haven't configured it well. Could it be that my main config file isn't reading it or telling the right pins to fire? Here is my main config http://pastebin.com/06uLnuyr And incase of successful card reading, will leosac give some kind of command-line notification?

xaqq, 2015-09-07:
Running with your configuration fails with:

```
exception: Module::Unable to init module AUTH_FILE: Actor Initialization Exception
|--> exception: Actor Initialization Exception
|----> exception: Module::AuthFile cannot load configuration
|------> exception: Configuration error in file {auth.xml}: Invalid config file content
```

Indeed, you have a few configuration errors in both the main config file and the auth file. This should work better.

auth: http://paste2.org/eC7kV01p
main: http://paste2.org/cIfcaVjK

Gbengat, 2015-09-07:
Thank you very much! Let me try again with the new configuration.

closed 2015-09-22 by Gbengat