leosac/access-control #69: WS-Notifier Module not working for HTTPS URLs

Marcus-L, 2016-01-04:
Module: libwsnotifier.so, using <url> with https://

From the log, getting an error message with every notification attempt:
[2016-01-04 16:40:02.249] [console] [warning] curl_easy_perform() failed: Out of memory

Relevant code:
https://github.com/islog/leosac/blob/develop/src/modules/ws-notifier/WebServiceNotifier.cpp

Possible solution:
http://stackoverflow.com/questions/166617/curle-out-of-memory-error-when-posting-via-https

"add the flag CURL_GLOBAL_SSL to curl_global_init"

xaqq, 2016-01-07:
Hello,

Thank you for the report. Your possible solution is indeed correct.
However to work with self-signed cert (which I used to reproduce the issue) a bit more work is needed -- mostly providing additional configuration option to prevent certificate validation.

I have a patch mostly ready that I will probably push tomorrow.

closed 2016-01-08 by xaqq

Marcus-L, 2016-01-16:
> Closed #69 <https://github.com/islog/leosac/issues/69> via 975000a

Thanks so much for the quick update!

commit 975000a1 (2016-01-08): Improve WS-Notifier module with regard to SSL.

It's now possible to connect to an HTTPS webservice to publish
notification. This fixes #69.

Additional configuration option are now available to control
wether or not SSL shall be used, aswel as option to
validation certificate / hostname / use a custom CA.