leosac/access-control #71: multiple requests

choleman, 2016-06-14:
First off, pardon if this is not where to do.

1) Where can I find current up-to-date authfile settings, I've got things configured but still wont find the username

2) enhancements to ws-notifier requests
    a) send more than card_id, ie: what door, 
    b) better pin handling
    c) if username is know, pass along
    d) again having a bear of a time trying to log access, ive tried MULTIPLE different approaches

3)  having a bear of a time with proper wiegand codes ie: code on fob says 0001234561231 but in ws-notifier shows "4623456" as id, I've decoded this down to binary and padding 0's left/right and can replicate the correct # given by leosac to make a match, but am I doing something wrong or a bug

4) in auth file, again key fob id doesnt correctly correlate to hex value leosac expects

5)  improvements to remote control since a rest app/api is undesired

I'm not much a c programmer but willing and already started to work on a web front-end to handle users, logging, and have thought of the idea of being able to set up a script to have leosac pull from the web front-end in master/slave mode, to update auth/kernel files and have that front end dish out updated files and config.

It appears that this is a side project persay amongt friends maybe, but like the software and wanting to contribute and enhance if possible.

Thank you

Maxhy, 2016-06-14:
Let me just react on your latest point for now.

It is a side project among coworkers and friends yes, but it also have the support of ISLOG company which remains the project initiator. Problem is that it will probably never be profitable so it's more a support by ideology than for business :wink:.
[We](https://github.com/orgs/islog/people) have great plans about webui but no much time to really do it for now. @xaqq started an expirement few months ago, see https://github.com/islog/leosac/tree/webui-experiment
To be honest we are also more all C/C++ developers, with specialization on RFID / NFC and security (see other open source project https://github.com/islog/liblogicalaccess) and web UI is not our best. If we can work together on this it would be great, we (@xaqq, @Liryna and I) have some plans with webui that we can share for discussion of course.

For us a rest API where the app remains on the browser is the desired design at the end for previous experiences, why did you said it is *undesired*?

choleman, 2016-06-14:
I had looked around and didn't see the webui expirement. I'll look at that.  Unfornately it seems as the repo is kinda riddled with outdated information and some things are hard to find.  

In regards to RemoteControl and REST.   I thought I had saw something somewhere where a comment was made that "REST isnt needed as it'd be accomplished thru the RemoteControl module".   I may have misunderstood, I'll have to find the comment and re-read.

Myself, I'm an IT guy, but I've been into programming/scripting/html pretty much my whole life, I can read/understand C, but can't write it.  I'm not an RFID guy nor a micro/hardware guy either, but I get thru ok.

My last rfid door system, I created with a pi, custom soldered board, leds, buzzers and rfid reader.   Worked well but as we came into summer (Phoenix Arizona USA, very very hot) the reader wouldn't read when it got hot.    To by-pass this, I implemented a sms gateway persay that would traverse/bounce thru a router, server and into the pi to open the door.      The system was actively installed and used at a company I work for with about 15 employees and got to a point I had to replace it.

I decided to create a new system, ordered a wiegand reader etc.   Then discovered that wiegand isn't reliable via scripting languages and the decoder pretty much had to be in C and thus I found leosac.   Having issues with the reader thinking it was sysgpio I ordered a piface but then realized I had done the power/ground incorrectly to the reader, so not sure if the piface is really required at the moment, but it's already installed (replacing my first system), so stuck with it for the time.

*I'd be interested in working together* to do the webui if you guys/gal can implement whats needed in the c code to facilitate things in the webui.     

My idea of webui:  
-mysql/sqlite db containing system/module config, door config, user config
-bootstrap, js, ajax
-rest/remotecontrol/ipc into leosac
-leosac configured as slave/master to pull from webserver config via remotecontrol, ipc, or ?
-logging/display of access

Can you point me into the right direction to get the most current implementation, best config, and what you guys currently use as dev enviroment/hardware, so I can recreate similar enviroment (I'll have a dev enviroment at my desk at home by weeks end, but also have a production enviroment/installation)

I know I typed alot, have to head into work shortly and will be unavail for 12-13 hours again.

choleman, 2016-06-14:
Also, I started to do the webui, literally just yesterday, planning on it being a one-off and developing for myself, I started it in mysql/php.

Obviously, if trying to incorporate it into leosac and confining it to a pi, it'd be better to either

1) continue with what you guys have started
2) re-start in sqlite and python  or

mysql is difficult to get running on a pi (havent been able to yet)
php is able to run on a pi, but dependicies are more than out-of-the-box/standard pi

ngix runs better 
and
python has many modules for creating web services, rest, and apps (cherrypy)

xaqq, 2016-06-15:
Hello,

1)
The documentation is located here: http://www.leosac.com/doc_output/develop/html/pages.html.
Direct link to the AuthFile module documentation: http://www.leosac.com/doc_output/develop/html/d0/d25/mod_auth_file_main.html

To be 100% certain you have the latest documentation, I recommend running `doxygen` from the source tree. 

2)
a) I'm not sure about door, but adding which reader has read the card should be fairly simple.
b) If you mean reporting case when the user used a card + pin or simply pin, then this should also be fairly simple.
c) The authentication process is separated from the wiegand and ws-notifier module. The auth module is currently bypassed in those notification. However, it'd be possible for the ws-notifier module to start notifying authentication result in addition to wiegand event. 
d) I do not understand what you mean here.

3) Do you know what format (wiegand26, wiegand35, ...) you tag is / you reader is configured to read? When notifying a card read, the ws-notifier module tries to convert the card number to an integer. For the wiegand format, this means ignoring the FacilityCode and simply reading UID bits.
However, this is only implement for wiegand26, so other format will have their whole frame (including parity bits) converted to integer.
If you see the message "Not using format to convert WiegandCard to integer because no format match." this would explain what happens.

4) The hex value Leosac expects is the hex value produced by the `n` bits sent by the card reader.

About the Web UI.
===============

> In regards to RemoteControl and REST. I thought I had saw something somewhere where a comment was made that "REST isnt needed as it'd be accomplished thru the RemoteControl module

Yes, the RemoteControl was implemented to allows master/slave replication early on in the project. It was easiest than providing a REST API. However, with the web-ui experimentation, I've started implementing a WebSocket module. This WebSocket might end-up being the "standard" way to communicate with Leosac. This is still very a work in progress and subject to change.

An other point about the Web UI is that we would like to avoid a second "server process", whether this is a python server or php application.
The Web-UI frontend (currently written with EmberJS) talks directly to the Leosac daemon, over websocket. Sadly, as this is under development, the websocket API is not stable nor documented.
The header file for the API is https://github.com/islog/leosac/blob/webui-experiment/src/modules/websock-api/api/api.hpp

The database for the WebUI is currently managed by Leosac itself (though there is very little code as we only read credential from the db when authenticating against the web-ui). It is backed by SQLite.

> Can you point me into the right direction to get the most current implementation, best config, ...

I recommend you clone the webui-experiment branch and build it. For the web-ui, you can live with very little config. When last working my Leosac daemon was configured to load 1 module: the `WEBSOCK_API` module.

```
<?xml version="1.0"?>
<kernel>
    <instance_name>fixe</instance_name>
    <plugin_directories>
        <plugindir>.</plugindir>
    </plugin_directories>

    <network>
        <enabled>false</enabled>
    </network>

    <modules>
        <module>
            <name>WEBSOCK_API</name>
            <file>libwebsock-api.so</file>
            <level>1</level>

            <module_config>
                <port>8888</port>
            </module_config>
        </module>
    </modules>
</kernel>
```

To develop the WebUI, I didn't yet use a raspberry, and did everything from my Debian desktop.
If you'd like to look at we did currently, I recommend reading a bit about EmberJS as this is the framework we use in the browser. It seems flexible enough for our needs as it let us communicate however we want with the c++ backend.

If you have more questions, feel free to ask !

choleman, 2016-06-15:
1) I think thats the file Ive been using, ill post a sample of the auth.xml tonight.  now understanding the other questions this might be part of my problem

2)
a) that would be sufficient
b) *I* only care about card or pin, not card+pin.     
   1)  currently ws-notifier passes each digit singularly, and doesnt 
c)  #2 here is moot for me if leosac used the internal sqlite and a webui to be honest

3) 
a) Im blonde, so i might have messed up, I have the cards in the file as "34 bits", is it 35?   I didnt read the reader that close I guess.
b) yes that is the error I get
c)
fob 0006625564
ws-notifier sends 848072256
by converting fob id to binary,concat "1000000" to end, then conver back to hex I get
hex 00328c8e40  which leosac expects and accepts the fob
also i think i had to drop a digit somewhere, this would obviously be the parity bit, so ok, i kinda understand now.
anyway with this, I can now match and log a user

WebUI)

Ok, that explains alot actually.       Ill read up on the EmberJs (not familiar), clone the repo and setup.  My main machine is windows still, i have linux servers, and use the pi directly when doing my dev stuff that but should work regardless.

So I understand, theory is if using websocket-api, leo will internally log, configure, etc to its own sqlite thus negating ws-notifier etc?   

While that sounds awesome, to implement as a browser object strictly and communication via websocket sounds awesome again, it poses certain problems that I see.
1) every change, daemon has to reload/restart/hup
2) potentially harder to change ui 
3) more things are hardcoded
4) more code in leo, and actually less ui code

Ill have to clone and look to really understand I guess how it all works, I see the benefit, but I also see issues.   but.

xaqq, 2016-06-15:
Regarding your tag's format. Wiegand34 is a valid format, so it's likely that you are correct here. It's simply that leosac doesn't handle converting from raw bits to a card number for card that have wiegand34 format. A patch to support this would look like this https://gist.github.com/xaqq/49277193c48d889a26c2fc76c72477c6.

> So I understand, theory is if using websocket-api, leo will internally log, configure, etc to its own sqlite thus negating ws-notifier etc? 

No, the websocket-api and ws-notifier module have 2 different purposes. websocket-api will let client talk to, and configure leosac, while ws-notifier's goal is simply to make a remote call to a user-defined webservice to notify him that something happened.

About your observations regarding webui:
1) I think its the opposite. Since the daemon is directly talking to the web-ui, it may apply change internally without having to save the configuration then reload. It can apply the change "in-memory" and save the configuration, while avoiding reloading everything.
2) I'm not sure either. The UI is pure JS and relies on the websocket-api to retrieve / push data. However, no html/js/css code will ever be sent by the leosac daemon. All UI is in-browser JS.
3) I think this is related to 2). Since the UI is full JS there isn't anything (ui-wise) to hardcode. Unless I misunderstood you.
4) Yes, more code in Leosac. But *maybe* the code will be easier because it lives in the daemon and can do things an external process might not be able to do.

If you are doing dev directly on the pi you might face some problem (in addition to rather long compilation time). I haven't really tested the webui branch on the pi. If you are familiar with Linux it might be easier (and faster) to setup a Linux VM.

Maxhy, 2016-06-15:
About WebUI again.
Providing API is on current experimentation stage as @xaqq said and Leosac must definitely have it's own default webui at the end.

As you noticed the community is very small for now and even if we have great plans for the open source community, time is limited. So Leosac WebUI can be handled as an autonomous project from Leosac Core with its own git repo etc..., but it should be under *Leosac branding* (not a real brand btw :smile:). What I mean here is that once your tests goes to something be sure we will be happy to consider it as the official one if good enough (smart code, flexible architecture). License must be AGPLv3 to guarantee project good health for everyone.
Please do your tests, but once this done I don't know what is your goal here. If it is just to match your current requirements then thanks for sharing. If this is to provide a killing app for the community then please feel free to discuss the roadmap here, we may have some long terms ideas ;).

choleman, 2016-06-17:
xaqq:

I see the patch, and yeah I agree, it'd look something like that =)

I'm currently using the deb package, this weekend Ill set up my pi to compile from source, and pull down a git and try that patch.  =)

If the web-socket and ws-notifier are 2 different purposes, and leosac doesnt log internally and to log you must use the ws/tcp-notifier, then you're relying on another piece of software/daemon to accomplish a "complete" system, or am I missing something?

1)  ok, ill give you that
2) what server the .js file to the browser in the first place?
3) every function the js-ui invokes has to call a command to the leosacd, which has to know that command and interpret.  
4) most likely, depending upon the function, but yes most likely.

and I agree about compile time etc, but thats not matter to me really.   I can start the make and walk away for the day, it'll be done by the time I get home, besides I use a pi2, and not the original pi.

I also have a few linux vm's on a work server that has 8core, ht, 48g ram.   but normally up in the morning for an hour or 2 then gone for 10-12 hours then home. so make time doesnt mean much to me really.

choleman, 2016-06-17:
maxhy:

again, unless Im missing something, and I might be.    leo still has to log to a different software???   So whats really the problem of having an html/php/python front-end to leo?     

Now I agree, if we are talking config, if you dont want the front-end to touch the files directly and you want to code the functions into remotecontrol so that the front-end could config and manipulate leo via a websocket, thats awesome!     The front-end woudl still serve up the ember config ui or whatnot and log the access, and do this that..

Not trying to sound cocky, but.   Yes I understand your concerns, and although you have noone beating down your front door to help create it, I personally have decided to use leo.    And in doing so it lacks in areas, a ui.   I could create something myself, for myself and keep it that way, but instead had decided to reach out to the current group to figure our your ideas and to figure out how I could help instead of just creating something for myself.     

In addition i'm not much of a person to "re-create the wheel", and if I need to create something myself for an open source project to fulfill a lacking need, then i'd like to give that back also, but asking you guys what your thoughts/plans are so that we DONT duplicate efforts.   I'd rather create something to solve a problem vers solving a temporary problem, in which i'll sit back and wait if I can.  lol

So, yes, I'm ok with AGPLv3 (or whatever gpl license you prefer), and my goal is/was to help the project along, within your vision; if possible. autonomous, blessed, or non-blessed.   Again I'd rather start a path to a solution to solves everyones problems (blessed) vers just my own (non-blessed).

xaqq, 2016-06-17:
> If the web-socket and ws-notifier are 2 different purposes, and leosac doesnt log internally and to log you must use the ws/tcp-notifier, then you're relying on another piece of software/daemon to accomplish a "complete" system, or am I missing something?

The reason leosac doesn't log internally is that we didn't have the use case. We needed to notify a remote, independent system that a card had been read, thus developing ws/tcp-notifier. I agree that it makes sense to have the possibility to log from within leosac, aswel as querying the log using the websocket api.

Right now, nothing serves the JS file to the browser (expect emberjs' builtin dev server). In production I plan to have nginx running on the PI, serving the files.

> again, unless Im missing something, and I might be. leo still has to log to a different software??? So whats really the problem of having an html/php/python front-end to leo? 

As said before, leosac will be able in the future to log internally.
It's not so much about problems, it's that it seems (to me at least) simpler (and maybe cleaner) to have the remote API built into the daemon. The main reason being that: you'd need a way for leosac to talk to the python server (and vice-versa).

Thus the flow would be: `client's web browser` <-> `python webserver` <-> `leosac`. I aim for doing
`client's web browser` <-> `leosac`.
The goal being that the web UI is "simply" a way to display data coming from the leosac daemon, and providing a friendly interface to input data into leosac. Of course we would need to extend, implement and document a robust websocket API, but I think in the end its worth it to skip the "middleware php/python webservice".

> I'm currently using the deb package, this weekend Ill set up my pi to compile from source, and pull down a git and try that patch. =)

Okay. If this fails (the web-ui branch) please try in a Linux vm.

choleman, 2016-06-17:
Again, I understand and agree a simple direct  browser <> leo would be the best and not a 3 way.

So I *THINK*, now that I understand almost exactly, I know how *I* will proceed at least.

Python/php can:
   - can import modules that can talk websocket, ipc.    
   - can interact directly with the files (auth, kernel, etc)
   - has zmq module also
   - can log, and do whatever etc.
   - serve files directly also (no need for nginx even)
   -  another benefit,if leo is on an internally natted network, you redirect 1 port, 80, instead of 2 ports, 80 and 8888

I think what I'll do, again for myself but can push to a git repo or into a branch here will be a front-end thats a logger endpoint, configurator, etc etc.

I'll set it up so that front end can be dual option
  1) a front-end that  does this this this
  2) serve emberjs file to direct comm to leo

Whats lacking now, I can implement as #1, and as things get programmed, tested, debugged; I'll add the same functionality into #2.   In the end #2 would be the "preferred" way, but till its accomplished I or other people could use #1.     But when completed there would be 2 options, 

This serves my immediate use needs, serves your end goals  and I don't have to beg/plead/wait to have a certain funtionality implemented.    IF I was a c programmer, I'd of course simply add the functionality directly into leo myself, but alas I'm not, so this is the approach I must take.

What do you think?

choleman, 2016-06-18:
??

Also, is there a way to toggle a gpio?      Im installing my 2nd reader into a secondary door to the same pi, I want to be able to scan a fob, and it flips whatever the state of gpio pin, scan again, it flips again, on or off.

Why?

These are internal doors, only certain people are allowed to open the doors, once the door is opened, its unlocked for the day, when they go to leave, they scan and it'd lock the doors.

choleman, 2016-06-19:
So I've finally sitting down to pull a git and try to compile.   Which branch should I pull from, master, develop or webui-experiment?

My git skills aren't the greatest, or is there a way to pull the latest/greatest dev and get the webui-experiment?

Maxhy, 2016-06-20:
Python/php can be a temporary solution to provide REST API today but isolation and good architecture is important here from the beginning. All logic should remain on Leosac Core otherwise it will result with solutions which looks like ugly patches to resolve one user case, trace is part of it sorry. So if you are more comfortable with php/python, why not going to this architecture
 - Leosac Core, for all logics
 - Leosac Python REST API, to provide REST API only. You can plan missing feature on Core with mock REST API for now to progress on the ui at the same time, and create tickets for these missing features on core.
 - Leosac JS WebUI, for the web user interface, fully browser based as we already discussed
Also, we believe in a modular architecture and for us even the webui should be modular (for instance, we attempt to add offline locks support soon or later, we thought that the UI should mainly remains the same that was online doors). So I recommend you to *think plugin* ;)

xaqq, 2016-06-20:
Hello,

> Also, is there a way to toggle a gpio? Im installing my 2nd reader into a secondary door to the same pi, I want to be able to scan a fob, and it flips whatever the state of gpio pin, scan again, it flips again, on or off.

This use case is indeed supported. The module to configure in order to do so (react to authentication event) is the [doorman module](http://www.leosac.com/doc_output/develop/html/d6/d2b/mod_doorman_main.html). You would set your GPIO as `target` and send the `TOGGLE` command.

> So I've finally sitting down to pull a git and try to compile. Which branch should I pull from, master, develop or webui-experiment?

The `webui-experiment` branch. However, I don't think you'll be able to build it on/for the PI yet. IIRC some dependency are "hard" to build for on the PI, and I haven't tried to fix that yet. The `develop` branch should work fine on the PI tho. 

Regarding the webui, what Maxhy said makes sense if you don't want to develop directly against the leosac daemon. Note that a fully JS based UI would be great as it would be easier to eventually port it to work directly against leosac daemon.

choleman, 2016-06-20:
I think I used maybe inappropriate wording to explain my idea.

Yes, I'd write a js ui, as I think everyone agrees upon.  Not sure if thats EmberJS, or something else, I still havent looked at that.

But what I was saying was that'd interact with leo directly for what it supports, NOW..   What leo doesn't currently support I would write in python as a "middleware" or mock/psuedo interface that would in turn do whatever it needed, that the jsUI would still interact with.   So my use of the word "front-end" above, really should have been the term "middleware".    As leo starts to have more functions etc, it would start moving away from the middleware and start talking directly to leo.   

But in the end, most likely you end up with possible a somewhat complete python middleware that you could configure the jsUI to interace with either, and since the code is written, no need to delete it to the bits of heaven.

Now, by all means, once I start, and I start popping in requests and you guys want to knock them out quickly, etc, then I wouldnt write the middleware for that particular function.  

I guess I was trying to be nice by saying/thinking, its an open-source project, and hard to place demands on a group of people to add functionality that I personally need/want, when I could "hack" a way around it, until it WAS programmed by thus said group of people.      So same page, I think, just poor wording on my side and maybe some misunderstanding?

choleman, 2016-06-20:
In the case of TOGGLE, thats great to hear, and I had looked at the DOORMAN module, unfornately a lot of documentation seems to be somewhat out-dated or lacking and there are a bunch of documents with conflicting information.  

ie:
1) toggle not documented anywhere
2) auth will find the card, the door, grant/deny/etc, but no matter what ive tried it wont tie to a username it just says "someone".
3) some things in auth/authfile says <user> and other spots says <id>

Well I started down a path last night, that hasnt finished yet, and I'll wipe my pi tonight and reinstall jessie.      I took a guess at trying branch webui and it required boost1.55, which the current ver of raspbian, wheazy, didn't have, nor could install.  Trying to be smart and lazy at the same time, I thought maybe jessie would have it, but instead of just loading jessie from scratch (20 minutes), I tried to upgrade and as of this morning its still not done.  Prolly would have been, but some packages require interaction.

So I should use branch develop?   So for example, if you drop a patch in ie for the wiegand 34bit, that'd be in the develop branch?   New requests for functionality for use in the jsUI would be there too?

xaqq, 2016-06-20:
Regarding the web UI I think I understand how you want to proceed now. I think its a good idea !

About TOGGLE
--------------------

> In the case of TOGGLE, thats great to hear, and I had looked at the DOORMAN module, unfornately a lot of documentation seems to be somewhat out-dated or lacking and there are a bunch of documents with conflicting information. 

While lacking (and possibly outdated) is semi-acceptable, if you find conflicting information please tell me where so I can fix it up / clear it up / delete it.

1) `TOGGLE` is "documented" [here](http://www.leosac.com/doc_output/develop/html/d0/ddf/hardware_management.html#hardware_spec_gpio_toggle). I put documented between quote because yes, this kind of documentation would too be technical for the general end-user.
Indeed the command was documented as part of the internal message passing protocol that describes modules interfaces.

2) When you says it "just says someones" do you mean as part of the standard output log?
3) Not sure I understand. 

About branches
---------------------

The current state is a bit annoying. `develop` and `webui-experiment` are rather close (expect for webui related feature, obviously). Yes, a patch for wiegand34 would land in `develop` and I would *rebase* `webui-experiment` so it also has the patch.

I guess features that support the web ui would land in `webui-experiment`, up until we point we consider the webui not an experiment anymore and merges it into `develop`.

I gave a (not too long) try this week end to build the `webui-expirement` branch on the PI. It didn't work. It's fixable, but I think you may go crazy trying to build it. This is something I need to fix on my end. (Some dependencies needs to be built manually, however I encountered a gcc "internal compiler error" so I had to install a newer gcc, and then I end-up facing other build issues).

If you want to work on the PI to try to fix the username / invalid tag number issue you initially reported, go ahead and use `develop`. If you want to take a look at the webui stuff, use `webui-expirement` but do so in a Linux VM (at least until I fix the build).

choleman, 2016-06-21:
1) TOGGLE works great..  I just hadnt seen that option in the regular documentation is all.
2) [console] [info] AUTH_FRONTDOOR GRANTED access to target FREEDOM.FRONTDOOR for someone   -  Something is wrong in the authfile obviously, i have the schedules, the mapping, the user, credentials etc, but something is missing cause not picking up name
3) was pointing out that some documentation says/examples 1 way, while another page will referent a different conflicting way.

Tonight I plan on loading jessie then

1) Ill pull a git of develop on the pi to get regular patches etc
2) on my vm, i know how to make etc, but how do I cross compile it for the pi's arm processor vers the vm's intel processor?   Or is that non issue since its all linux?   Never really cross-compiled something before.   Or am I over thinking things and you're saying best to compile on the vm and dev on the vm also for the ui?   If that is the case, which module do I use to simulate a card scan, is that the instrumentation or is the test&reset module?

3) since you were so kind to toss in the weigand 34bit patch for me, could you toss a patch into WSNotifer on the develop branch, that would send at least cardid, door, and granted_or_denied??

choleman, 2016-06-21:
Ive tried for 3 hours now, I cant build webui-experiment, demands boost 1.55, and ive tried to build from source and everything and its an error after error.  I have 1.53.    Re-installing 1.53 and gonna try develop.    This is in my vm, which is a centos7

choleman, 2016-06-21:
-- Could NOT find AsciiDoc (missing:  ASCIIDOC_EXECUTABLE) 
CMake Warning (dev) at libzmq/tests/CMakeLists.txt:132 (message):
  Test 'test_large_msg' is not known to CTest.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- cotire 1.6.9 loaded.
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/leosac/build
[root@linux1 build]# make
[  0%] Building CXX object libzmq/CMakeFiles/libzmq.dir/src/address.cpp.o
c++: error: unrecognized command line option ‘-std=c++14’
make[2]: *** [libzmq/CMakeFiles/libzmq.dir/src/address.cpp.o] Error 1
make[1]: *** [libzmq/CMakeFiles/libzmq.dir/all] Error 2
make: *** [all] Error 2
[root@linux1 build]# 

This is as far as I could get with develop

seems like a 0mq prob but its 0mq 4.2

xaqq, 2016-06-21:
Hello,

1) Good to know.
2) Yes this is normal. The log message doesn't display the username. See [here](https://github.com/islog/leosac/blob/develop/src/modules/auth/auth-file/AuthFileInstance.cpp#L76). This is just a display issue.
3) If you have concrete link please send. You can create a new issue about this.

-------

1) Ok
2) Cross compiling is not easy. What I meant was indeed develop and test everything on the VM. The instrumentation module is what you need to simulate card insertion.
Take a look at [leopytest](https://github.com/islog/leosac/tree/develop/test_helper/leopytest) its a very small python library to ease automating testing. It contains a class that let you talk to the instrumentation module. The `WiegandClient` class is what does the work.
3) I'm gonna see if I can do something that make sense. Probably this w-e, maybe this week if I find time.

I've never used centos. Maybe the package there are too old. I don't really know.
The compilation error from your last post make me think your compiler is too old (it doesn't understand the `-std=c++14` flags. You probably need to upgrade to a more recent version.

choleman, 2016-06-21:
2) gotcha, thats no prob to me then, i was guessing it was a misconfigured auth file was all
3) when I run into them again, Ill send

1) i didnt get to the pi last night, was trying to setup the vm.  what linux flavor do you guys use?   I generally use Centos, but I can spin up another vm of whatever flavor.       If you point me to what flavor etc, Ill try web-exp again

2) thank you
3) thank you again, no rush, again cant get anything to compile yet anyway and dont have the pi sat up to compile develop yet either so, but again thank you

xaqq, 2016-06-21:
Am I running Debian myself. Jessie with some package from stretch (testing). Maybe try with Debian Stretch for your VM.

choleman, 2016-06-22:
k, looks like develop is  compiling in a vm with debian after a few apt's.   so far maybe 3 minutes and at 50%

choleman, 2016-06-22:
[ 96%] Building CXX object src/modules/ws-notifier/CMakeFiles/ws-notifier.dir/WebServiceNotifier.cpp.o
/opt/leosac/develop/src/modules/ws-notifier/WebServiceNotifier.cpp:20:23: fatal error: curl/curl.h: No such file or directory
 #include <curl/curl.h>
                       ^
compilation terminated.
src/modules/ws-notifier/CMakeFiles/ws-notifier.dir/build.make:77: recipe for target 'src/modules/ws-notifier/CMakeFiles/ws-notifier.dir/WebServiceNotifier.cpp.o' failed
make[2]: *** [src/modules/ws-notifier/CMakeFiles/ws-notifier.dir/WebServiceNotifier.cpp.o] Error 1
CMakeFiles/Makefile2:4669: recipe for target 'src/modules/ws-notifier/CMakeFiles/ws-notifier.dir/all' failed
make[1]: *** [src/modules/ws-notifier/CMakeFiles/ws-notifier.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

close but no cigar, what package do I need for the curl headers?

Liryna, 2016-06-22:
Hi @choleman ,

You need `libcurl-dev`package.

choleman, 2016-06-22:
Thank you!

libcurl4-openssl-dev to be exact, but BINGO, develop compiled!!    

webui: 
= Could NOT find ODB (missing: ODB_EXECUTABLE ODB_LIBODB_FOUND sqlite)
tried to install libsqlite3-dev and a few other -dev's but still an error, what package is that?

xaqq, 2016-06-22:
This is ODB: http://www.codesynthesis.com/products/odb/download.xhtml

The packages are (IIRC):
  + libodb-dev
  + libodb-sqlite-dev
  + odb

choleman, 2016-06-23:
Having a bitch of a time with the ODB

libtool: compile:  g++ -DHAVE_CONFIG_H -I.. -I.. -I/usr/lib/gcc/x86_64-linux-gnu/4.9/plugin/include -g -O2 -MT validator.lo -MD -MP -MF .deps/validator.Tpo -c validator.cxx 
validator.cxx:1589:1: internal compiler error: in possible_polymorphic_call_targets, at ipa-devirt.c:1557
 }
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
Preprocessed source stored into /tmp/ccM3qO77.out file, please attach this to your bugreport.
Makefile:947: recipe for target 'validator.lo' failed
make[2]: *** [validator.lo] Error 1
make[2]: Leaving directory '/root/odb/odb-2.4.0/odb'
Makefile:426: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/root/odb/odb-2.4.0/odb'
Makefile:373: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

internal compiler error?  any clues, I havent found much yet on google...  

again develop loaded fine though

xaqq, 2016-06-23:
Hello,
Yes I encountered this bug while trying to compile on the raspi.
This is a bug in gcc <5 I believe. Install gcc 5.1 would fix the issue.

choleman, 2016-06-23:
Looks like I finally got ODB compile last night and installed.

but now on compile in webui

CMake Error at /usr/share/cmake-3.0/Modules/FindPackageHandleStandardArgs.cmake:136 (message):
  Could NOT find ODB (missing: ODB_EXECUTABLE ODB_LIBODB_FOUND sqlite)
Call Stack (most recent call first):
  /usr/share/cmake-3.0/Modules/FindPackageHandleStandardArgs.cmake:343 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindODB.cmake:108 (find_package_handle_standard_args)
  CMakeLists.txt:26 (find_package)

odb is in /usr/local/bin/odb

how do I point it to the right directory?

choleman, 2016-06-23:
with develop, trying to load leosac, copied xml files from my pi

Exception propagated to main(). Will now exit.
exception: Module::Unable to init module PIFACEDIGITAL_GPIO: Actor Initialization Exception
|--> exception: Actor Initialization Exception
|----> exception: Cannot open PifaceDigital device (is SPI module enabled ?)

do I simulate a piface, need to enable spi, or do I remove that module from the config and simulate it differently?

Once I can get these branches to compile, at least in my vm, then I can start doing something.   havent even started nor tried on the pi.       Thats not really a point until theres a reason to compile on the pi (helpful/major patches); but assuming if thats the case since you said cross-compiling is a bear; you could run a armd arch compile into a package for me or is that a hard/manual thing?   Never built a package before so Im unsure.

BTW: Thanks for all the help so far in the guidance in setting up my enviroment.

xaqq, 2016-06-23:
Regarding ODB... I'm surprised it's not automatically found. Have you tried cleaning (`rm -rf`) your leosac `build` directory and run `cmake ..` again ? I don't really know otherwise.

Indeed the PifaceDigital module cannot run on a non-raspberry pi system. You'll have to remove it from the configuration. The Instrumentation can be used so you can simulate GPIO.

Building a package is semi-manual yeah.

You're welcome !

Maxhy, 2021-05-03:
Closing in favor to https://github.com/leosac/leosac-web

closed 2021-05-03 by Maxhy