leosac/access-control #72: Ability to connect door switch for exiting without authorisation

spratters, 2016-07-12:
Hi there,
We currently have a few Pis with PiFaces running Leosac, and all is working fine with a Wiegand reader. The trouble we're having is enabling the user to exit through the door without using a card, but by pressing a button. The exit button we have has 4 wires to it (2 x NO and 2 x NC). Is there a possibility of something like skipping authorisation and going directly to the doorman module when the button is pressed. It would be handy to be able to use to doorman to keep the door unlocked for a certain amount of time when exiting.
Thanks for your time.

xaqq, 2016-07-12:
Hello,

First let me tell you that I'm not familiar with push button and their wirings. Correct me if I'm wrong, but I'm going to assume that pushing the button translates to a GPIO interrupt, that would be recognized by the `piface_digital` module.

Currently the `doorman` module is only able to handle message from the authorization module. Indeed you specify `auth_contexts` when configuring it.

I don't know if you guys write C++, but if you're interested in submitting a pull request to support this use case, here are a few pointers:
  
  + GPIOs emits a `S_INT:GPIO_NAME` when an interrupt is triggered. This message is emitted on the global "Leosac bus" and therefore could be read by the doorman module.
  + In addition to `auth_context`, `doorman` module could subscribe to messages from GPIO.
  + The infrastructure to process an action (including keeping a door opened for a given amount of time) is already working. The only piece of work is to support listen to GPIO messages.
  + I believe all required code change would happen in the `doorman` module. 

> It would be handy to be able to use to doorman to keep the door unlocked for a certain amount of time when exiting.

This is already possible. 
```
<action>
  <on>GRANTED</on>
  <target>gpio_door</target>
  <cmd>
    <f1>ON</f1>
    <f2>2500</f2>
  </cmd>
</action>
```

The `<f2>` parameters controls the duration of the `ON` state. This is documented [here](http://www.leosac.com/doc_output/develop/html/d0/ddf/hardware_management.html#hardware_spec_gpio_on).

violuke, 2016-07-12:
TThanks @xaqq, I work with @spratters. We're not C++ developers but PHP developers who are happy to stray from our comfort zone :)

We've been playing about, but not getting far. You're right that the button translates to a GPIO interrupt, that is recognised by the `piface_digital` module. If we try connecting one button wire to ground and one to a spare input pin (the "normally closed" wires), "listening" to it in the `gpio` config and purely as a test, setting the wiegand reader `low` pin to this, it does trigger the wiegand module when pushing the button. Obviously it now cannot read a card properly as the module is not connected to, but this does seem to indicate how the interrupt works.

We'd rather not need to change code or build from source. Basically we have configured two  wiegand readers. One is real, and one is fake. The fake one always reads the "button push" as a card like this:
```
Source Name: VIO_FAKE_WIEGAND
Source Type: WiegandCard
Number of bits: 26
Card id: 00
```
Adding that card (`00`) to the auth file does then follow through to the door opening. The "real" reader is also reading cards ok and unlocking the door correctly. The only issue we're having now is that the "real reader" seem to trigger a read from both. As in, if we scan a card, both readers try to read the card, the result is the "fake reader" giving us a card of `00:00:00:00:00:00:00` (which the auth does not allow). The "real reader" gives us  the correct output and unlocks the door.

The button ("fake reader") when pushed only triggers a read on the fake reader, so reads `00` which we have in the auth file and therefore the door unlocks. The "fake reader" (button) does seem to get triggered twice per button push. I'm guessing that this is when it goes in and back out, but this doesn't seem to cause a problem.

This means that in practise the door locks and unlocks when it should, but when scanning a valid card we always get both the `GRANTED` and `DENIED` run (so we get our error sounding beep, but the door does unlock).

Can you help with this? We'd really appreciate it if you can :)

Is it just a config issue which is causing both readers to try to read a card when we scan a card? Our `kernel.xml` is as follows:
```
<?xml version="1.0"?>
<kernel>
	<instance_name>first_access_vio_door</instance_name>
	<plugin_directories>
		<plugindir>/usr/lib/leosac</plugindir>
		<plugindir>/usr/lib/leosac/auth</plugindir>
	</plugin_directories>

	<network>
		<enabled>false</enabled>
	</network>

	<log>
		<enable_syslog>true</enable_syslog>
		<min_syslog>INFO</min_syslog>
	</log>

	<modules>
		<!-- Piface digital module -->
		<module>
			<name>PIFACEDIGITAL_GPIO</name>
			<file>libpifacedigital.so</file>
			<level>2</level> <!-- all levels must be different, lower they are the higher the priority -->

			<module_config>
				<gpios>

					<gpio>
						<name>vio_door</name>
						<no>0</no>
						<direction>out</direction>
						<value>false</value>
					</gpio>

					<gpio>
						<name>wiegand_green</name>
						<no>2</no>
						<direction>out</direction>
						<value>false</value>
					</gpio>

					<gpio>
						<name>wiegand_buzzer</name>
						<no>4</no>
						<direction>out</direction>
						<value>false</value>
					</gpio>

					<!-- Prepare GPIO for our Wiegand reader -->
[... 322 lines cut ...]
[2016-07-12 15:37:23.345] [console] [info] Using AuthSource: Text representation of auth source:
		Source Name: FAKE_WIEGAND
		Source Type: WiegandCard
		Number of bits: 26
		Card id: 00
[2016-07-12 15:37:23.346] [console] [info] Access is granted through schedule 'all'
[2016-07-12 15:37:23.347] [console] [info] AUTH_CONTEXT_2 GRANTED access to target  for someone
[2016-07-12 15:37:23.347] [console] [debug] DOORMAN HERE
[2016-07-12 15:37:23.348] [console] [debug] ACTION (target = wiegand_buzzer)
[2016-07-12 15:37:23.348] [console] [debug] would do : ON to target: wiegand_buzzer
[2016-07-12 15:37:23.348] [console] [debug] would do : 100 to target: wiegand_buzzer
[2016-07-12 15:37:23.349] [console] [debug] Next timeout: 99
[2016-07-12 15:37:23.349] [console] [debug] ACTION (target = vio_door)
[2016-07-12 15:37:23.349] [console] [debug] would do : ON to target: vio_door
[2016-07-12 15:37:23.350] [console] [debug] would do : 5000 to target: vio_door
[2016-07-12 15:37:23.350] [console] [debug] Next timeout: 98
[2016-07-12 15:37:23.351] [console] [debug] ACTION (target = vio_door)
[2016-07-12 15:37:23.351] [console] [debug] would do : ON to target: vio_door
[2016-07-12 15:37:23.351] [console] [debug] would do : 5000 to target: vio_door
[2016-07-12 15:37:23.352] [console] [debug] Next timeout: 96
[2016-07-12 15:37:23.449] [console] [debug] Next timeout: 0
[2016-07-12 15:37:23.449] [console] [debug] UPDATING
[2016-07-12 15:37:23.450] [console] [debug] Next timeout: 4902
[2016-07-12 15:37:28.352] [console] [debug] UPDATING
```

violuke, 2016-07-13:
Thanks @xaqq, @spratters and I have been looking at this again this morning. We've set up the `MONITOR` module and will put the logs at the bottom of this message. It seems that an interrupt is wrongly coming from the `exit_button` pin when the card is scanned. We've removed and re-wired all connections and they do all seem to be solid and neat and the same thing is still happening. We also tried moving `exit_button` from input pin `7` to `6` and adjusting the config, but exactly the same thing happened. We also tried changing the earth used by the `exit_button` but that made no difference (as expected).

I'm not really sure what to do now? It is a workable solution and we're in a hurry, so we'll probably have to stick with our work around for now.

Ideally there'd be a new really simple module for a push button which simply listened to the interrupt on a single `in` pin and told the doorman `GRANTED` (no authorisation required). I guess it might be best if it could also debounce the button push (as I think this is why the button push fires multiple times (usually 2, but I think I have seen 3)). I'll try and have a look at this myself at some point but I'm not likely to have it working soon. Is this something you could help with?

Below is the logs (with `exit_button` on pin `6`):
```
Logger not set-up yet ! Will display log message as is.
Creating Leosac Kernel...
[2016-07-13 08:51:46.467] [console] [info] Network configuration disabled. Will not touch system setting.
[2016-07-13 08:51:46.469] [console] [debug] Adding {/usr/lib/leosac} in library path
[2016-07-13 08:51:46.469] [console] [debug] Adding {/usr/lib/leosac/auth} in library path
[2016-07-13 08:51:46.470] [console] [info] Attempting to load module nammed PIFACEDIGITAL_GPIO (shared lib file = libpifacedigital.so)
[2016-07-13 08:51:46.470] [console] [info] Loading library at: /usr/lib/leosac/libpifacedigital.so
[2016-07-13 08:51:46.471] [console] [debug] library file loaded (not init yet)
[2016-07-13 08:51:46.472] [console] [info] Attempting to load module nammed WIEGAND_READER (shared lib file = libwiegand.so)
[2016-07-13 08:51:46.472] [console] [info] Loading library at: /usr/lib/leosac/libwiegand.so
[2016-07-13 08:51:46.474] [console] [debug] library file loaded (not init yet)
[2016-07-13 08:51:46.475] [console] [info] Attempting to load module nammed AUTH_FILE (shared lib file = libauth-file.so)
[2016-07-13 08:51:46.475] [console] [info] Loading library at: /usr/lib/leosac/auth/libauth-file.so
[2016-07-13 08:51:46.477] [console] [debug] library file loaded (not init yet)
[2016-07-13 08:51:46.478] [console] [info] Attempting to load module nammed DOORMAN (shared lib file = libdoorman.so)
[2016-07-13 08:51:46.478] [console] [info] Loading library at: /usr/lib/leosac/libdoorman.so
[2016-07-13 08:51:46.480] [console] [debug] library file loaded (not init yet)
[2016-07-13 08:51:46.480] [console] [info] Attempting to load module nammed MONITOR (shared lib file = libmonitor.so)
[2016-07-13 08:51:46.481] [console] [info] Loading library at: /usr/lib/leosac/libmonitor.so
[2016-07-13 08:51:46.483] [console] [debug] library file loaded (not init yet)
[2016-07-13 08:51:46.485] [console] [info] Creating GPIO vio_door, with no 0. direction = out
[2016-07-13 08:51:46.485] [console] [debug] trying to bind to inproc://vio_door
[2016-07-13 08:51:46.486] [console] [info] Creating GPIO wiegand_green, with no 2. direction = out
[2016-07-13 08:51:46.486] [console] [debug] trying to bind to inproc://wiegand_green
[2016-07-13 08:51:46.486] [console] [info] Creating GPIO wiegand_buzzer, with no 4. direction = out
[2016-07-13 08:51:46.487] [console] [debug] trying to bind to inproc://wiegand_buzzer
[2016-07-13 08:51:46.487] [console] [info] Creating GPIO wiegand_data_high, with no 3. direction = in
[2016-07-13 08:51:46.518] [console] [debug] trying to bind to inproc://wiegand_data_high
[2016-07-13 08:51:46.518] [console] [info] Creating GPIO wiegand_data_low, with no 1. direction = in
[2016-07-13 08:51:46.519] [console] [debug] trying to bind to inproc://wiegand_data_low
[2016-07-13 08:51:46.519] [console] [info] Creating GPIO exit_button, with no 6. direction = in
[2016-07-13 08:51:46.519] [console] [debug] trying to bind to inproc://exit_button
[2016-07-13 08:51:46.520] [console] [info] Creating GPIO exit_button_fake, with no 0. direction = in
[2016-07-13 08:51:46.520] [console] [debug] trying to bind to inproc://exit_button_fake
[2016-07-13 08:51:46.521] [console] [info] Module PIFACEDIGITAL_GPIO is now initialized. Thread id = 1059
[2016-07-13 08:51:46.521] [console] [info] Module PIFACEDIGITAL_GPIO initialized. (level = 2)
[2016-07-13 08:51:46.522] [console] [info] Creating WiegandReader: VIO_WIEGAND_1
	 Green Led:
	 Buzzer:
	 GPIO Low: wiegand_data_low
	 GPIO High: wiegand_data_high
[2016-07-13 08:51:46.523] [console] [info] Creating WiegandReader: FAKE_WIEGAND
	 Green Led:
	 Buzzer:
	 GPIO Low: exit_button
	 GPIO High: exit_button_fake
[2016-07-13 08:51:46.524] [console] [info] Module WIEGAND_READER is now initialized. Thread id = 1060
[2016-07-13 08:51:46.524] [console] [info] Module WIEGAND_READER initialized. (level = 5)
[2016-07-13 08:51:46.525] [console] [info] Creating AuthFile instance AUTH_CONTEXT_1. Target door =
[2016-07-13 08:51:46.525] [console] [debug] Will load tree
[2016-07-13 08:51:46.526] [console] [debug] Tree loaded
[2016-07-13 08:51:46.527] [console] [debug] Ready
[... cut ...]
[2016-07-13 08:51:57.675] [monitor_stdout] [info] F0: {S_vio_door} ; F1: {ON} ;
[2016-07-13 08:51:57.675] [monitor_stdout] [info] F0: {S_wiegand_buzzer} ; F1: {ON} ;
[2016-07-13 08:51:57.675] [monitor_stdout] [info] F0: {S_wiegand_buzzer} ; F1: {OFF} ;
[2016-07-13 08:52:02.457] [console] [debug] Next timeout: 0
[2016-07-13 08:52:02.457] [console] [debug] Next timeout: 0
[2016-07-13 08:52:02.457] [console] [debug] UPDATING
[2016-07-13 08:52:02.458] [monitor_stdout] [info] F0: {S_vio_door} ; F1: {OFF} ;
```

xaqq, 2016-07-13:
Hello,
Thanks for testing this.

Looking at the log file, we indeed see interrupts coming from the `exit_button` pin. While it's possible that this is software issue, it really seems like the interrupt are really getting triggered. I have no idea why though.

I can think of one more test you could do. The Piface has 4 integrated push buttons, that IIRC map to input pin 0-3. Could you try removing your hardware button, and use the Piface button instead. If the `exit_button` as-a-piface-button doesn't receive unwanted interrupt, this may point to a wiring problem. Otherwise it may point to an other issue. 

 > Ideally there'd be a new really simple module for a push button which simply listened to the interrupt on a single in pin and told the doorman GRANTED (no authorisation required). I guess it might be best if it could also debounce the button push (as I think this is why the button push fires multiple times (usually 2, but I think I have seen 3)). I'll try and have a look at this myself at some point but I'm not likely to have it working soon. Is this something you could help with?

Either something like this yes, or improving the doorman module so that it can listen to GPIO interrupt would work too. That's something worth doing indeed, however I can't give you a timeline.
WRT debouncing the button, I don't know if this is possible. Maybe ignoring additional interrupt in a given timeframe could work. I don't really know.

Maxhy, 2021-05-03:
Will be closed as properly tracked on #78.

closed 2021-05-03 by Maxhy