leosac/access-control #73: Add Username in log

thestyxx, 2016-08-14:
In the log, I have : 
[info] AUTH_CONTEXT_1 GRANTED access to target Test.door_A for someone
and the line before the badge number.

To easily read it would it be possible more detailled user data ? Like: 
[info] AUTH_CONTEXT_1 GRANTED access to target Test.door_A for <USER> (<Firstname> <Lastname>). OR
[info] AUTH_CONTEXT_1 GRANTED access to target Test.door_A for <GROUP>/<USER> (<Firstname> <Lastname>). 

So it is easy with a grep to find all the access of an user or a group.

THanks
Guy

xaqq, 2016-08-16:
Hello,

This has indeed been ask more than once. I've pushed a commit that mostly implement your first example.
However, instead of `firstname`/`lastname` I used the `name` property (which is a unique ID) to identify which user has tried to authenticate. The reason being that firstname/lastname could be not unique and maybe harder to parse. Using the unique identifier make more sense from a logging perspective. It should not reduce your ability to grep for user.

I have not added display of group, because a user can be a member of more than one group. I'm not sure how you feel about this, but it may be better to check access from group based on access from user, instead of listing group membership anytime an access happens.

Let me know how this works for you.

thestyxx, 2016-08-16:
Hello,

thanks for the commit, however I have no compilation environment, I was using the APK.  I can test but later.

Leosac will be used in real life environment mamaging the accesses within a buidling.  The group is actually also used to identify thru which access it was granted. Ex: people renting / building manager / concierge. While people could be part of the 3 groups, they are likely to belong to one. I would anyway prefer to know from which (first) group they are getting access, again, in case of degugging that will lead me faster to the right group and schedule if someone is still getting an access that he shouldn't.

the first/lastname are also usefull, I will show you the real life example.

User                 FirstName       LastName        Group
BIEN_01_01      Marie              Lechamps        Rental
BIEN_01_02      Jean-Pierre     Durant             Rental
BIEN_02_01      Damien           Picariello         Rental
BIEN_02_01      Danielle          Zordan            Rental
BIEN_02_01      Damien           PIerrard           Rental
(...)
BIEN_ANX_01   Mireille           Donin              Concierge
BIEN_ANX_02   Florin              Kouros            Concierge
ADMIN             Guy                 Admin            Admin

The concierge appartement can change and the number is different anyway in all the building. So the group is actually giving the function. The USER field is actually the building name, appartement number and name of the person. If I do know who is 'BIEN_02_01', it is not the same for my colleague or the person replacing me. As all the badges are individual up to 4 per appartement, the name of the person within the same appartement is important to avoid having to check without the paperwork who got this badge.  

So, giving the group that was giving the access and the name related to the USER (not the name alone) is not meaningless.

Ex : RENTAL/BIEN_02_01  (Damien PIerrard) will immediately tell me that the renter of appartement 02 in BIEN with badge belonging to Damien Pierrard is getting an access.

Having multiple Damien Pierrard is not a problem as they are unique in their USER/BADGE number.

I forgot to mention that the goal is tu use it in serveral building and get the syslog back in the central site. 

Guy

Just to mention, the push button is widely used : we removed it from the doors, some kids were using the postal flap to pass their arm and push the button. In addition having to push a button AND pull the door is not easy. So we moved the button 2m away from the door. you push the button and the door is open for 5 secons : 1 hand is only used and not acessible from outisde.  This is standard on the box we use to control the doors now with an entry stating that the door was opened by the inside button.

In addition, the buttons could be linked to the button on the doorphone, letting us know which renter is giving access from his appartement to someone. This would give us an advantage in security. 

We had some dammage within the building, a guy was pushing all the bell buttons on the doorphone and one just opened letting the person enter and degrade. We were able to 'see' the person on the camera, but actually the person who opened blindly the door is also reponsible of the damage, and this info was missing. (it would need a second piface but the stacking board allow 4 of them).

I just share here the problems of the access control in real life in appartement rentals :) 

Guy

thestyxx, 2016-08-16:
Sorry, 

I cannot edit the post on my mobile. When a badge is defined and disabled (we never reuse a badge and put it as disabled to trace stolen and lost badges, is it labeled unknown or disabled?

Maxhy, 2021-05-03:
Should be fixed as expected, feel free to reopen otherwise.

closed 2021-05-03 by Maxhy