leosac/access-control #75: sysfsgpio door GPIO

mjmayer, 2016-09-17:
I have leosac successfully reading from an RFID card and granting access. However, I'm having trouble getting the GPIO pin used to signal the door to open. In my test case I'm just using an LED. The LED is constantly lit at a low level. Like GPIO pin 3 is receiving voltage all the time. I've tested GPIO pin 3 using python code to set the GPIO output to 1 and to 0 and it behaves as expected. But leosac seems to leave it in a half on state.

Here is my debug output.
> [2016-09-17 04:39:48.919] [console] [debug] DOORMAN HERE
[2016-09-17 04:39:48.921] [console] [debug] ACTION (target = my_door_gpio)
[2016-09-17 04:39:48.929] [console] [debug] would do : ON to target: my_door_gpio
[2016-09-17 04:39:48.932] [console] [debug] would do : 20000 to target: my_door_gpio
[2016-09-17 04:39:48.940] [console] [debug] Remaining = 1
[2016-09-17 04:39:48.942] [console] [debug] Next timeout: 19998
[2016-09-17 04:40:08.940] [console] [debug] Next timeout: 0
[2016-09-17 04:40:08.941] [console] [debug] Turning off SysFsGPIO pin.
/>

xaqq, 2016-09-17:
Hello !

This sounds like a weird situation. There really isn't any "half on" state: it's either on or off (`1` or `0` as a value in the `/sys/..../gpioX/value` file).

It could be a wiring problem, but since you said you tried with an external program its unlikely.
Just to confirm, what you observe is that Doorman granting access has no effect on the LED, the LED is always "half on", even when Leosac starts?

Can you paste your configuration file?

It'd be interesting if you could try loading the (undocumented) "STDIN_CONTROLLER" module. With this module enable, you can use the standard input to send command to Leosac. You could try typing `my_door_gpio ON 20000` and see if something happens.

Last question: are you using the Debian package for version 0.6.3 ? (latest available release)

mjmayer, 2016-09-17:
The best that I can tell, the direction of the GPIO is being set to "in" despite the configuration setting it to out. I determined this by looking at 'cat /sys/class/gpio/gpio3/direction' and showed in. I used echo out > /sys/class/gpio/gpio3/direction and the led started to turn on and off with a card swipe.
```
pi@raspberrypi:~ $ cat /sys/class/gpio/gpio3/direction 
in
```

LED appears "half on" even before leosac is started. If i use an external program to set the GPIO either to 1 or 0. The "half on" state goes away and the led is either on or off.

```
<gpio>
          <name>door_lock</name>
          <no>3</no>
          <direction>out</direction>
          <value>0</value>
 </gpio> 
```

[Complete Config](http://pastebin.com/Skd3WTvd)

Running version of leosac:
```
pi@raspberrypi:/sys/class/gpio $ dpkg -s leosac
Package: leosac
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 5672
Maintainer: Arnaud Kapp <kapp.arno@gmail.com>
Architecture: armhf
Version: 0.6.3-1
Depends: libboost-regex1.55.0, libboost-serialization1.55.0, libc6 (>= 2.15), libcurl3 (>= 7.16.2), libgcc1 (>= 1:4.4.0), libstdc++6 (>= 4.9), libboost-serialization1.49.0, libtclap-dev
Conffiles:
 /etc/leosac.d/factory/kernel.xml baf8a1670dd8471da6a6d0a837b2683b
 /etc/leosac.d/factory/README.md 300c8f303f71d090466dcc538b1724e2
 /etc/init.d/leosacd 8e8e333977a3c0e6bc51d8f110689449
Description: open source access control

```

xaqq, 2016-09-19:
Hello,

Thank you for providing additional information.
I am able to reproduce the issue, and am trying to find the root cause for it.

xaqq, 2016-09-19:
Well, it turns out this a documentation issue.
The example specifies:
   +  `<direction_path>/sys/class/gpio/__PLACEHOLDER__/value</direction_path>`

But it should be:
   +  `<direction_path>/sys/class/gpio__PLACEHOLDER__/direction</direction_path>`

If you fixes this in your configuration file, it should then work as expected.

commit 783c30a1 (2016-09-19): Problem: sysfsgpio example is incorrect.

The configuration example specified the `value` file instead of
the `direction` for GPIO direction.
This fixes #75.
closed 2016-09-19 by xaqq