leosac/access-control #76: Build from Develop Branch

mjmayer, 2016-10-01:
I had leosac working when installing from the rpm, but I wanted some logging features in the develop branch. I'm having some problems building from source. I'm essentially following the directions in the [documentation](https://github.com/leosac/leosac/wiki/Build-and-install) for the master branch

Is there documentation or a script for building from the develop branch?

Here is the error I'm getting.
[image]

xaqq, 2016-10-02:
Hello,

The develop branch is under heavy development at the moment and does not build on the raspi at this point.

Can you point me to the logging features you want? It may lead to a commit that can be built on the pi.

mjmayer, 2016-10-02:
Hi @xaqq,

I was looking at #73 which was a request for username in log. You made a commit that looks like it provided a username in the log in https://github.com/leosac/leosac/commit/9d54ad3ab679fe48d7ad02f3aa3b7a7e458fda81

xaqq, 2016-10-02:
The commit you linked should build okay on the pi.
Let me know if you have issue building it.

mjmayer, 2016-10-03:
I was able to get commit https://github.com/leosac/leosac/commit/9d54ad3ab679fe48d7ad02f3aa3b7a7e458fda81 to build after meeting all the dependencies. After completing a 'make install' I attempted to run leosac using the config files that worked previously. It partially starts then exits. If I run without sudo it indicates there was a seg fault. But with sudo there is no indication.
[image]

xaqq, 2016-10-03:
The fact that the program crashes like that makes me think that it tried to load modules that were built for an other version of leosac.
Can you try to run leosac "locally" from the build directory ? Copy and edit the configuration file so that library path point to the build directory.

mjmayer, 2016-10-03:
You were correct. I removed the version from the repo, reran 'make install'. I saw the 'make install' put the libraries in /usr/local/lib/leosac/ . I updated the config files where appropriate.

Your suggestion of running leosac "locally" from the build directory worked for me and put me on the right track. Thanks for your help.

closed 2016-10-03 by mjmayer

xaqq, 2016-10-03:
You're welcome. Glad it works :)