leosac/access-control #77: I need help

griggsy, 2017-01-28:
Hi, Your wiki link is down and i just started using leosac and was confused how to setup the software side of things, like adding cards and the like.

Liryna, 2017-01-28:
Hi @griggsy ,

Link was fixed 👍  https://github.com/leosac/leosac/wiki

griggsy, 2017-01-29:
i can't seem to install leosac using the .deb file.

griggsy, 2017-01-29:
My raspberry pi is stuck on committing changes when installing the .deb file
I can't seem to fix this issue

griggsy, 2017-01-29:
[image]
This is the error trying to run the command to install the dependencies.

Liryna, 2017-01-29:
I think without `:` at the end it will work 👍

xaqq, 2017-01-29:
Try running apt-get install --fix-missing then reinstall the .deb

This should pull the required and missing dependencies and allow you to
install.

griggsy, 2017-01-30:
@xaqq i tried that command and tried reinstalling the deb and is says running and when i run the command listed on the install page it gets the same error as last time 
[image]

griggsy, 2017-01-31:
Fixed it, now just to figure out how to configure the modules as the wiki is confusing

griggsy, 2017-01-31:
Think I have understood the modules bit. But how do I get it to load the auth File and config file, also how do I get the card Id to add the the auth file ?

xaqq, 2017-01-31:
You have to specify them in the main configuration file. Have you followed [this](http://www.leosac.com/doc_output/develop/html/d5/d38/install_guide.html) guide ?

The complete configuration for this example is available [here](https://github.com/leosac/leosac/tree/develop/cfg/example/guide_rpi_piface_wiegand). Hopefully it is still valid.

You configure card ids in the auth file by manually writing XML. An you run the whole thing by typing `/path/to/leosac -k /path/to/config/file.xml`

griggsy, 2017-01-31:
I mean is there a way to scan the card on my wiegand reader and see the card id to write into my auth file?

I understand how to add the modules into the config file. 😀

griggsy, 2017-02-01:
I have tired that command but i get an error like this 
[image]
This is my config 
it wont let me upload as xml so it is hosted on pastebin 
http://pastebin.com/LDJFkmay

xaqq, 2017-02-03:
Hello, 
Your xml config file is not valid XML.
You're missing a few opening tag (didn't see `<modules>`) and some closing tag near the `doorman` module configuration.

closed 2021-05-03 by Maxhy