leosac/access-control #78: Support for door contact/request to exit on GPIOs

andrew867, 2017-02-05:
Just a quick question as I couldn't find my answer when looking at the documentation and code. Is it currently possible to use a door contact (on a GPIO) to sense if/when a door is opened after a successful scan? What about assigning a GPIO as a request to exit, ex: when using a mag lock that is fail safe the magnet is always on so you need a way to either scan on an exit reader or press a button to exit. Same with monitored doors, ex: if you wanted to make sure someone didn't hold the door open or prop it open you would use a door contact along with a request to exit device (like a Kantech T-REX) to automatically sense that someone is at the door and about to open it.

If these features don't exist then consider this a feature request. :)

xaqq, 2017-02-06:
Hello,

> Is it currently possible to use a door contact (on a GPIO) to sense if/when a door is opened after a successful scan?

No, not really.

> What about assigning a GPIO as a request to exit, ...

You can read #72, there are similarity with what you want to do it seems.
Currently you'd have to work around the limitation of the doorman module (it can only understand message from the authentication module).

The "door contact device" + "request to exit" to trigger opening the door is an interesting combo but it isn't supported (yet?).

djarbz, 2018-05-12:
I think this would be a great feature! 
I install professional level door access systems and one thing all the good ones have is the ability to monitor door contacts(NC input when secure) and a request to exit device (NO input when no one is attempting to exit).
Bonus points 
-Relock the strike when the door contact is opened.
-Unlock the strike when RQE is tripped.
-Similar to #72 unlock the door with a pushbutton (receptionist lets someone in the lobby door) with a relock when the DC is opened.

Maxhy, 2022-04-22:
Request to Exit button has been added.
Door contact sensor as well but it needs a new "Alarm module" first to be useful. For now it is just for informational purpose on logs.

Maxhy, 2022-05-25:
Alarm module has been added as well.

closed 2022-05-25 by Maxhy