leosac/access-control #79: PB with files localization

joelferry, 2017-03-07:
Hello, many thanks for you software, it's seem to be a good job
sorry for my english, I'm a french user (and newby)
could you help me for localization of auth.xml file because the answer of the soft is "No owner..." but I've created the owner before. my auth.xml file is in the same path as config file "usr/share/leosac/cfg but I've seen in an issue than I must have a path + file in usr/bin, but I've just have a file leosac in this path but no path named leosac 
best regards
joel

Liryna, 2017-03-07:
Bonjour @joelferry ,

Afin de t'aider dans la configuration, peux-tu m'expliquer un peu se que tu veux faire avec leosac exactement ? Peux-tu me partager les fichiers de configuration ?

joelferry, 2017-03-07:
Bonjour

Formidable, tu parles français, ça va être plus simple.

Merci d’abord pour cette réponse rapide.

En fait, j’ai installé LEOSAC en clonant sur Github j’ai réussi à faire fonctionner le logiciel sans trop de gros problèmes.

Mais lorsque je lis un badge, il n’accède pas au fichier auth.xml qui est situé dans le même répertoire que mon fichier de config.

ci-joint mon fichier leosac.xml et auth.xml

j’ai vu dans un post qu’i y avait un répertoire situé dans usr/bin nommé leosac, or je n’ai qu’un fichier exécutable nommé leosac dans ce répertoire.

J’ai une RPI 3 avec un module PIFACE et un lecteur RFID

Je voulais déjà lire et autoriser un badge, puis par la suite gérer une petite base de données sur MySQLLite.

J’ai même imaginé faire un petit bout de programme pour enregistrer directement un badge lu dans cette base après avoir levé un PIN pour mettre le système en mode enregistrement, car je n’ai pas toujours besoin d’avoir un nom associé au badge.

Pour info, je suis installateur de systèmes de motorisation et de contrôle d’accès mais surtout un passionné de développement (mais étant un peu faignant, je ne vais pas réinventer ce qui existe surtout lorsque c’est bien fait.

Merci encore de cette rapide réponse. Si je réussi à vendre ce système, existe-il un système de rémunération pour l’auteur du logiciel LEOSAC ?

Amitiés

Joël FERRY

joelferry, 2017-03-08:
Hello, Liryna,
j'ai réussi à faire fonctionner le tout, j'ai modifié mon fichier auth.xml et ça fonctionnne, formidable
par contre est-ce normal que je sois obligé de passer le badge à 4 ou 5 reprises avant que la lecture soit bonne, le lecteur me lit en 24 bits, puis en 25 et enfin en 26 et à ce moment j'ai le bon badge
autre question, mon badge porte le n° 0000426206 et il est lu  83:40:6f:40 comment est faite la conversion ? j'ai essayé ce convertisseur http://www.ccdesignworks.com/wiegand_calc.htm mais ça ne marche pas pouvez-vous m'expliquer ?
par contre je suis absent du 10 au 21/03 inclus
amitiés
Joel Ferry

xaqq, 2017-03-08:
Bonjour,

>  est-ce normal que je sois obligé de passer le badge à 4 ou 5 reprises avant que la lecture soit bonne

Non, cependant cela m'est deja arrive. Deux raisons possible a premiere vue:

  +   Un soucis de branchement.
  +   Le programme ne tourne pas en root / n'a pas les permissions d'augmenter la priorite
       du thread qui gere les interrupts. Y a til dans les logs un message genre: "Setting realtime priority for PIFACEDIGITAL_GPIO failed." 

A propos du badge. Je ne sais pas ce que represente le numero 0000426206 mais cela n'est pas un "card code" valide en wiegand 26. En effet, en wiegand 26 (et comme indique dans ton lien) le numero de badge est encode sur 16bits. Hors la valeur max d'un nombre a 16 bits est de 65535.

Leosac et ton lien retrouve bien les meme informations. Petite explication.
Les bits sont lus depuis le lecteur. Leosac "construit" deux valeurs pour un badge lu:
 
  + Une chaine de character hexadecimal (ton "83:40:6f:40")
  + Un nombre (ici 26) qui est le nombre de bits lu. IMPORTANT: La chaine hexa contient 32bits de donnees, hors seul 26 ont un sens ici. Cela est utilise apres pour la conversion vers un nombre entier.

Ainsi, pour retrouver le "card code" wiegand, 2 etapes sont necessaires:

  + Convertir la paire (chaine-hexa, nombre-de-bit) en un nombre "raw". Pour ton example
cela donne 34406845.
  + Extraire le numero de bage (dans notre cas, les 16 bits qui nous interresse).  Cela donne 32990

Plus d'info: https://github.com/leosac/leosac/blob/develop/src/core/credentials/RFIDCard.cpp#L39-L79

joelferry, 2017-03-09:
Bonjour et merci de cette attention,

Voilà un des messages du user.log :

[syslog] [warning] Received too many interrupt. Resetting current counter.

Par contre, j’ai aussi ces messages :

Mar  9 07:17:30 raspberrypi leosac: [2017-03-09 07:17:30.308] [syslog] [error] Configuration Error (module LED_BUZZER) GPIO my_LED_gpio doesn't exist.

Mar  9 07:17:30 raspberrypi leosac: [2017-03-09 07:17:30.359] [syslog] [error] Configuration Error (module LED_BUZZER) GPIO my_buzzer_gpio doesn't exist.

Mar  9 07:17:30 raspberrypi leosac: [2017-03-09 07:17:30.367] [syslog] [error] Configuration Error (module DOORMAN) Object my_door_gpio cannot be found.

J’ai surement dû louper quelque chose, mais je suis nouveau et surtout pas très à l’aise avec l’anglais.

J’ai aussi viré la PIFACE pour installer le module SYSFSGPIO et là, la lecture des bits varie entre 17 et 24, jamais plus 26

Quant au code 0000426206, c’est le numéro qui est gravé sur le badge, serait-ce du wiegand 40 ?

Merci encore de votre aide

Joël FERRY

xaqq, 2017-03-31:
Bonjour,

> [syslog] [warning] Received too many interrupt. Resetting current **counter.**

Ca signifie que le module Wiegand a recu trop d'interrupt. Cela pourrait eventuellement indiquer des problemes de branchements. 

>  Mar 9 07:17:30 raspberrypi leosac: [2017-03-09 07:17:30.308] [syslog] [error] Configuration Error (module LED_BUZZER) GPIO my_LED_gpio doesn't exist.
Mar 9 07:17:30 raspberrypi leosac: [2017-03-09 07:17:30.359] [syslog] [error] Configuration Error (module LED_BUZZER) GPIO my_buzzer_gpio doesn't exist.
Mar 9 07:17:30 raspberrypi leosac: [2017-03-09 07:17:30.367] [syslog] [error] Configuration Error (module DOORMAN) Object my_door_gpio cannot be found.

Cela n'est pas trop grave, ca indique qu'il y a des references vers des objects inexistant. L'ideal est de clean le fichier de config d'enlever ce qui est inutilise.

> Quant au code 0000426206, c’est le numéro qui est gravé sur le badge, serait-ce du wiegand 40 ?

Je ne saurais pas dire.

Maxhy, 2021-05-03:
I assume this is fixed.
@joelferry feel free to comment about your experience with Leosac and this customer over the years now (even in French ;)). Thanks.

closed 2021-05-03 by Maxhy