leosac/access-control #80: use relays instead of digital output

makasper, 2017-03-25:
I would like to use a relay instead of the digital Output to open my door.
I'm working with the released and compiled Version for raspberry.
The released Version is more then one year old. When will normally new versions released?
It would be perfect to have a new release with the above mentioned Feature.

best regards
Markus

xaqq, 2017-03-31:
Hello,

Yes the last release is indeed quite old.
I hope I'll be able to put up a new release during april.

Regarding your issue, I'm not sure I understand.
Are you using a PifaceDigital device? If yes, then you can use its relay (as GPIO output 0 and 1, respectively).

makasper, 2017-03-31:
Thanks a lot, I thought that the relays have to be controlled on a different way.
Now I checked the Manual again and reconfigured it to GPIO 0, and now it works.
Now I can start testing on my real door, but it would be great if you could find some time in April to generate a new release.

xaqq, 2017-04-03:
Good to know that it works.
I'm going to close this issue since it's solved :)

closed 2017-04-03 by xaqq