leosac/access-control #84: Without PiFace

sercanbektas, 2017-06-19:
Hi,
I already install leosac and try to run it on Raspberry Pi 3. I have a Wiegand reader. Is it possible to read 26 bit wiegand data without PiFace module? If yes can you share an example config file

xaqq, 2017-06-19:
Hello,

Yes it is possible to do so.
You'll have to use the SYSFS_GPIO module.

Have a look at http://www.leosac.com/doc_output/develop/html/d2/d1b/mod_sysfsgpio_main.html for the documentation of SYSFS_GPIO.

You can have a look at https://github.com/leosac/leosac/blob/develop/cfg/dev-test/rpi-sysfs.xml too or the wiegand + piface tutorial and just use SYSFS_GPIO instead of PIFACEDIGITAL_GPIO.

xaqq, 2017-06-29:
Hello,
Did you manage to configure how you like? Can I close this?

sercanbektas, 2017-06-29:
Hello,
Thank you very much.Sorry for my late reply.Yes i can start program but i
cant see wiegand code which i swap to reader. Can you help me to show mu
wiegand  data on console. I want to learn my cards wiegand data and give
permission to that card. I have 37 bit card and nothing happens when i swap
card.
Best Regards

şunu yazdı:

xaqq, 2017-06-29:
Hello,

Can you show your config file?
Are you confident in your wiring of the pi and the reader?

sercanbektas, 2017-06-30:
Hello,
Config file is what you sent me before and wiring photos is in attachments.
Thanks for your interest.That will be very helpfull for me. If i see that
card is read successfuly that is enough for me.
Best Regards

şunu yazdı:

xaqq, 2017-07-01:
Hello,

Attachment do not work :(

sercanbektas, 2017-07-01:
Hello,
I attached again...sorry:(

şunu yazdı:

xaqq, 2017-07-03:
Hello,
What I meant was that sending attachment by email doesn't work at all on github.
They aren't attached to the issue. You have to upload them yourself and provide a link to them.

sercanbektas, 2017-07-03:
[image]
[image]
[image]
[image]

xaqq, 2017-07-04:
Hello,

So I'm not sure I understand correctly, but as far as I can tell your reader isnt powered, is it?
It seems you wired data0/data1 and the ground, but the reader seems to receive no power.

 What is the reference of your reader?

Rokia88, 2019-12-13:
I am facing the same issue , have you managed to make it work without piface ?

Maxhy, 2022-03-10:
I have updated the documentation on https://leosac.github.io/leosac-doc/doc_output/develop/ see https://leosac.github.io/leosac-doc/doc_output/develop/d2/d23/page_module_wiegand.html#mod_wiegand_wiring

For the record, Wiegand format decoding and advanced RFID features will be part of another Leosac update.

Closing this ticket then, feel free to comment for additional questions on that matter. Thanks.

closed 2022-03-10 by Maxhy