leosac/access-control #86: SQL implemenation and documentation

nixator, 2017-07-24:
Hello, I successfull install Leosac but I miss documentation about SQL - default schema, example data etc. I need SQL for read auth attempts and checking these incidents on cameras so I need something "machine friendly" like SQL with auth data. 

I tryed configure MySQL in Leosac but it doesnt do anything.

Thanks

xaqq, 2017-07-24:
Hello,

I imagine that you installed Leosac using the 0.6.3 Debian package? If yes, then this version is not SQL aware, and has zero SQL support.

### In the future

The current `develop` branch brings SQL support to the table, but I've not yet made a release (despite people asking and me promising to do so :/). The reason is that, while the is "core" SQL support, modules do not work with SQL yet. So even if you used the latest from git repository, you could not do what you want yet. Note that in the future, I expect your use case to be supported through the websocket API (or directly querying the database if you prefer).

### Right now

Despite Leosac not being ready for what you want to do, there is a possible workaround
that could interest you.

You could use the [WebServiceNotifier](http://www.leosac.com/doc_output/develop/html/d8/d3d/mod_ws-notifier_main.html) module to be notified of card that are read by the Wiegand module.
If you setup a small HTTP server, this module could allow to receive "card read" event. You could then store them yourself for later analysis or do what you need with it.

However, this solution is far from perfect: the module doesn't know if the authentication was a success or a failure, it just knows that a credential was read. The WebServiceNotifier module could be (probably easily) be extended to report an "authentication result" in addition to a "credential read" event.

nixator, 2017-07-25:
Hello, 

WebServiceNotifier is exactly what I needed. I thank you for a great project.

nixator, 2017-07-25:
Pity that WS-notifier send only card-id. If it could have sent also auth-source it will be fantastic.

xaqq, 2017-07-25:
Hey,

> Pity that WS-notifier send only card-id.

Yes, unfortunately, WS-Notifier is very simple and only send the card-id (after trying to convert it to Wiegand26 or 34 format).

If you write C++ I'd gladly accept a patch extending the features of WS-Notifier so that it can provide the source, or even authentication result. Though if I find some time I may write one myself to provide that functionality.

> Or is possible to run multiple instances of WS-notifier module for separate auth-source and target pair?

Modules are not designed to be run multiple times, so running multiple instances would not work.
Maybe you could workaround that limitation (haha this sounds so hacky forgive me) and try to run two instance of Leosac itself. I've never tried that, but on top of my head it could work if you are using SYSFS_GPIO module. I believe that with PIFACEDIGITAL_GPIO it may not work.

PS: Are you running on raspberry pi with multiple reader attached? Regarding compiling latest `develop` commit, did you compile the project the Pi itself?

nixator, 2017-07-26:
Wow, I see, that you made commit with requested improve. Big thanks to you. Im going to recompile and test it...

Yes, we have attached two readers via GPIO directly. 

Do you have some PayPal, It will be my honor to send you some reward.

xaqq, 2017-07-27:
Hey,

Yes hopefully this commit work for you. Just note that the `auth_source` will be prefixed by `S_` (this is due to an internal implementation detail).

Since you asked, I've set up a paypal account for the project, dev@leosac.com if you wish to donate.

Be sure to let me know if this patch works for you !

nixator, 2017-07-27:
Hello I was not able to compile develop branch because some odb libraries missing. I use gcc 4.9.2.

```
CMake Error at /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find ODB (missing: ODB_EXECUTABLE pgsql mysql sqlite boost)
```

xaqq, 2017-07-28:
Hello,

Ah I see, you compiled the rather old master branch, right?
I was a bit surprised someone compiled the current develop :P
The build process is a bit complicated and I haven't tried it on the Pi
yet. I am using
cross compilation for now.

Ok then your best bet would be to checkout tag 0.6.3 and apply/port my last
commit that you
are interested in, and compile that.

I need to cut a release, but I'll be going on vacation with no internet
tomorrow, I doubt i'll have time
in the meantime.

nixator, 2017-07-28:
Hello, after little fight because in old 0.6.3 version of WS module was little different code and I had rewrite some things but I win and Leosac send auth source. Thank you very much. I send you little gift.

http://imageshack.com/i/pmMNWpN6p

closed 2017-08-05 by nixator

xaqq, 2017-08-10:
Hey !
Thank for the gift and congratz on getting it working :)