leosac/access-control #89: MySQL and replication

mojo3120, 2017-08-13:
Hi,

Was about to post here and ask for help after installing the debian package and spending a few hours trying to get it to write to mysql, but just found the closed post from a week ago which answers that question. Is SQL support likely in the near future or will that be a while?

Would like to implement this in a multi site project. Until sql support is implemented, was thinking an alternative solution may be to write a simple web interface with mysql back end. Setup a php or python cron job on each unit to connect to remote mysql and generate a new auth.xml file from database. Feasible?

Thanks for the project. Have checked out quite a few and this seems to be the most well written and thought out.
-Andrew

xaqq, 2017-08-14:
Hello Andrew,

I hope we can release SQL support at the end of the year. However, I doubt support for multi-site / cluster will be available yet. I would like to implement better cluster support than "multiple Leosac instances pointing at shared database"... no idea about the time frame tho.

Yes your solution is feasible. Actually, someone already used Leosac like this in production.
He was generating auth.xml files using his own system, synchronizing (`rsync`) them to Leosac instances and restarting those instances. What you want to do is similar, and is a simple and clean setup that should work with no issues.
So the best way (for now) to use Leosac in a multi site configuration is to use in "normal mode" (no replication etc) and perform the synchronization / generation of auth configuration yourself.

mojo3120, 2017-08-17:
Sounds good. 

Are stacked pifaces with different spi addresses supported? Will probably roll my own pcb, but it'll be the same schematically as the piface.

xaqq, 2017-08-21:
Hey,

Regarding stacked piface, I've never tried as I don't have a Piface rack to test.
Looking at Leosac code, this would currently be unsupported: we open the device using hardware address 0, and this is hardcoded.

However, I **think** it should be fairly to support. AFAIK its possible to call `pifacedigital_open()` with a given hardware address, which should be (mostly) enough to support the use case.

mojo3120, 2017-09-01:
This is the board I designed. I'd be glad to mail you a copy if you'd like.

[image]
[image]

xaqq, 2017-09-04:
Hello,

Yeah I'd be interested in a sample of your board !
However, I don't have skill reading hardware schematics, so I will need some help to make sense of it.

Is this a board composed of multiple `mcp23s17` (the underlying device on the piface if I remember correctly) devices? If yes, how do I know which "hardware adress" an input/output on the board maps to ?

Shout me an email at kapp [DOT] arno [AT] gmail [DOT] com !

pavel604, 2017-09-05:
Hi mojo3120,

I am interested in fabrication of a similar board for RPI/Touchscreen/NFC, 
Would be interested to hear your opinion on custom PCB shops, pricing, lead times etc. who accept Eagle files
Thanks

mojo3120, 2017-09-06:
> Hi mojo3120
I am interested in fabrication of a similar board for RPI/Touchscreen/NFC,
Would be interested to hear your opinion on custom PCB shops, pricing, lead times etc. who accept Eagle files
Thanks

I've had around 10 orders from osh park. They're great for smaller boards, but they charge by the sq inch so can get relatively expensive for larger boards. Average turnaround time is probably 1.5 weeks (they're US made).

I ordered these boards from elecrow, which will be my second order from them. Don't really have any complaints on my last order. The boards aren't as pretty as osh park, but no problems with the fab and they all worked just fine. They're dirt cheap! 10 copies of this board (10cm x 10cm) cost $10 + shipping. They don't accept eagle brd files, but you can generate your own gerbers pretty easily. They provide a cam file to use in their wiki, along with instructions how to generate gerbers.

Dirtypcb's is another chinese fab I've heard good things about, but never used them personally.

nixator, 2019-01-24:
Hello, is this project alive? I going to do generaly same board so If this project is alive it can save some time... Thanks!

nixator, 2020-07-16:
after some time I made HW update... https://photos.app.goo.gl/cX6htVHS83aTDqEaA

Maxhy, 2021-05-03:
MySQL should be supported now, closing this ticket then.

@nixator the project has several users and a lot of interest. Even from ourselves, for several reasons we (original contributors and project owners) put it on hold for a few years. That do not stop any contribution from other sources if anyone want to contribute no worries. Future is clearer now and we have plan to continue the work in a few months.
Otherwise your work is good thanks for the contribution 👍 , improving the hardware documentation and HW contribution listing will be one of the first goal when awake from sleep.

closed 2021-05-03 by Maxhy