leosac/access-control #92: SPI Transaction Error

sercanbektas, 2017-10-05:
Hi,
When we run leosac with piface module there is an info about "There was a error during the SPI Transaction"

xaqq, 2017-10-05:
Hello,

Can you provide some additional information please:
   + What version of leosac are you running?
   + Are you running leosac as root?
   + What is your linux-kernel version ?
   + Are you sure linux-kernel's SPI module is enabled?
   + The full log output.

Thanks

sercanbektas, 2017-10-05:
Leosac and rapberry is the latest one. Today we pull and upgrade raspberry.
Yes we are running as root
Yes SPI module is enabled.

xaqq, 2017-10-05:
Can you provide the log output please?

The error you get "There was a error during the SPI Transaction" is a rather low-level error from either `libpifacedigital` or `libmcp23s17`.

Can you show your configuration file? Maybe an error in config could cause this.

sercanbektas, 2017-10-06:
Linux raspberrypi 4.9.52-v7+ #1038 SMP Fri Sep 29 16:26:52 BST 2017 armv7l GNU/Linux,

-----
[2017-10-06 08:10:38.796] [console] [info] Network configuration disabled. Will not touch system setting.
[2017-10-06 08:10:38.797] [console] [debug] Adding {.} in library path
[2017-10-06 08:10:38.797] [console] [debug] Adding {/root/leosac/lib/leosac} in library path
[2017-10-06 08:10:38.798] [console] [info] Attempting to load module nammed PIFACEDIGITAL_GPIO (shared lib file = libpifacedigital.so)
[2017-10-06 08:10:38.798] [console] [info] Loading library at: ./libpifacedigital.so
[2017-10-06 08:10:38.810] [console] [debug] library file loaded (not init yet)
[2017-10-06 08:10:38.810] [console] [info] Attempting to load module nammed WIEGAND (shared lib file = libwiegand.so)
[2017-10-06 08:10:38.810] [console] [info] Loading library at: ./libwiegand.so
[2017-10-06 08:10:38.819] [console] [debug] library file loaded (not init yet)
[2017-10-06 08:10:38.819] [console] [info] Attempting to load module nammed AUTH_FILE (shared lib file = libauth-file.so)
[2017-10-06 08:10:38.819] [console] [info] Loading library at: ./libauth-file.so
[2017-10-06 08:10:38.850] [console] [debug] library file loaded (not init yet)
[2017-10-06 08:10:38.850] [console] [info] Attempting to load module nammed DOORMAN (shared lib file = libdoorman.so)
[2017-10-06 08:10:38.850] [console] [info] Loading library at: ./libdoorman.so
[2017-10-06 08:10:38.862] [console] [debug] library file loaded (not init yet)
mcp23s17_write_reg: There was a error during the SPI transaction.
mcp23s17_write_reg: There was a error during the SPI transaction.
mcp23s17_write_reg: There was a error during the SPI transaction.
mcp23s17_write_reg: There was a error during the SPI transaction.
mcp23s17_write_reg: There was a error during the SPI transaction.
[2017-10-06 08:10:38.865] [console] [info] Creating GPIO my_door_gpio, with no 3. direction = out
[2017-10-06 08:10:38.865] [console] [debug] trying to bind to inproc://my_door_gpio
mcp23s17_read_reg: There was a error during the SPI transaction.
[2017-10-06 08:10:38.872] [console] [info] Creating GPIO wiegand_data_high, with no 4. direction = in
[2017-10-06 08:10:38.873] [console] [debug] trying to bind to inproc://wiegand_data_high
mcp23s17_read_reg: There was a error during the SPI transaction.
[2017-10-06 08:10:38.873] [console] [info] Creating GPIO wiegand_data_low, with no 2. direction = in
[2017-10-06 08:10:38.874] [console] [debug] trying to bind to inproc://wiegand_data_low
mcp23s17_read_reg: There was a error during the SPI transaction.
mcp23s17_write_reg: There was a error during the SPI transaction.
mcp23s17_read_reg: There was a error during the SPI transaction.
[2017-10-06 08:10:38.875] [console] [info] Module PFDigital initiliazed.
[2017-10-06 08:10:38.876] [console] [info] Module {PIFACEDIGITAL_GPIO} initialized. (level = 2
mcp23s17_read_reg: There was a error during the SPI transaction.
[2017-10-06 08:10:38.924] [console] [info] Creating READER MY_WIEGAND_1 (green led = ), buzzer {}
[2017-10-06 08:10:38.925] [console] [info] Module WiegandReader initialized.
[2017-10-06 08:10:38.925] [console] [info] Module {WIEGAND} initialized. (level = 5
[2017-10-06 08:10:38.927] [console] [info] Creating Auth instance AUTH_CONTEXT_1
[2017-10-06 08:10:38.927] [console] [info] Auth instance (AUTH_CONTEXT_1) subscribe to MY_WIEGAND_1
[2017-10-06 08:10:38.928] [console] [info] Module AuthFile initiaziled.
[2017-10-06 08:10:38.928] [console] [info] Module {AUTH_FILE} initialized. (level = 41
[2017-10-06 08:10:38.930] [console] [info] Creating Doorman instance A_DOORMAN_INSTANCE
[2017-10-06 08:10:38.931] [console] [info] Doorman Module initialized.
[2017-10-06 08:10:38.932] [console] [info] Module {DOORMAN} initialized. (level = 50

[config.txt](https://github.com/leosac/leosac/files/1361573/config.txt)

xaqq, 2017-10-07:
Hey,

This is weird. Can you test that you can control your piface device using the pifacetools?
This is to checks whether its a possible leosac issue or if its an issue with your system or device.

Maxhy, 2021-05-03:
Closing for inactivity.

closed 2021-05-03 by Maxhy