leosac/access-control #94: Multiple piface support

xaqq, 2017-10-07:
Pinging @mojo3120

I received your board and added support for multiple piface device.
See https://github.com/leosac/leosac/tree/multi-piface

I have an issue with reading interrupt state from the underlying devices.
Assume I have a reader plugged in 0 and 1 of hardware device 1 and another one
in 0 1 (or elsewhere, doesn't seem to matter) on hardware device 2.

Interrupts trigger correctly, but reading interrupt is incorrect. Whenever a reader reads a card,
the code also detects interrupts from the other device.
I need to investigate more.

Have you successfully manage to read interrupts from your device using some tools?

Maxhy, 2021-05-03:
Ping @mojo3120 as this is likely not not be merged without his support. Thanks.

Maxhy, 2022-03-05:
@xaqq was this still working properly with an unique PiFace and without the board from @mojo3120?
If it is the case then it is fine to assume issue is on the board and we can merge as it is. Otherwise I will leave open.

Thanks.

Maxhy, 2022-03-10:
It has already been merged.

closed 2022-03-10 by Maxhy