leosac/access-control #95: What am I doing wrong? SPI transaction error. Please help.

gorkemhacioglu, 2017-10-11:
Here is my all files and error log.
`pi@raspberrypi:~ $ cd leosac
pi@raspberrypi:~/leosac $ cd build
pi@raspberrypi:~/leosac/build $ sudo ./leosac -d /home/pi/leosac/build -k /home/pi/leosac/cconfig.xml
[2017-10-11 13:37:45.746] [console] [info] Network configuration disabled. Will not touch system setting.
[2017-10-11 13:37:45.747] [console] [debug] Adding {/home/pi/leosac/build} in library path
[2017-10-11 13:37:45.747] [console] [info] Attempting to load module nammed PIFACEDIGITAL_GPIO (shared lib file = libpifacedigital.so)
[2017-10-11 13:37:45.748] [console] [info] Loading library at: /home/pi/leosac/build/libpifacedigital.so
[2017-10-11 13:37:45.750] [console] [debug] library file loaded (not init yet)
[2017-10-11 13:37:45.750] [console] [info] Attempting to load module nammed WIEGAND (shared lib file = libwiegand.so)
[2017-10-11 13:37:45.751] [console] [info] Loading library at: /home/pi/leosac/build/libwiegand.so
[2017-10-11 13:37:45.752] [console] [debug] library file loaded (not init yet)
[2017-10-11 13:37:45.752] [console] [info] Attempting to load module nammed AUTH_FILE (shared lib file = libauth-file.so)
[2017-10-11 13:37:45.753] [console] [info] Loading library at: /home/pi/leosac/build/libauth-file.so
[2017-10-11 13:37:45.755] [console] [debug] library file loaded (not init yet)
[2017-10-11 13:37:45.756] [console] [info] Attempting to load module nammed DOORMAN (shared lib file = libdoorman.so)
[2017-10-11 13:37:45.756] [console] [info] Loading library at: /home/pi/leosac/build/libdoorman.so
[2017-10-11 13:37:45.758] [console] [debug] library file loaded (not init yet)
mcp23s17_write_reg: There was a error during the SPI transaction.
mcp23s17_write_reg: There was a error during the SPI transaction.
mcp23s17_write_reg: There was a error during the SPI transaction.
mcp23s17_write_reg: There was a error during the SPI transaction.
mcp23s17_write_reg: There was a error during the SPI transaction.
[2017-10-11 13:37:45.760] [console] [info] Creating GPIO my_door_gpio, with no 3. direction = out
[2017-10-11 13:37:45.760] [console] [debug] trying to bind to inproc://my_door_gpio
mcp23s17_read_reg: There was a error during the SPI transaction.
[2017-10-11 13:37:45.761] [console] [info] Creating GPIO wiegand_data_high, with no 4. direction = in
[2017-10-11 13:37:45.761] [console] [debug] trying to bind to inproc://wiegand_data_high
mcp23s17_read_reg: There was a error during the SPI transaction.
[2017-10-11 13:37:45.761] [console] [info] Creating GPIO wiegand_data_low, with no 2. direction = in
[2017-10-11 13:37:45.761] [console] [debug] trying to bind to inproc://wiegand_data_low
mcp23s17_read_reg: There was a error during the SPI transaction.
mcp23s17_write_reg: There was a error during the SPI transaction.
mcp23s17_read_reg: There was a error during the SPI transaction.
[2017-10-11 13:37:45.762] [console] [info] Module PFDigital initiliazed.
mcp23s17_read_reg: There was a error during the SPI transaction.
[2017-10-11 13:37:45.762] [console] [info] Module {PIFACEDIGITAL_GPIO} initialized. (level = 2
[2017-10-11 13:37:45.820] [console] [info] Creating READER MY_WIEGAND_1 (green led = ), buzzer {}
[2017-10-11 13:37:45.821] [console] [info] Module WiegandReader initialized.
[2017-10-11 13:37:45.821] [console] [info] Module {WIEGAND} initialized. (level = 5
[2017-10-11 13:37:45.822] [console] [info] Creating Auth instance AUTH_CONTEXT_1
[2017-10-11 13:37:45.823] [console] [info] Auth instance (AUTH_CONTEXT_1) subscribe to MY_WIEGAND_1
[2017-10-11 13:37:45.823] [console] [info] Module AuthFile initiaziled.
[2017-10-11 13:37:45.823] [console] [info] Module {AUTH_FILE} initialized. (level = 41
[2017-10-11 13:37:45.824] [console] [info] Creating Doorman instance A_DOORMAN_INSTANCE
[2017-10-11 13:37:45.825] [console] [info] Doorman Module initialized.
[2017-10-11 13:37:45.825] [console] [info] Module {DOORMAN} initialized. (level = 50
`
`https://utis-my.sharepoint.com/personal/gorkem_hacioglu_utis_com_tr/_layouts/15/guestaccess.aspx?docid=0826fb8cd619a4326b49afa683bce392e&authkey=AZ7TT0Wf5aussDkldhFt-N4&expiration=2017-12-31T21%3a00%3a00.000Z`
[config_auth_files.zip](https://github.com/leosac/leosac/files/1375750/config_auth_files.zip)

xaqq, 2017-10-15:
Hello,

Can you please check, using the pifacedigital tools, that you can correctly control your piface?

gorkemhacioglu, 2017-10-15:
Yes. I can control my piface with tools

xaqq, 2017-10-16:
Okay, a few follow-up questions:
  + Are you sure you are running leosac as root?
  + Are you sure the hardware address of your piface (controlled through jumpers) is set to 0?
  + Are you running leosac 0.6.3 (installed with the debian package) ?
  + What is your kernel version?
  + What model of raspberry pi is it?

I'm trying to gather as much info as I can because I don't know why you and @sercanbektas are experiencing these issues.

closed 2021-04-25 by gorkemhacioglu