leosac/access-control #97: Roller Shutter

Chapster5, 2017-10-18:
Hi there,

Looking at this as a possible solution to controlling a roller shutter so when a card is authorised the shutter opens for x amount of time then closes back down. Would this be possible with this software and system? 

Cheers,

xaqq, 2017-10-24:
Hello,

I don't know how shutter works, but maybe.
Do you have some control wire that open and close the shutter?

The software can drive up a pin for open the shutter and an other one to close it.
So with a relay its probably possible. 
Can you explain the hardware of the shutter a bit more ?

Chapster5, 2017-10-25:
Many thanks for your response that sounds great. The Roller shutter currently has a rfid control on it at the moment but from what I was told the control board has extra pin ins to attached external hardware such as a button or key control. So this sounds like it should work great!