leosac/access-control #98: Build failure - db pragma xxxxx is not associated with a declaration

knnniggett, 2018-01-14:
I've been working through build issues with the develop branch, mostly missing package dependencies, when I came upon an issue I had not seen before. 

The same output shown below was generated on two different Debian stretch machines, one virtual arm64 and the other arm - raspi zero. 

Relevant build output is shown below. 

This might be relevant, but I'm not sure:
https://www.codesynthesis.com/pipermail/odb-users/2016-December/003581.html

That was over a year ago. One would think the fix for that issue would have made its way into the debian repos by now.

I am very unfamiliar with odb, and could use some help troubleshooting this.

installed gcc and odb packages:
```
abauer@vDebian:~/build/leosac/build$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/6/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 6.3.0-18' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 6.3.0 20170516 (Debian 6.3.0-18) 

abauer@vDebian:~/build/leosac/build$ dpkg -l |grep odb
ii  libodb-2.4                           2.4.0-1+b1                     amd64        Common ODB Runtime Library
ii  libodb-boost-2.4                     2.4.0-1+b1                     amd64        Boost ODB runtime library
ii  libodb-boost-dev                     2.4.0-1+b1                     amd64        Boost ODB runtime library (development)
ii  libodb-dev                           2.4.0-1+b1                     amd64        Common ODB Runtime Library (development)
ii  libodb-mysql-2.4                     2.4.0-3                        amd64        ODB Runtime Library for MySQL
ii  libodb-mysql-dev                     2.4.0-3                        amd64        ODB Runtime Library for MySQL (development)
ii  libodb-pgsql-2.4                     2.4.0-1+b1                     amd64        ODB Runtime Library for PostgreSQL
ii  libodb-pgsql-dev                     2.4.0-1+b1                     amd64        ODB Runtime Library for PostgreSQL (development)
ii  libodb-qt-2.4                        2.4.0-2+b1                     amd64        Qt ODB runtime library
ii  libodb-qt-dev                        2.4.0-2+b1                     amd64        Qt ODB runtime library (development)
ii  libodb-sqlite-2.4                    2.4.0-1+b1                     amd64        ODB Runtime Library for SQLite
ii  libodb-sqlite-dev                    2.4.0-1+b1                     amd64        ODB Runtime Library for SQLite (development)
ii  odb                                  2.4.0-5+b1                     amd64        C++ Object-Relational Mapping (ORM)
```

Build output:
```
[  7%] Generating core-schema-sqlite.cxx, core-schema-pgsql.cxx
/home/abauer/build/leosac/src/core/auth/User.hpp:154:12: error: db pragma 'id' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:154:15: error: db pragma 'auto' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:160:12: error: db pragma 'unique' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:161:12: error: db pragma 'not_null' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:162:12: error: db pragma 'type' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:162:12: error: db pragma 'type' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:165:12: error: db pragma 'type' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:165:12: error: db pragma 'type' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:172:12: error: db pragma 'value_not_null' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:172:12: error: db pragma 'value_not_null' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:172:27: error: db pragma 'inverse' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:175:12: error: db pragma 'not_null' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:178:12: error: db pragma 'value_not_null' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:178:12: error: db pragma 'value_not_null' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:178:27: error: db pragma 'inverse' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:184:12: error: db pragma 'value_not_null' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:184:12: error: db pragma 'value_not_null' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:184:27: error: db pragma 'inverse' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:191:12: error: db pragma 'transient' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/User.hpp:194:12: error: db pragma 'version' is not associated with a declaration
/home/abauer/build/leosac/src/core/credentials/Credential.hpp:75:12: error: db pragma 'id' is not associated with a declaration
/home/abauer/build/leosac/src/core/credentials/Credential.hpp:75:15: error: db pragma 'auto' is not associated with a declaration
/home/abauer/build/leosac/src/core/credentials/Credential.hpp:86:12: error: db pragma 'value_not_null' is not associated with a declaration
/home/abauer/build/leosac/src/core/credentials/Credential.hpp:86:12: error: db pragma 'value_not_null' is not associated with a declaration
/home/abauer/build/leosac/src/core/credentials/Credential.hpp:86:27: error: db pragma 'inverse' is not associated with a declaration
/home/abauer/build/leosac/src/core/credentials/Credential.hpp:89:12: error: db pragma 'version' is not associated with a declaration
/home/abauer/build/leosac/src/core/auth/Door.hpp:67:12: error: db pragma 'id' is not associated with a declaration
[... 147 lines cut ...]
/home/abauer/build/leosac/src/hardware/Device.hpp:43:28: error: db pragma 'table' is not associated with a declaration
/home/abauer/build/leosac/src/hardware/GPIO.hpp:40:12: error: db pragma 'object' is not associated with a declaration
/home/abauer/build/leosac/src/hardware/GPIO.hpp:40:19: error: db pragma 'callback' is not associated with a declaration
/home/abauer/build/leosac/src/hardware/GPIO.hpp:40:49: error: db pragma 'table' is not associated with a declaration
/home/abauer/build/leosac/src/hardware/GPIO.hpp:40:49: error: db pragma 'table' is not associated with a declaration
/home/abauer/build/leosac/src/hardware/RFIDReader.hpp:41:12: error: db pragma 'object' is not associated with a declaration
/home/abauer/build/leosac/src/hardware/RFIDReader.hpp:41:19: error: db pragma 'callback' is not associated with a declaration
/home/abauer/build/leosac/src/hardware/RFIDReader.hpp:41:49: error: db pragma 'table' is not associated with a declaration
/home/abauer/build/leosac/src/hardware/RFIDReader.hpp:41:49: error: db pragma 'table' is not associated with a declaration
/home/abauer/build/leosac/src/hardware/LED.hpp:35:12: error: db pragma 'object' is not associated with a declaration
[... cut ...]
/home/abauer/build/leosac/src/hardware/Buzzer.hpp:41:49: error: db pragma 'table' is not associated with a declaration
/home/abauer/build/leosac/src/hardware/Buzzer.hpp:41:49: error: db pragma 'table' is not associated with a declaration
src/db/CMakeFiles/leosac_db.dir/build.make:446: recipe for target 'src/db/core-schema-sqlite.cxx' failed
make[2]: *** [src/db/core-schema-sqlite.cxx] Error 1
CMakeFiles/Makefile2:284: recipe for target 'src/db/CMakeFiles/leosac_db.dir/all' failed
make[1]: *** [src/db/CMakeFiles/leosac_db.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

```

xaqq, 2018-01-22:
Hello,

I was able to reproduce on a fresh Strech install.
It seems the issue you mention is not resolved or has happened again.

I tried to build the ODB compiler myself using gcc-6 provided by raspbian, but it failed.

knnniggett, 2018-01-22:
Thanks for the response. Odb failed to build on my end using gcc 6 as well. I forgot what the exact error was. I seem to recall odb was looking for some function call not (or perhaps no longer) in gcc 6.

Trying to build with gcc 4.9 produced a different error. Something about time or date... googling revealed the issue was the compiler... so gcc 5 it is then...

Last night I managed to get the development branch to build, using Debian Jessie after I installed gcc-5, g++-5, and gcc-5-plugin-dev from debian unstable, but I also had to rebuild all the odb and boost libraries as well, since they depend on the version of gcc. It was not a pretty sight, but it did succeed.

Today I'm trying to build on Debian Stretch w/ the same gcc-5 packages.

xaqq, 2018-01-25:
Hey,

Yes I confirm that I also had issue with stock gcc (4.9) and didn't manage to build.
I'm glad you managed to build natively on Jessie. It's a start :D

> Today I'm trying to build on Debian Stretch w/ the same gcc-5 packages.

Did it work? If you got `develop` to build on Stretch I'd be interested to see a mini guide so we can add it to the documentation as it would be really helpful to others.

knnniggett, 2018-01-26:
Right now I'm experiencing more failure than success as I go through a learning curve. 
I stopped working on Stretch after I got a redeclaration error during the build, which I did not receive when performing the same steps on Jessie. I know that description is not very specific. I plan to come back to that. 

In the meantime, I discovered the Docker scripts for this project. That is what I'm playing with right now. The documentation under /doc/guides describes a tool called `leosaccli` as the glue that ties all the Docker bits together. However, I cannot find `leosaccli` in the project. Is this another case of outdated documentation, or am I just not looking in the right place?

If I can work my way to the point where I can consistently build the develop branch, then I'd be glad to give back to this project. The (missing) cmake library checks and the documentation could definitely use some love. If you are looking for volunteers, that is something I can help with, once I get a better understanding of this project.

xaqq, 2018-01-26:
Yes, the docker script are still experimental I would say. I've pushed the updated documentation on gh-pages now to reflect what was in the repository.

At http://www.leosac.com/doc_output/develop/html/d2/de2/page_guide_docker_cc.html you can find a partial guide  for cross-compiling using the `leosaccli` tool. However documentation is incomplete.

The idea is:
    + Go to `python` directory, and install `leosaccli`. `pip install -e .` should work. You might wanna use a virtualenv to avoid polluting your global install.
    + Build the `cross_compile` container. (1)
    + Run it to perform cross compilation.

(1): To build the cross_compile container you need some "cross-compile-resources": that's basically the `tar` version of required libraries. See https://github.com/leosac/bin-resources. So you would need to clone https://github.com/leosac/bin-resources from leosac root directory I believe, then run `leosaccli dev docker build cross_compile`.

> If you are looking for volunteers, that is something I can help with, once I get a better understanding of this project.

Lots of things needs to be improved in this project, therefore contributions are always welcomed :D. Build system and doc being an important part of this.

xaqq, 2018-02-02:
Hello,

I am pinging everyone who lately had trouble with installing/building/running leosac: @knnniggett @sirtoggle @fusionfoto

I cross-compiled the current develop and made a (pre) release. https://github.com/leosac/leosac/releases/tag/v0.7.0

I built it from a fresh x64vm using the leosac-cli and docker. I made a few commits along the way to (hack)fix issues.

I would be great if you could try to run it and provide feedback !

knnniggett, 2018-02-04:
@xaqq Thank you for all the work you've put into this project. This project stands out with the number of features, support for industry standard wiegand readers, and ability to be expanded further through modules.

I have successfully built the 0.7.0 pre-release and have it running on a raspberry pi 2, following these instructions: https://leosac.github.io/leosac-doc/doc_output/develop/d2/de2/page_guide_docker_cc.html

I went ahead and created the container using `leosaccli dev docker build cross_compile` even though the instructions did not mention that. Is this step no longer required?

It looks like an old page fragment is present in the github.io site:
https://leosac.github.io/leosac-doc/doc_output/develop/d9/d24/md_doc_guides_guide_full_cc.html

Not sure why that page exists since there is no corresponding source file under the doc subfolder.

knnniggett, 2018-02-05:
UPDATE:
Good news. Last night I was able to track down why the odb package in the debian repos triggers these error messages. Even better, it appears this exact issue has already been reported to the odb developers and fixed. 

I've reported the issue to debian, and included a patch which fixes it:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=889664

This now makes it possible to, once again, build the latest leosac into a package w/o building anything from-source. There is still more work to be done to make this possible, but I did want to give the project developers a heads up that this is something I plan to work on. As a temporary measure, I'll build odb packages for amd64 & armhf and make them accessible.

xaqq, 2018-02-05:
> I went ahead and created the container using leosaccli dev docker build cross_compile even though the instructions did not mention that. Is this step no longer required?

This step is still required, might be a documentation / documentation generation issue. The whole cross-compile run through the this docker container so it is required to have it.

Regarding ODB packages issue, this is indeed good news ! Would be nice indeed to be able to build from the raspberry natively (even tho its slow).
Thanks for researching this issue :)

knnniggett, 2018-02-11:
Going to close this issue out since it falls into the category of "not our bug".

closed 2018-02-11 by knnniggett

knight-of-ni, 2018-07-22:
Good news! Debian patched the ODB package in their repo and the Debain bug report is now resolved:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=889664

I'm not sure how long this will take to trickle down to the Raspbian repos.