leosac/access-control #99: Building on RPi against 0.6.3

fusionfoto, 2018-01-16:
Hi, Trying to compile native on RPi. I'm sure this is just a versioning thing, but I'm using the 0.6.3 tag (based on earlier suggestions on here) and I get this error from it appears boost. I'm using boost 1.62. 

Any suggestions on versions to move around or tags to use? I have the ability to build on an Intel linux box, I'm just not clear on the cross-compile process for ARM.

Thanks in advance!

[ 68%] Generating CXX prefix source src/cotire/leosac_lib_CXX_prefix.cxx
[ 68%] Generating CXX prefix header src/cotire/leosac_lib_CXX_prefix.hxx
[ 68%] Building CXX precompiled header src/cotire/leosac_lib_CXX_prefix.hxx.gch
Scanning dependencies of target leosac_lib
[ 69%] Building CXX object src/CMakeFiles/leosac_lib.dir/core/kernel.cpp.o
In file included from /home/pi/leosac/src/core/kernel.cpp:21:0:
/usr/include/boost/property_tree/ptree_serialization.hpp: In function âvoid boost::property_tree::detail::load_children(Archive&, boost::property_tree::basic_ptree<K, D, C>&)â:
/usr/include/boost/property_tree/ptree_serialization.hpp:66:24: error: âlibrary_version_typeâ in namespace âbsaâ does not name a type
             const bsa::library_version_type library_version(
                        ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/property_tree/ptree_serialization.hpp:69:16: error: âlibrary_version_typeâ is not a member of âbsaâ
             if(bsa::library_version_type(3) < library_version){
                ^~~
/usr/include/boost/property_tree/ptree_serialization.hpp:69:47: error: âlibrary_versionâ was not declared in this scope
             if(bsa::library_version_type(3) < library_version){
                                               ^~~~~~~~~~~~~~~
src/CMakeFiles/leosac_lib.dir/build.make:76: recipe for target 'src/CMakeFiles/leosac_lib.dir/core/kernel.cpp.o' failed
make[2]: *** [src/CMakeFiles/leosac_lib.dir/core/kernel.cpp.o] Error 1
CMakeFiles/Makefile2:2480: recipe for target 'src/CMakeFiles/leosac_lib.dir/all' failed
make[1]: *** [src/CMakeFiles/leosac_lib.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

knnniggett, 2018-01-20:
I ran into the same build error last week, and I fixed it by installing a missing boost library. I'm trying to retrace my steps to remember which one. 

Looking at the boost header files, it looks like bsa::library_version_type is defined in /usr/include/boost/property_tree/ptree_serialization.hpp. On my system, that file is part of libboost-serialization1.62-dev. So install that package, or perhaps better install libboost1.62-dev.

Note that I obtained the boost 1.62 libraries by upgrading raspbian from Jessie to Stretch. It appears that the package maintainers changed the naming of the boost packages in Stretch to include the version of boost in the package name. So depending on how you got your boost 1.62 libraries, the package names you are looking for may or may not have the boost version in the name... e.g. libboost-serialization1.62-dev Vs. libboost-serialization-dev.

Let me know if that works. Once you get past this build error, you may run smack into the error I described in #98. If you don't get that error, then I'd be really interested in learning the difference between your build environment and mine. We both are building natively on rpi's.

fusionfoto, 2018-01-22:
Thanks for getting back to me... 

Unfortunately, I already have that library:

I got them from apt-get install -y libboost-dev or libboost-all-dev.

This is the whole list:

pi@raspberrypi:~ $ aptitude search libboost |grep "i A"
i A libboost-atomic-dev - atomic data types, operations, and memory ordering constraints (default version)
i A libboost-atomic1.62-dev - atomic data types, operations, and memory ordering constraints
i A libboost-atomic1.62.0 - atomic data types, operations, and memory ordering constraints
i A libboost-chrono-dev - C++ representation of time duration, time point, and clocks (default version)
i A libboost-chrono1.62-dev - C++ representation of time duration, time point, and clocks
i A libboost-chrono1.62.0 - C++ representation of time duration, time point, and clocks
i A libboost-context-dev - provides a sort of cooperative multitasking on a single thread (default version)
i A libboost-context1.62-dev - provides a sort of cooperative multitasking on a single thread
i A libboost-context1.62.0 - provides a sort of cooperative multitasking on a single thread
i A libboost-coroutine-dev - provides a sort of cooperative multitasking on a single thread (default version)
i A libboost-coroutine1.62-dev - provides a sort of cooperative multitasking on a single thread
i A libboost-coroutine1.62.0 - provides a sort of cooperative multitasking on a single thread
i A libboost-date-time-dev - set of date-time libraries based on generic programming concepts (default version)
i A libboost-date-time1.61.0 - set of date-time libraries based on generic programming concepts
i A libboost-date-time1.62-dev - set of date-time libraries based on generic programming concepts
i A libboost-date-time1.62.0 - set of date-time libraries based on generic programming concepts
i A libboost-dev - Boost C++ Libraries development files (default version)
i A libboost-exception-dev - library to help write exceptions and handlers (default version)
i A libboost-exception1.62-dev - library to help write exceptions and handlers
i A libboost-fiber-dev - cooperatively-scheduled micro-/userland-threads (default version)
i A libboost-fiber1.62-dev - cooperatively-scheduled micro-/userland-threads
i A libboost-fiber1.62.0 - cooperatively-scheduled micro-/userland-threads
i A libboost-filesystem-dev - filesystem operations (portable paths, iteration over directories, etc) in C++ (default version)
i A libboost-filesystem1.61.0 - filesystem operations (portable paths, iteration over directories, etc) in C++
i A libboost-filesystem1.62-dev - filesystem operations (portable paths, iteration over directories, etc) in C++
i A libboost-graph-dev - generic graph components and algorithms in C++ (default version)
i A libboost-graph-parallel-dev - generic graph components and algorithms in C++ (default version)
i A libboost-graph-parallel1.62-dev - generic graph components and algorithms in C++
i A libboost-graph-parallel1.62.0 - generic graph components and algorithms in C++
i A libboost-graph1.62-dev - generic graph components and algorithms in C++
i A libboost-graph1.62.0 - generic graph components and algorithms in C++
i A libboost-iostreams-dev - Boost.Iostreams Library development files (default version)
i A libboost-iostreams1.61.0 - Boost.Iostreams Library
i A libboost-iostreams1.62-dev - Boost.Iostreams Library development files
i A libboost-locale-dev - C++ facilities for localization (default version)
i A libboost-locale1.62-dev - C++ facilities for localization
i A libboost-locale1.62.0 - C++ facilities for localization
i A libboost-log-dev - C++ logging library (default version)
i A libboost-log1.62-dev - C++ logging library
i A libboost-log1.62.0 - C++ logging library
i A libboost-math-dev - Boost.Math Library development files (default version)
i A libboost-math1.62-dev - Boost.Math Library development files
i A libboost-math1.62.0 - Boost.Math Library
i A libboost-mpi-dev - C++ interface to the Message Passing Interface (MPI) (default version)
i A libboost-mpi-python-dev - C++ interface to the Message Passing Interface (MPI), Python Bindings (default version)
i A libboost-mpi-python1.62-dev - C++ interface to the Message Passing Interface (MPI), Python Bindings
i A libboost-mpi-python1.62.0 - C++ interface to the Message Passing Interface (MPI), Python Bindings
i A libboost-mpi1.62-dev - C++ interface to the Message Passing Interface (MPI)
i A libboost-mpi1.62.0 - C++ interface to the Message Passing Interface (MPI)
i A libboost-program-options-dev - program options library for C++ (default version)
i A libboost-program-options1.62-dev - program options library for C++
i A libboost-program-options1.62.0 - program options library for C++
i A libboost-python-dev - Boost.Python Library development files (default version)
i A libboost-python1.62-dev - Boost.Python Library development files
i A libboost-python1.62.0 - Boost.Python Library
i A libboost-random-dev - Boost Random Number Library (default version)
i A libboost-random1.62-dev - Boost Random Number Library
i A libboost-random1.62.0 - Boost Random Number Library
i A libboost-regex-dev - regular expression library for C++ (default version)
i A libboost-regex1.61.0 - regular expression library for C++
i A libboost-regex1.62-dev - regular expression library for C++
[... 3 lines cut ...]
i A libboost-serialization1.62.0 - serialization library for C++
i A libboost-signals-dev - managed signals and slots library for C++ (default version)
[... cut ...]
i A libboost-type-erasure-dev - C++ runtime polymorphism based on concepts (default version)
i A libboost-type-erasure1.62-dev - C++ runtime polymorphism based on concepts
i A libboost-type-erasure1.62.0 - C++ runtime polymorphism based on concepts
i A libboost-wave-dev - C99/C++ preprocessor library (default version)
i A libboost-wave1.62-dev - C99/C++ preprocessor library
i A libboost-wave1.62.0 - C99/C++ preprocessor library
i A libboost1.62-dev - Boost C++ Libraries development files
i A libboost1.62-tools-dev - Boost C++ Libraries development tools

xaqq, 2018-01-22:
Hello,

What version of raspbian are you running on the Pi? And what version of the Pi are you using ?
@knnniggett: @fusionfoto wont run into #98 as this is related to database code that did not exist on the 0.6.3 tag.

0.6.3 should be easier to build than current develop.

@fusionfoto Have you tried to the .deb package ? (assuming you just want to run leosac).

fusionfoto, 2018-01-23:
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.3 (stretch)
Release:        9.3
Codename:       stretch

I haven't tried to use the deb package yet.

xaqq, 2018-01-25:
Hello,

I noticed that in #100 @knnniggett  had issue with the debian package. However hes running on a raspi Zero and i never tried on this hardware.
Any chance you could give a try to the deb package and report back?

fusionfoto, 2018-01-30:
I will give it all a try and report back! thanks.

fusionfoto, 2018-01-30:
so at first glance, this is what I got... 

root@raspberrypi:/var/cache/apt/archives/partial# ls -al
total 24
drwx------ 2 _apt root  4096 Jan 23 16:25 .
drwxr-xr-x 3 root root 20480 Jan 23 16:25 ..
root@raspberrypi:/var/cache/apt/archives/partial# wget https://github.com/leosac                                                                                                                                                          /leosac/releases/download/0.6.3/leosac_0.6.3-1_armhf.deb
--2018-01-30 13:12:40--  https://github.com/leosac/leosac/releases/download/0.6.                                                                                                                                                          3/leosac_0.6.3-1_armhf.deb
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/212435                                                                                                                                                          83/05cc68c2-8e29-11e5-9bbf-efb30b3785fb?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-C                                                                                                                                                          redential=AKIAIWNJYAX4CSVEH53A%2F20180130%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-                                                                                                                                                          Date=20180130T181240Z&X-Amz-Expires=300&X-Amz-Signature=328b4df92cfc965f0f9c9e7b                                                                                                                                                          a02d0dd9274aa3b34c5f64055ed0dd4fc3f2d323&X-Amz-SignedHeaders=host&actor_id=0&res                                                                                                                                                          ponse-content-disposition=attachment%3B%20filename%3Dleosac_0.6.3-1_armhf.deb&re                                                                                                                                                          sponse-content-type=application%2Foctet-stream [following]
--2018-01-30 13:12:40--  https://github-production-release-asset-2e65be.s3.amazo                                                                                                                                                          naws.com/21243583/05cc68c2-8e29-11e5-9bbf-efb30b3785fb?X-Amz-Algorithm=AWS4-HMAC                                                                                                                                                          -SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180130%2Fus-east-1%2Fs3%2Faws4                                                                                                                                                          _request&X-Amz-Date=20180130T181240Z&X-Amz-Expires=300&X-Amz-Signature=328b4df92                                                                                                                                                          cfc965f0f9c9e7ba02d0dd9274aa3b34c5f64055ed0dd4fc3f2d323&X-Amz-SignedHeaders=host                                                                                                                                                          &actor_id=0&response-content-disposition=attachment%3B%20filename%3Dleosac_0.6.3                                                                                                                                                          -1_armhf.deb&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-produc                                                                                                                                                          tion-release-asset-2e65be.s3.amazonaws.com)... 52.216.132.171
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-pr                                                                                                                                                          oduction-release-asset-2e65be.s3.amazonaws.com)|52.216.132.171|:443... connected                                                                                                                                                          .
HTTP request sent, awaiting response... 200 OK
Length: 1811628 (1.7M) [application/octet-stream]
Saving to: âleosac_0.6.3-1_armhf.debâ

leosac_0.6.3-1_armh 100%[===================>]   1.73M  3.41MB/s    in 0.5s

2018-01-30 13:12:41 (3.41 MB/s) - âleosac_0.6.3-1_armhf.debâ saved [1811628/1811                                                                                                                                                          628]

root@raspberrypi:/var/cache/apt/archives/partial# dpkg -i leosac_0.6.3-1_armhf.d                                                                                                                                                          eb
Selecting previously unselected package leosac.
(Reading database ... 149809 files and directories currently installed.)
Preparing to unpack leosac_0.6.3-1_armhf.deb ...
[... cut ...]
  Package libboost-serialization1.55.0 is not installed.
 leosac depends on libboost-serialization1.49.0; however:
  Package libboost-serialization1.49.0 is not installed.

dpkg: error processing package leosac (--install):
 dependency problems - leaving unconfigured
Processing triggers for systemd (232-25+deb9u1) ...
Errors were encountered while processing:
 leosac
root@raspberrypi:/var/cache/apt/archives/partial#

So its asking for two conflicting/old libraries. I will attempt to resolve that and report back.

knnniggett, 2018-01-30:
Yeah, the deb package on the releases page won't work with Debian Stretch because is has dependencies only found in Debian Jessie.

I patched up the 0.6.3 release branch in my own fork to build on Debian stretch. The following steps were successful in an amd64 vm, and it is building right now on my pi zero.  

Try this:
```
sudo apt-get install --no-install-recommends -y \
git wget build-essential cmake libtclap-dev devscripts fakeroot debhelper \
libsodium-dev libpython2.7-dev libunwind-dev libcurl4-openssl-dev \
libboost-serialization-dev libboost-regex-dev asciidoc-base

wget https://raw.githubusercontent.com/knnniggett/leosac/release-0.6.3/deb.sh
chmod +x deb.sh

./deb.sh -u https://github.com/knnniggett/leosac -b release-0.6.3
```

fusionfoto, 2018-01-30:
So building the package threw a bunch of warnings, but I think you knew that with the comments.

The debian package built, and I installed that and it seems ok. 

Now I need to figure out the next part... which is either connecting directly to the Weigand reader or the piface... Thanks so much for your help so far!

knnniggett, 2018-01-30:
Glad to hear. Hope the rest goes well.  Ultimately I'd like to take what we learn hear to update the project pending approval from the developers of course. I'm glad to help any way I can.

You beat me by the way.... my pi zero is at 78%. It's like I'm back to working on an IBM XT.

sirtoggle, 2018-01-31:
mine seems to end at. 

 dpkg-source --before-build leosac_0.6.3
dpkg-buildpackage: info: host architecture armhf
dpkg-checkbuilddeps: error: Unmet build dependencies: libsodium-dev libunwind-dev
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
debuild: fatal error at line 1116:
dpkg-buildpackage -rfakeroot -us -uc failed

knnniggett, 2018-01-31:
@sirtoggle sounds like you skipped the first step. The first step installs the packages it is complaining about. Just copy and paste the entire block into the command line:
```
sudo apt-get install --no-install-recommends -y \
git wget build-essential cmake libtclap-dev devscripts fakeroot debhelper \
libsodium-dev libpython2.7-dev libunwind-dev libcurl4-openssl-dev \
libboost-serialization-dev libboost-regex-dev asciidoc-base
```

sirtoggle, 2018-01-31:
looks like i did miss a line, ewps

pi@raspberrypi:~ $ sudo apt-get install --no-install-recommends -y \
> git wget build-essential cmake libtclap-dev devscripts fakeroot debhelper \
> libboost-serialization-dev libboost-regex-dev asciidoc-base
Reading package lists... Done

knnniggett, 2018-01-31:
Everything "should" just work. Let me know how it went...

sirtoggle, 2018-02-01:
My ssh connection timed out last nite, it had been at 75% at that time. and right now there is no deb file in a tmp folder.. So I am going to run it again but I think tomorrow so I can monitor it as it wouldn't be done by the time I go to sleep tonight.

fusionfoto, 2018-02-01:
Is there a set of minimum example scripts or config files I can use to set up leosacd? When I run the one I built, the daemon dies silently and I'm not really even sure where the docs are to start the initial config.

thanks

sirtoggle, 2018-02-01:
https://github.com/leosac/leosac/wiki/Configuration

knnniggett, 2018-02-02:
Oh, I had better mention that the build consumes a lot of memory.

On the pi zero, or any other pi with 512mb ram, you will certainly need to increase the swap space like so: https://www.bitpi.co/2015/02/11/how-to-change-raspberry-pis-swapfile-size-on-rasbian/

If your pi has 1GB ram, you might be able to get away with simply setting gpu_mem 16 in your pi config file, or just increase your swap using the instructions above.

sirtoggle, 2018-02-02:
Awesome, thank you for the info. Just updated the swap and started again.

sirtoggle, 2018-02-02:
So it was at 61% last I was connected to the Pi, Went to bed. and this morning no tmp folder and no deb file. Guess I will try again today and try to keep the connection between my laptop and the pi up and running
or its just too much for it to compile on the pi and I will need to do a box on my pc to the SD card then boot the Pi with that?

knnniggett, 2018-02-02:
I'm not sure if you meant to imply you are disconnecting your ssh session, or maybe I just have not had enough coffee yet this morning .....but if you really are disconnecting, logging out, shutting down your laptop, etc before the build is complete, the child processes (i.e. the deb.sh script) will be killed off the moment your parent process (e.g. your remote shell) is stopped. 

I didn't time it, but it took my slow pi zero w something like 3-4 hours to complete. If you can't leave an ssh session going, undisturbed, for that length of time then you should consider using something like the "at" tool to schedule the build for you (in which case you can log out then come back later), or perhaps it is simpler to just run the script direct from the local console.

I just did something like this from the command last night for something else I was working on: 
`echo /your/path/to/deb.sh | at 19:30`

or if you want to run it 1 minute from the moment you enter the command:
`echo /your/path/to/deb.sh | at now + 1 min`

Come back later and it will be done.  It should email you the output of the script, assuming you have local mail setup on the pi.

sirtoggle, 2018-02-02:
Oh I didn't realized that ending the ssh ended everything the pi was doing.. Again sorry, I am new to linux and the pi.

sirtoggle, 2018-02-02:
it finished, 5 hours, pi b+

sirtoggle, 2018-02-02:
Okay, one last dumb question now. is where did it install too. there isn't a /leosac/ nor do I see it in /usr/local/ folder w/ auth.conf and other files to edit..

knnniggett, 2018-02-02:
Package installations have a prefix of `/usr` rather than `/usr/local`.

Unlike Windows, the files for any particular package are spread around the file system.

The leosac binary itself is in `/usr/bin`, but you normally don't need to know that. You start it via the service file rather than call the binary directly: `sudo service start leosacd`.

Before you do that however, you need to build your kernel.xml file. That file should be placed under `/etc/leosac.d/`.

There is various documentation around here that shows how to create the kernel.xml. I followed this one: 
http://www.leosac.com/doc_output/develop/html/d5/d97/page_guide_rpi_piface_wiegand.html#install_guide_basic_cfg

knnniggett, 2018-02-02:
So random thought. If you are curious and want to see where all the files live, which belong to the package named leosac, do this: `dpkg -L leosac`

fusionfoto, 2018-02-06:
Is it just me or are those leosac.com urls broken? Google and direct click both seem to have that problem.

fusionfoto, 2018-02-06:
https://leosac.github.io/leosac-doc/doc_output/develop/d5/d97/page_guide_rpi_piface_wiegand.html seems to work for me...

knnniggett, 2018-02-06:
Yeah, you're right.
It looks like the documentation was recently moved from www.leosac.com to leosac.github.io

xaqq, 2018-02-07:
Yes I have created a leosac-doc repository to push the generated documentation. Indeed the resulting doc was about 100MB so this bloated the main repo for nothing.

I'm curious about what is broken tho, because the link @fusionfoto provided seems to work ok on my side and the "Documentation" link on www.leosac.com points to http://leosac.github.io/leosac-doc/doc_output/develop/ which is correct.

Maxhy, 2021-05-03:
Original issue is resolved, closing.

closed 2021-05-03 by Maxhy