leosac/access-control #101: Dependency uninstallable

sirtoggle, 2018-01-28:
This package is uninstallable
Dependency is not satisfiable: libboost-regex1.55.0

I have searched the internet for a way to install this lib. and there were a few packages that said it was in there, installed them and still get this error. I found one that said this had to be compiled but when going through the instructions I got stuck with a command qmake, and it's not a found command. 
Any ideas?

Liryna, 2018-01-28:
Hi @sirtoggle ,

https://packages.debian.org/fr/jessie/libboost-regex-dev

sirtoggle, 2018-01-28:
**@Liryna, Thank you, but having more problems still.**

_pi@raspberrypi:~ $ sudo gdebi libboost-regex1.55.0_1.55.0+dfsg-3_armhf.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Reading state information... Done
This package is uninstallable
Dependency is not satisfiable: libicu52 (>= 52~m1-1~)_

**So I downloaded libicu52, and then.**

_pi@raspberrypi:~ $ sudo gdebi libicu52_52.1-8+deb8u6_armhf.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Reading state information... Done
This package is uninstallable
Failed to satisfy all dependencies (broken cache)_

I am new to this.

Liryna, 2018-01-28:
Which version of debian are you running ?
The most easy would be to run debian jessie with 
`deb http://ftp.de.debian.org/debian jessie main`
in the repository list and install the package directly instead of download one by one.

sirtoggle, 2018-01-28:
I am running raspbian, 

pi@raspberrypi:~ $ uname -r
4.9.59+
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.1 (stretch)
Release:        9.1
Codename:       stretch

Liryna, 2018-01-28:
Ok so you can get the 1.62
https://packages.debian.org/stretch/libboost-regex-dev

sirtoggle, 2018-01-28:
I had 1.62 installed. But leosac is looking for 1.55

knnniggett, 2018-01-28:
The deb 0.6.3 deb package on the leosac releases page has a strict package dependency on boost regex 1.55. This version of boost is the default for debian/raspbian Jessie, while Stretch uses boost 1.62. Unfortunately, as the OP just found out, one goes down the dependency rabbit hole (been there) trying to manually install older packages, which might cause other problems with the operating system.

A couple options the op can do are to go back to Debian (Raspbian) Jessie or build Leosac. There is a deb.sh script in the root folder of the project which should let you build your own package, but I have not tried it with 0.6.3 release. Deb.sh does not work currently with latest develop.

On a side note, I've made a lot of progress on a shell script which builds the Leosac develop branch on either Jessie or Stretch. It builds perfectly on amd64, and today I've begun testing directly on a pi zero. As you can imagine, running directly on the pi is quite slow so testing takes a long time. If anyone is interested, I will post the script shortly. I need to do some additional testing however to verify the script  won't irreparably break anything.

sirtoggle, 2018-01-28:
I tried following the build leosac directions on the site, and also became stuck.

knnniggett, 2018-01-28:
Yeah, the instructions are out of date. I just went through this myself. 
Try running deb.sh after git cloning the project. If you have not built deb packages on the host machine before, the script will fail due to missing packages, but it will tell you which package it looking for.  Install that package and run the script again. This worked for me for leosac 0.4.1 (at the time I had not realized 0.6.3 was the newest release).

sirtoggle, 2018-01-28:
trying it now..

thanks again for the help from both of you.

sirtoggle, 2018-01-28:
quick update, seems to be building everything right now, 7% and growing.

sirtoggle, 2018-01-28:
./sigh

make[3]: Leaving directory '/tmp/tmp.v4rI9vvdqV/leosac_0.4.1/obj-arm-linux-gnueabihf'
[ 59%] Built target libzmq-static
make[2]: Leaving directory '/tmp/tmp.v4rI9vvdqV/leosac_0.4.1/obj-arm-linux-gnueabihf'
Makefile:130: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/tmp/tmp.v4rI9vvdqV/leosac_0.4.1/obj-arm-linux-gnueabihf'
dh_auto_build: make -j3 returned exit code 2
debian/rules:6: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
debuild: fatal error at line 1116:
dpkg-buildpackage -rfakeroot -us -uc failed

knnniggett, 2018-01-28:
That portion of the output is coming from the debian build scripts. It just says the build failed without telling you why.

To get a better idea of why it failed, you need to scroll up, possibly quite a bit, to see what exactly caused the error. It can be hard to spot in all the output. You are looking for the compiler or linker giving a long message that starts with "error:". Warnings can be ignored.

sirtoggle, 2018-01-28:
here is what i found . unless -werror counts. Some libs missing, but I can't find some to install. like windows.h

/usr/include/boost/property_tree/ptree_serialization.hpp:66:24: error: ‘library_version_type’ in namespace ‘bsa’ does                                                                                                                         not name a type
             const bsa::library_version_type library_version(
                        ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/property_tree/ptree_serialization.hpp:69:16: error: ‘library_version_type’ is not a member of ‘bsa                                                                                                                        ’
             if(bsa::library_version_type(3) < library_version){
                ^~~
/usr/include/boost/property_tree/ptree_serialization.hpp:69:47: error: ‘library_version’ was not declared in this sco                                                                                                                        pe
             if(bsa::library_version_type(3) < library_version){
                                               ^~~~~~~~~~~~~~~
-----------------------------------------------------------------------------------------------------------------

src/modules/rpleth/CMakeFiles/rpleth.dir/build.make:89: recipe for target 'src/modules/rpleth/CMakeFiles/rpleth.dir/R                                                                                                                        plethModule.cpp.o' failed
make[3]: *** [src/modules/rpleth/CMakeFiles/rpleth.dir/RplethModule.cpp.o] Error 1
make[3]: Leaving directory '/tmp/tmp.SviBcMPTbx/leosac_0.4.1/obj-arm-linux-gnueabihf'
CMakeFiles/Makefile2:621: recipe for target 'src/modules/rpleth/CMakeFiles/rpleth.dir/all' failed
make[2]: *** [src/modules/rpleth/CMakeFiles/rpleth.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....

----------------------------------------------------------------------------------------------------------------

sirtoggle, 2018-01-28:
also it couldn't find kqueue, so I installed that package

knnniggett, 2018-01-29:
Ah, so you have arrived at #99

knnniggett, 2018-01-30:
@sirtoggle We are having some success building the 0.6.3 release with the steps outlined here:
https://github.com/leosac/leosac/issues/99#issuecomment-361730413

Give it a shot.

sirtoggle, 2018-01-31:
I will head on over there, thanks for the heads up.

EricPaquet, 2018-07-18:
I'll see that too

EricPaquet, 2018-07-18:
I'll look at <b> that </b>

EricPaquet, 2018-07-18:
I'll see %252E

Maxhy, 2021-05-03:
Fixed with #118, thanks again @knight-of-ni :)

closed 2021-05-03 by Maxhy