leosac/access-control #111: Logging possibilities

raumfisch, 2018-02-11:
Hello, 

i'm looking for a possibility to log failed and successful authentication attempts and save them to file or database backend. 
Every notification module i found so far is working with the SIMPLE_WIEGAND strategy and sending only the card_id of scanned cards. What i need is a complete log with user, time and used key of a WiegandCardAndPin credential, or the card_id and time of a failed login attempt.

On the documentation page is written:

> This optional module can be used to filter events coming from the core program. Everything Leosac does is sent to logger modules, that way its simple to implement custom notification systems, like automatic e-mail alerts for users that were blacklisted. The default implementation simply prints out what is happening on the system in real time.

I'm not able find and documentation about these logging possibilities, did i miss something?
Can you tell me a way to accomplish what i'm looking for?

Thank you,

xaqq, 2018-02-14:
Hey,

I am assuming the documentation you quote is from the wiki. Sadly the wiki is completely outdated, to the point that it should not be used anymore.

Indeed the WSNotifier module only support SIMPLE_WIEGAND.
Sadly what you want to do is not possible yet because it has not been needed until now. It should be fairly easy to add support for other type of credential to WSNotifier. However, WSNotifier do not watch authentication result, simply attempts; it cannot map an attempt to a result.

I assume you have an Authentication module enabled. Does it not print the credential to the log when it receives it? I think right now your best bet is to read from leosac log.

If this does not work, I can probably find the time to implement support for other type of credential into WSNotifier.

raumfisch, 2018-03-13:
Hey,  
in the meantime i developed my own simple logging using the WSNotifier module und my own database. I'm also not looking for successful attempts, but 

If anyone is interested, i published my code here: https://github.com/raumfisch/leosac-keytool
It's a nodejs based keytool with the possibility wo manage users, keys and pin codes. The feature to manage schedules and groups will follow. 
The tool is using it's own mongo database and is writing the auth.xml file to the disk.
Documentation will follow. 

I recently ran into a different issue: 
I'm looking or a way to create the hex-codes written to the auth.xml files. The WSNotifier module is only giving me decimal numbers, for example, 61685 will result in the hex-code "ca:78:7a:c0"

Could you tell me where i can look for the algorithm to convert those numbers? I searched the internet without results yet. Only found some online Wiegand-calculators, but they are giving me different codes than those used by leosac.

Many thanks!

Maxhy, 2022-04-22:
Today we have Audit events but this is currently only used when managing the entities through the API. We should add new event types for access granted/refused to properly store and track such events. At the same time we could publish a new standardized ZMQ message for new modules to forward such event, like WSNotifier.