leosac/access-control #119: RPi+Piface+Wiegand reader tutorial is missing diagrams

systeemgod, 2018-04-05:
Hi,

I want to build a setup as described in the tutorial, however the wiring diagrams seem to be missing (e.g. the 'wiring everything' section).  Do you have these diagrams or images available somewhere?

Nico

brianredbeard, 2018-06-06:
Hope this helps.

The idea is to create a [voltage divider](https://elinux.org/RPi_GPIO_Interface_Circuits#Voltage_divider) which brings the voltage down from +5v to something roughly in the range of +3.3v.

[image]

To double check that everything is wired correctly this C program is very useful: [`wiegand.c`](https://gist.github.com/hsiboy/9598741).  Just make sure you referencing the [wiringPi](http://wiringpi.com) pins, not [BCM pins](https://pinout.xyz).

systeemgod, 2018-06-06:
Thanks for the info!

Nico

closed 2021-05-03 by Maxhy

dplicato, 2022-04-02:
Hi,
I found that the diagram were missing at leosac 0.7.0 doc, but they are present at 0.8.0 doc [here](https://leosac.github.io/leosac-doc/doc_output/develop/d5/d97/page_guide_rpi_piface_wiegand.html).

About the hardware it says that are only needed:
1. A Raspberry Pi
2. A Piface Digital board

In the "Wiring everything" section there is no reference to the need of a voltage divider and it seems that is possible to directly connect the 5v wiegand data wires to the piface digital inputs.

Is this correct?

Thank you.

Maxhy, 2022-04-02:
Yes that's correct. When using a Piface board/hat you don't need a voltage divider / level shifter. Piface inputs should support up to 5v. Piface Digital manufacture several products so it may be safe to check their documentation first depending the model you are using.

Uzzi, 2022-06-15:
Hi Could you help me about https://wiki.geekworm.com/Raspberry_Pi_DIDO_Board and https://ae01.alicdn.com/kf/H1af968b7b7ec4bce83c5e4152c6be2b2z.jpg connections schema diagram?
Thank you