leosac/access-control #120: Compile Error on Raspberry Pi

pwallner, 2018-04-18:
Hello,

when I compile your source, I receive this error:
```
CMake Error at /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find ODB (missing: ODB_EXECUTABLE ODB_LIBODB_FOUND pgsql sqlite
  boost)
Call Stack (most recent call first):
  /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindODB.cmake:108 (find_package_handle_standard_args)
  CMakeLists.txt:56 (find_package)
```

How can I solve this problem?

Thank you,
Patrick

knnniggett, 2018-04-18:
I guess you could call me an interested third party. I'm not on development team, but they have not been responsive lately so I thought I'd respond. 

I've been able to build leosac successfully, and I've got an open issue describing how one can use a docker container to easily build a leosac deb package.

The error message you received is saying you are missing the ODB executable, which is one of many dependencies.  If you want to build from source, here is the documentation to accomplish that:
https://leosac.github.io/leosac-doc/doc_output/develop/d6/d12/page_guide_install_from_source.html

Alternatively, I've got a docker container set up, which will spit out an arm deb, for you:
See #116 

It will build an arm deb package for you, either building natively or on an x86_64 host.

pwallner, 2018-04-24:
Caused by the reason of the missing odb, ich build my own odb's (2.4.0) with boost 1.67.

The first command now works:
`cmake -DCMAKE_BUILD_TYPE=Release ..`
but with
`make`

I get this error at 7%:
```
[  7%] Generating ../../odb_gen/tools/SingleTimeFrame_odb_pgsql.cpp, ../../odb_g                                                                    en/tools/SingleTimeFrame_odb_sqlite.cpp, ../../odb_gen/tools/SingleTimeFrame_odb                                                                    .cpp
In file included from /home/pi/Downloads/mcpat-odb/leosac/src/tools/Schedule.hpp                                                                    :25:0,
                 from /home/pi/Downloads/mcpat-odb/leosac/src/tools/ScheduleMapp                                                                    ing.hpp:178,
                 from /home/pi/Downloads/mcpat-odb/leosac/src/core/auth/User.hpp                                                                    :208,
                 from /home/pi/Downloads/mcpat-odb/leosac/src/core/audit/AuditEn                                                                    try.hpp:180,
                 from <odb-epilogue-1>:1:
/home/pi/Downloads/mcpat-odb/leosac/src/tools/SingleTimeFrame.hpp:38:8: error: r                                                                    edefinition of ‘struct Leosac::Tools::SingleTimeFrame’
 struct SingleTimeFrame
        ^
/home/pi/Downloads/mcpat-odb/leosac/src/tools/SingleTimeFrame.hpp:38:8: error: p                                                                    revious definition of ‘struct Leosac::Tools::SingleTimeFrame’
 struct SingleTimeFrame
        ^
src/db/CMakeFiles/leosac_db.dir/build.make:71: recipe for target 'odb_gen/tools/                                                                    SingleTimeFrame_odb_pgsql.cpp' failed
make[2]: *** [odb_gen/tools/SingleTimeFrame_odb_pgsql.cpp] Error 1
CMakeFiles/Makefile2:284: recipe for target 'src/db/CMakeFiles/leosac_db.dir/all                                                                    ' failed
make[1]: *** [src/db/CMakeFiles/leosac_db.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

pwallner, 2018-05-10:
Ok, I found out
- compiling with gcc < 5.x and "hacked" source not possible (see above answer)
- compiling with gcc 5.3.0 -> build erros
- compiling with gcc 7.3.0 -> on going (9%)

So here are the necessary files for trying to build leosac:
[gcc-7.3.0_armhf.deb](https://www.mcpat.com/.leosac/7.3.0/gcc-7.3.0_armhf.deb)
[libboost-1.62-dev_armhf.deb](https://www.mcpat.com/.leosac/7.3.0/libboost-1.62-dev_armhf.deb)
[libboost-date-time1.62.0_armhf.deb](https://www.mcpat.com/.leosac/7.3.0/libboost-date-time1.62.0_armhf.deb)
[libboost-date-time1.62-dev_armhf.deb](https://www.mcpat.com/.leosac/7.3.0/libboost-date-time1.62-dev_armhf.deb)
[libboost-filesystem1.62.0_armhf.deb](https://www.mcpat.com/.leosac/7.3.0/libboost-filesystem1.62.0_armhf.deb)
[libboost-filesystem1.62-dev_armhf.deb](https://www.mcpat.com/.leosac/7.3.0/libboost-filesystem1.62-dev_armhf.deb)
[regex](https://www.mcpat.com/.leosac/7.3.0/libboost-regex1.62.0_armhf.deb)
[regex-dev](https://www.mcpat.com/.leosac/7.3.0/libboost-regex1.62-dev_armhf.deb)
[serialization](https://www.mcpat.com/.leosac/7.3.0/libboost-serialization1.62.0_armhf.deb)
[serialization-dev](https://www.mcpat.com/.leosac/7.3.0/libboost-serialization1.62-dev_armhf.deb)
[system](https://www.mcpat.com/.leosac/7.3.0/libboost-system1.62.0_armhf.deb)
[system-dev](https://www.mcpat.com/.leosac/7.3.0/libboost-system1.62-dev_armhf.deb)
[libcutl](https://www.mcpat.com/.leosac/7.3.0/libcutl-1.10.0_armhf.deb)
[libcutl-dev](https://www.mcpat.com/.leosac/7.3.0/libcutl-1.10.0-dev_armhf.deb)
[libodb](https://www.mcpat.com/.leosac/7.3.0/libodb-2.4.0_armhf.deb)
[libodb-boost](https://www.mcpat.com/.leosac/7.3.0/libodb-boost-2.4.0_armhf.deb)
[libodb-boost-dev](https://www.mcpat.com/.leosac/7.3.0/libodb-boost-dev-2.4.0_armhf.deb)
[libodb-dev](https://www.mcpat.com/.leosac/7.3.0/libodb-dev-2.4.0_armhf.deb)
[libodb-mysql](https://www.mcpat.com/.leosac/7.3.0/libodb-mysql-2.4.0_armhf.deb)
[libodb-mysql-dev](https://www.mcpat.com/.leosac/7.3.0/libodb-mysql-dev-2.4.0_armhf.deb)
[libodb-pgsql](https://www.mcpat.com/.leosac/7.3.0/libodb-pgsql-2.4.0_armhf.deb)
[libodb-pgsql-dev](https://www.mcpat.com/.leosac/7.3.0/libodb-pgsql-dev-2.4.0_armhf.deb)
[libodb-sqlite](https://www.mcpat.com/.leosac/7.3.0/libodb-sqlite-2.4.0_armhf.deb)
[libodb-sqlite-dev](https://www.mcpat.com/.leosac/7.3.0/libodb-sqlite-dev-2.4.0_armhf.deb)
[odb](https://www.mcpat.com/.leosac/7.3.0/odb-2.4.0_armhf.deb)
[odb-2.4.0-6b-patches.tgz](https://www.mcpat.com/.leosac/7.3.0/odb-2.4.0-patches.tgz)

Next step:
- try to compile leosac to 100%
- docker from above, need a larger SD in my Rpi...

Regards
Pat

knnniggett, 2018-05-10:
Yes, the latest version of Leosac is known not build on Debian Jessie. The gcc 4.8.x compiler that ships with Jessie crashes. I thought I mentioned that in one of the other issues. There are other problems with Jessie too if I recall.  Come to think of it, I don't recall if I was ever successful building the latest master branch on Jessie. You need to build on Debian Stretch.

Docker is a much easier solution. However, if you go that route, you really ought to build on a fairly fast x86_64 machine. Follow the instructions shown on my Dockerhub page to enable arm emulation on the x86_64 host machine, and it will build an armf deb for you. It will take a long time, but it will finish.

If you build on a Pi, it will take even longer, and you almost certainly will need to increase the swap file to something like 2GB.  No matter how you build Leosac, the build consumes more than 1GB of RAM, which is more than the Pi has.

Good luck.

pwallner, 2018-05-10:
And now I hang on 38%

```
[ 38%] Building CXX object src/CMakeFiles/leosac_lib.dir/core/RemoteControl.cpp.o
In file included from /home/pi/Downloads/mcpat-odb/extern/leosac/deps/zmqpp/src/zmqpp/zmqpp.hpp:61:0,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/config/ConfigManager.hpp:29,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/RemoteControl.hpp:23,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/RemoteControl.cpp:20:
/home/pi/Downloads/mcpat-odb/extern/leosac/deps/zmqpp/src/zmqpp/compatibility.hpp:36:5: Warnung: »_WIN32« ist nicht definiert, wird zu 0 ausgewertet [-Wundef]
 #if _WIN32
     ^~~~~~
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/RemoteControl.cpp: In Elementfunktion »void Leosac::RemoteControl::handle_msg()«:
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/RemoteControl.cpp:113:20: Fehler: »class zmqpp::message« hat kein Element namens »get_property«
     bool ret = msg.get_property("User-Id", user_pubkey);
                    ^~~~~~~~~~~~
src/CMakeFiles/leosac_lib.dir/build.make:206: recipe for target 'src/CMakeFiles/leosac_lib.dir/core/RemoteControl.cpp.o' failed
make[2]: *** [src/CMakeFiles/leosac_lib.dir/core/RemoteControl.cpp.o] Error 1
CMakeFiles/Makefile2:190: recipe for target 'src/CMakeFiles/leosac_lib.dir/all' failed
make[1]: *** [src/CMakeFiles/leosac_lib.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

pwallner, 2018-05-10:
Thx, but after compiling libzmq now I'm at 39%

```
[ 38%] Building CXX object src/CMakeFiles/leosac_lib.dir/core/RemoteControl.cpp.o
In file included from /home/pi/Downloads/mcpat-odb/extern/leosac/deps/zmqpp/src/zmqpp/zmqpp.hpp:61:0,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/config/ConfigManager.hpp:29,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/RemoteControl.hpp:23,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/RemoteControl.cpp:20:
/home/pi/Downloads/mcpat-odb/extern/leosac/deps/zmqpp/src/zmqpp/compatibility.hpp:36:5: Warnung: »_WIN32« ist nicht definiert, wird zu 0 ausgewertet [-Wundef]
 #if _WIN32
     ^~~~~~
[ 39%] Building CXX object src/CMakeFiles/leosac_lib.dir/core/RemoteControlSecurity.cpp.o
[ 39%] Building CXX object src/CMakeFiles/leosac_lib.dir/core/module_manager.cpp.o
In file included from /home/pi/Downloads/mcpat-odb/extern/leosac/deps/zmqpp/src/zmqpp/zmqpp.hpp:61:0,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/config/ConfigManager.hpp:29,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/module_manager.hpp:24,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/module_manager.cpp:20:
/home/pi/Downloads/mcpat-odb/extern/leosac/deps/zmqpp/src/zmqpp/compatibility.hpp:36:5: Warnung: »_WIN32« ist nicht definiert, wird zu 0 ausgewertet [-Wundef]
 #if _WIN32
     ^~~~~~
[ 39%] Building CXX object src/CMakeFiles/leosac_lib.dir/core/MessageBus.cpp.o
In file included from /home/pi/Downloads/mcpat-odb/extern/leosac/deps/zmqpp/src/zmqpp/zmqpp.hpp:61:0,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/MessageBus.hpp:21,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/MessageBus.cpp:20:
/home/pi/Downloads/mcpat-odb/extern/leosac/deps/zmqpp/src/zmqpp/compatibility.hpp:36:5: Warnung: »_WIN32« ist nicht definiert, wird zu 0 ausgewertet [-Wundef]
 #if _WIN32
     ^~~~~~
[ 39%] Building CXX object src/CMakeFiles/leosac_lib.dir/core/Scheduler.cpp.o
In file included from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/GenericTask.hpp:23:0,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/Scheduler.hpp:23,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/Scheduler.cpp:20:
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp:110:10: Fehler: »function« in Namensraum »std« bezeichnet keinen Templatetyp
     std::function<void(void)> on_completion_;
          ^~~~~~~~
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp:111:10: Fehler: »function« in Namensraum »std« bezeichnet keinen Templatetyp
     std::function<void(void)> on_success_;
          ^~~~~~~~
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp:112:10: Fehler: »function« in Namensraum »std« bezeichnet keinen Templatetyp
     std::function<void(void)> on_failure_;
          ^~~~~~~~
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp: In Elementfunktion »void Leosac::Tasks::Task::set_on_completion(Callback)«:
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp:78:9: Fehler: »on_completion_« wurde in diesem Gültigkeitsbereich nicht definiert
         on_completion_ = [=]() { c(); };
         ^~~~~~~~~~~~~~
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp:78:9: Anmerkung: empfohlene Alternative: »set_on_completion«
         on_completion_ = [=]() { c(); };
         ^~~~~~~~~~~~~~
         set_on_completion
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp: In Elementfunktion »void Leosac::Tasks::Task::set_on_success(Callback)«:
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp:90:9: Fehler: »on_success_« wurde in diesem Gültigkeitsbereich nicht definiert
         on_success_ = [=]() { c(); };
         ^~~~~~~~~~~
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp:90:9: Anmerkung: empfohlene Alternative: »success_«
         on_success_ = [=]() { c(); };
         ^~~~~~~~~~~
         success_
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp: In Elementfunktion »void Leosac::Tasks::Task::set_on_failure(Callback)«:
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp:102:9: Fehler: »on_failure_« wurde in diesem Gültigkeitsbereich nicht definiert
         on_failure_ = [=]() { c(); };
         ^~~~~~~~~~~
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/Task.hpp:102:9: Anmerkung: empfohlene Alternative: »set_on_failure«
         on_failure_ = [=]() { c(); };
         ^~~~~~~~~~~
         set_on_failure
In file included from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/Scheduler.hpp:23:0,
                 from /home/pi/Downloads/mcpat-odb/extern/leosac/src/core/Scheduler.cpp:20:
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/GenericTask.hpp: Im globalen Gültigkeitsbereich:
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/GenericTask.hpp:37:28: Fehler: »function« in Namensraum »std« bezeichnet keinen Templatetyp
     GenericTask(const std::function<bool(void)> &fct);
                            ^~~~~~~~
/home/pi/Downloads/mcpat-odb/extern/leosac/src/core/tasks/GenericTask.hpp:37:36: Fehler: expected »,« or »...« before »<« token
     GenericTask(const std::function<bool(void)> &fct);
                                    ^
[... cut ...]
         std::thread(std::bind(&Task::run, t)).detach();
                          ^~~~
                          rend
src/CMakeFiles/leosac_lib.dir/build.make:302: recipe for target 'src/CMakeFiles/leosac_lib.dir/core/Scheduler.cpp.o' failed
make[2]: *** [src/CMakeFiles/leosac_lib.dir/core/Scheduler.cpp.o] Error 1
CMakeFiles/Makefile2:190: recipe for target 'src/CMakeFiles/leosac_lib.dir/all' failed
make[1]: *** [src/CMakeFiles/leosac_lib.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

knnniggett, 2018-05-11:
Adding include <functional> to get the build to progress makes me think there is a deeper problem going on here. I didn't have to modify the source code to build leosac.

In any case, regarding the new problem, look at this block of code:
https://github.com/leosac/leosac/blob/develop/src/tools/Stacktrace.cpp#L24

Try setting LEOSAC_STACKTRACE_DISABLED=1 when you call cmake e.g.
`cmake -DLEOSAC_STACKTRACE_DISABLED=1 -DCMAKE_BUILD_TYPE=Release ..`

When I was testing the package building scripts under the pkg/debian folder, I set stacktrace to disabled as the default. Maybe I should submit a pr to make this the default for the from-source instructions as well.

knnniggett, 2018-05-11:
It's still trying to link to backtrace. The likely cause is the cmake cache. When you called cmake a second , third time, etc. you should always first delete the cmakecache file it created when it first ran.

pwallner, 2018-05-11:
Thank you! I will try it, in the meantime I found (and compiled) libbacktrace from here [ianlancetaylor/libbacktrace](https://github.com/ianlancetaylor/libbacktrace), but I will follow your hint and delete the cmakecache and restart from beginning.

pwallner, 2018-05-11:
I did a clean new cmake and this is in the `CMakeCache.txt` (you can see `backtrace`, is this ok?):
```
//Dependencies for the target
leosac_lib_LIB_DEPENDS:STATIC=general;dl;general;pthread;general;zmqpp;general;/usr/lib/libboost_date_time.so;general;/usr/lib/libboost_system.so;general;/usr/lib/libboost_serialization.so;general;/usr/lib/libboost_regex.so;general;/usr/lib/libboost_filesystem.so;general;/usr/local/lib/libodb.so;general;/usr/local/lib/libodb-pgsql.so;general;/usr/local/lib/libodb-sqlite.so;general;/usr/local/lib/libodb-boost.so;general;backtrace;general;scrypt;general;leosac_db;
```

knnniggett, 2018-05-11:
Maybe this is a difference kind of problem than I thought.

Do you have the matching libgcc-X-dev package installed for the compiler you are using?
For example Stretch ships with gcc 6, so the corresponding package, which contains libbaktrace.a, is libgcc-6-dev.

pwallner, 2018-05-11:
I compiled gcc 7.3.0 by myself, so I don't have a libgcc-6-dev package..
But I compiled the libbaktrace by myself (see answer above) and there is a libbaktrace.a compiled. I installed this libbacktrace package which I compiled and so I think leosac can finish, now I'm at 90% and on going.

pwallner, 2018-05-15:
I found this in ‘src/CMakeLists.txt‘
‘‘‘
…
target_link_libraries(${LEOSAC_BIN} ${LEOSAC_LIB} backtrace)
target_link_libraries(${LEOSAC_LIB} dl pthread zmqpp ${Boost_LIBRARIES} ${ODB_LIBRARIES} backtrace scrypt leosac_db)
…
‘‘‘
And
‘‘‘
…
if (${LEOSAC_STACKTRACE_DISABLED})
    # Disable stacktrace generation.
    target_compile_definitions(${LEOSAC_LIB} PUBLIC -DLEOSAC_STACKTRACE_DISABLED=1)
endif()
…
‘‘‘
Now I try this hack:
‘‘‘
…
if (${LEOSAC_STACKTRACE_DISABLED})
    target_link_libraries(${LEOSAC_BIN} ${LEOSAC_LIB})
    target_link_libraries(${LEOSAC_LIB} dl pthread zmqpp ${Boost_LIBRARIES} ${ODB_LIBRARIES} scrypt leosac_db)
else()
    target_link_libraries(${LEOSAC_BIN} ${LEOSAC_LIB} backtrace)
    target_link_libraries(${LEOSAC_LIB} dl pthread zmqpp ${Boost_LIBRARIES} ${ODB_LIBRARIES} backtrace scrypt leosac_db)
endif()
…
‘‘‘

Maxhy, 2021-05-03:
Can probably be closed since... but still asking @pwallner?

pwallner, 2021-05-04:
> Can probably be closed since... but still asking @pwallner?

Yes we can close it

closed 2021-05-04 by Maxhy

Maxhy, 2022-03-06:
@pwallner it was still a valid issue after further investigation. Build should have been properly fixed for Debian Bullseye now.