leosac/access-control #126: Start error

crusaderpi, 2019-10-16:
Hi, I'm getting the following error when trying to start leosac:

Running Leosac version 0.7.0-afde0af387b980cbfc82fa36fa15eba9f1ac6499
Logger not set-up yet ! Will display log message as is.
Creating Leosac Kernel...
[2019-10-16 15:53:32.486] [console] [info] [5510] Network configuration disabled. Will not touch system setting.
[2019-10-16 15:53:32.487] [console] [debug] [5510] Adding {/usr/lib/leosac} in library path
[2019-10-16 15:53:32.487] [console] [debug] [5510] Adding {/usr/lib/leosac/auth} in library path
[2019-10-16 15:53:32.487] [console] [info] [5510] Attempting to load module nammed SYSFS-GPIO (shared lib file = libsysfsgpio.so)
[2019-10-16 15:53:32.488] [console] [info] [5510] Loading library at: /usr/lib/leosac/libsysfsgpio.so
[2019-10-16 15:53:32.489] [console] [debug] [5510] library file loaded (not init yet)
[2019-10-16 15:53:32.490] [console] [info] [5510] Attempting to load module nammed WIEGAND_READER (shared lib file = libwiegand.so)
[2019-10-16 15:53:32.490] [console] [info] [5510] Loading library at: /usr/lib/leosac/libwiegand.so
[2019-10-16 15:53:32.504] [console] [error] [5510] FAILURE, full path was:{/usr/lib/leosac/libwiegand.so}: DynLib::dlopen(): /usr/bin/../lib/leosac/libwebsock-api.so: undefined symbol: _ZN3odb10query_base6appendERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
[2019-10-16 15:53:32.505] [console] [info] [5510] Not stopping module SYSFS-GPIO as it doesn't seem to run.
Exception propagated to main(). Will now exit.
exception: Cannot load modules. Search path was:
         -> /usr/lib/leosac
         -> /usr/lib/leosac/auth
free(): invalid pointer
Aborted

knight-of-ni, 2019-10-16:
> [2019-10-16 15:53:32.504] [console] [error] [5510] FAILURE, full path was:{/usr/lib/leosac/libwiegand.so}: DynLib::dlopen(): /usr/bin/../lib/leosac/libwebsock-api.so: undefined symbol: _ZN3odb10query_base6appendERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE

This means you are trying to run something that was built in a different, incompatible environment from the one you are using. 

How did you install Leosac? Are you trying to use the Debian package found on the releases page? You need to be running Debian Stretch to run that package.

crusaderpi, 2019-10-17:
I'm running buster, installed leosac with the deb. Will reformat the pi with stretch and try again.

Maxhy, 2021-05-03:
Closing for inactivity.

closed 2021-05-03 by Maxhy