leosac/access-control #127: Can not read Mifare classic card with HID iClass SE reader and RPI 3

Rokia88, 2019-12-08:
Hi,

I am trying to read the UID of a MFC card using a RPI 3 B and a Wiegand reader HID iCLASS SE . I can run Leosac without any error but can not read the UID of the card, it stops at this line of log "Module TEST_AND_RESET initialized. "
I am not using any module apart these two devices and a breadboard, level shifter, and a breadboard powersupp, I connected manually the wires of the reader to the breadboard, the reader is powered up and beeps when I swipe a card, changes the color, etc

Any help is really appreciated , I tried really hard for a long time but without success

thank you so much
[image]

AHTOLLlKA, 2020-02-24:
did you get it work?

Rokia88, 2020-03-29:
no :(

Maxhy, 2021-05-03:
@AHTOLLlKA @Rokia88 this is supported for sure. What setup did you do Leosac side?

Maxhy, 2022-02-24:
Closing for inactivity. Related to #84

closed 2022-02-24 by Maxhy