leosac/access-control #128: There is the community active?

r3c4ll, 2019-12-20:
Hi, I wish to ask some questions to the community/developers, to understand some aspects that for me aren't clear in the documentation.

There is an IRC channel (better for real time) or a forum to do that?

AHTOLLlKA, 2020-02-24:
probably not active....

knight-of-ni, 2020-02-24:
I am not one of the developers, but I have previously submitted pull requests to help modernize the project & update documentation.  I had been working with @xaqq to get pr's merged. At some point he stopped responding. I am not sure the reason, but it would be nice if he, or one of the other project owners, would respond with any plans they might have for leosac.

I've been running leosac on a pi zero for a couple years now, to control my garage door via proximity card. It works great. The application is very stable, no memory leaks, etc.

Maxhy, 2020-03-23:
Thank you for your interest into Leosac project.
Xaqq and myself put this project on hold in our life as we cannot work on it anymore, at least for now. It is not dead in a sense we still believe into this project, just not a good time. We will also support anyone who want to get involved and in short term get part of Leosac organization on GitHub... A webui created two years ago for this project that we wanted to incorporate into a major release, that never came, may be published if it helps

knight-of-ni, 2020-03-24:
@Maxhy Thanks for the response, and I get it. Things can change in one's life and suddenly you've got other priorities. I changed to a better day job in Feb 2019, which had the unfortunate side affect of leaving me with much less time to work on the project I am a part of, ZoneMinder.

I did a lot of searching before using Leosac for my own purposes. It was the only open-source access control project I could find that supported industry standard Wiegand protocol, had database support, and was generally usable out of the box. I look forward to future development whenever and by whoever that occurs.

Lastly, since you mentioned a webui, I'll mention something I've had on my to-do list. I had in mind to bring Leosac events into ZoneMinder, such that access control events are automatically linked to surveillance cameras. The end user could run a filter (a.k.a. a report) and see video footage of these events or monitor them as they occur in realtime. Right now it's just a pipe dream, but I hope to get around to it someday.

lkraav, 2020-06-05:
Hi. I'm also looking to avoid proprietary systems like ACCO Net by Satel. Mid-2020, is Leosac still the only choice? Looks like it may be too much work for a niche community project?

Maxhy, 2021-03-16:
FYI, we still cannot allocate our time on the project but we decided to share the initial project for a Leosac web frontend we started few years ago (to avoid dealing with config file and to have a proper web ui management system). Please see https://github.com/leosac/leosac-web if you are interested.

knight-of-ni, 2021-04-23:
@Maxhy Thank you posting leosac-web. It looks like communication is exchanged through Leosac's via the websocket API. 

Can you provide an XML config example, that shows how to enable the websocket API in Leosac? It appears to be missing from the documentation: https://leosac.github.io/leosac-doc/doc_output/develop/df/d46/page_module_websock_api.html

knight-of-ni, 2021-04-23:
UPDATE: I found an example here:
https://github.com/leosac/leosac/blob/develop/cfg/dev-test/ws-amd64.xml#L31

onlinegill, 2021-05-01:
> UPDATE: I found an example here:
> https://github.com/leosac/leosac/blob/develop/cfg/dev-test/ws-amd64.xml#L31

Could you please help me to setup? I have installed leosac but can't find how to use it.

Maxhy, 2021-05-03:
@onlinegill https://leosac.github.io/leosac-doc/doc_output/develop/d5/d97/page_guide_rpi_piface_wiegand.html#install_guide should be good enough as a starting point without web ui, for the webui you need to enable websocket and look at leosac-web documentation (consider it as a WIP that need dependencies updates).

Maxhy, 2022-03-25:
The project regained some activities recently and is likely to have good support again for a couple of years, starting in a few months. More information in S2 2022, please stand by 😃.

closed 2022-03-25 by Maxhy

tkwern, 2024-01-12:
Can you please provide  a quick Jan 2024 status update?