leosac/access-control #129: Raspi interface

nixator, 2020-07-16:
Hi, finally I decided to make custom PCB because previous version of my build was horrible "tangled cables" style horror and I was scared about safety because there weren't any fuses and so on... Here is result: 

[https://photos.app.goo.gl/cX6htVHS83aTDqEaA](url)

Jindra

knight-of-ni, 2020-07-16:
Very nice. So it looks like you built yourself an 8 door controller.
What are the pinouts for each rj45? Wiegand + lock relay? Anything else?

nixator, 2020-07-16:
Yes, each RJ-45 contain all necessary for one door 4 wires for wiegand reader and 2+2 for lock.  8 maybe 9 doors without SPI I/O epander is possible to do only with level shifter.  It runs for 3 years without any issue. I can provide eagle files or gerbers...

Jindra

knight-of-ni, 2020-07-16:
Good idea doubling up on the lock power pairs.

jayjlow, 2020-07-27:
> 
> 
> Yes, each RJ-45 contain all necessary for one door 4 wires for wiegand reader and 2+2 for lock. 8 maybe 9 doors without SPI I/O epander is possible to do only with level shifter. It runs for 3 years without any issue. I can provide eagle files or gerbers...
> 
> Jindra

Hi Jindra,

If you can supply the eagle and gerber files, that would be amazing!

-Joaquin

nixator, 2020-07-28:
> Hi Jindra,
>
> If you can supply the eagle and gerber files, that would be amazing!
>
> -Joaquin

http://www.jindrichnikles.cz/?p=178

Maxhy, 2020-07-28:
👍

jeremy-walker, 2020-08-04:
@nixator, This is a really impressive project. Leosac in and of itself is a very impressive effort and sorely needed in this space, and your work on top of it is really amazing. 

I have 8 doors with magnetic door locks and corresponding Wiegand RFID card readers that is currently controlled by a *horrible* proprietary system running on Windows XP that the vendor refuses to provide upgrades for or support. They would rather we throw these perfectly functioning devices out, and buy new ones so that they can provide *support* for it. 

Unfortunately I'm not terribly skilled at electronics so building one of these is beyond my skills. Would you consider building one of these out for me? I will be happy to pay for your time and materials and some more to make it worth your while. 

Do let me know.

nixator, 2020-08-04:
> @nixator, This is a really impressive project. Leosac in and of itself is a very impressive effort and sorely needed in this space, and your work on top of it is really amazing.
>
> I have 8 doors with [...]

Hi, I'm able to make another four devices, because PCB supplier do only batch of 5 pieces... So answer is yes, I can provide one unit for you.

jeremy-walker, 2020-08-04:
That's great to hear. Let me know if I should email you directly to work out the details.

nixator, 2020-08-05:
@jeremy-walker Yes, contact me here: 
jnikles (a) seznam dot cz

RIVA-Theo, 2021-04-12:
>Hi, I'm able to make another four devices, because PCB supplier do only batch of 5 pieces... So answer is yes, I can provide one >unit for you.

Hi Nixator. 

I know it's been a while since you replied to this thread but i would also be interested if you still have a spare board for your Leosac/Pi setup. I am interested in setting up my own implementation.

nixator, 2021-04-13:
Hi, I still have two PCB's, where are you from?

Jindra

Dne 12.04.2021 v 22:40 RIVA-Theo napsal(a):

RIVA-Theo, 2021-04-14:
> Hi, I still have two PCB's, where are you from? Jindra Dne 12.04.2021 v 22:40 RIVA-Theo napsal(a):
> […](#)

I am from New York City. I sent you an email to the address you listed above.

onlinegill, 2021-05-01:
> Hi, I still have two PCB's, where are you from? Jindra Dne 12.04.2021 v 22:40 RIVA-Theo napsal(a):
> […](#)

Hi, I sent you email.

onlinegill, 2021-05-01:
> @nixator, This is a really impressive project. Leosac in and of itself is a very impressive effort and sorely needed in this space, and your work on top of it is really amazing.
>
> I have 8 doors with [...]

Hi Jeremy were you able to get it? I am also looking to setup one as we have same issue like you.

jeremy-walker, 2021-05-02:
Hi @onlinegill, 

Yes, I was able to get it, and @nixator  was kind enough to not only build one out for me, but also help me set it up and sort out some installation issues specific to my setup. So if @nixator is willing to build you one, I strongly recommend you go for it. This is way better a solution than the crap they sell in the market place. Especially since it's paired with a standard linux box (raspi), it opens up a world of possibilities. 

Good luck. 

cheers -Jeremy

onlinegill, 2021-05-02:
> Hi @onlinegill,
>
> Yes, I was able to get it, and @nixator was kind enough to not only build one out for me, but also help me set it up and sort out some installation issues specific to my setup. So [...]

Thanks for reply. I sent him email couple days ago but got no response. I was wondering if you have alternative contact information. How many doors are you using?also looks like leosac hasn't updated since long time. I can make pcb but due to lack of parts I don't know if there are alternative parts can be used.

fabkoh, 2021-08-25:
hi @nixator , apologies for pulling up this old thread. I was wondering if you could share how you wired up for multiple card readers to the Pi?

nixator, 2021-08-25:
> 
> 
> hi @nixator , apologies for pulling up this old thread. I was wondering if you could share how you wired up for multiple card readers to the Pi?

Hi, here is schematic in Eagle sch file: http://www.jindrichnikles.cz/?p=178

fabkoh, 2021-09-03:
Hi @Maxhy , was wondering if you could point to the code where you do the processing of key presses on the Wiegand keypad please?

onlinegill, 2021-11-25:
> Hi @Maxhy , was wondering if you could point to the code where you do the processing of key presses on the Wiegand keypad please?
Hi any progress on it ?

Maxhy, 2022-03-05:
@nixator I would like to include your PCB schemas as an example to the default Leosac documentation. Would that be OK? I would like to highlight the license as well, any chance for you to pick a license popular on open hardware community? Thanks!

nixator, 2022-03-05:
Hi, no problem, no secret there...

Jindra
---------- Původní e-mail ----------

Komu: leosac/leosac ***@***.***>
Datum: 5. 3. 2022 19:10:06
"

or Android
"

Maxhy, 2022-04-12:
The community has a lot of issues with the hardware setup since Piface Digital got discontinued. I would also like to prepare the project for the next phase now that there is some activity again, to be more "robust" and that goes through having proper hardware support.

Based on what you did @nixator I believe the basic requirements for a "door unit board" that would match most use cases are:
| Type | Quantities | Details | Remarks |
| ---- | ---------- | ------- | ------- |
| Power | 1 | 12 VDC | 24 VDC support at the same time would be a plus |
| Reader connectors | 4 |  - D0 / Rx+ for Wiegand or RS485 input <br> - D1 / Rx- for Wiegand or RS485 input <br> - VDC+ <br> - GRND <br> - Green LED output (for Wiegand) <br> - Red LED output (for Wiegand)<br> - Buzzer output (for Wiegand) <br> - Tamper switch/alarm input | Outputs and tamper switch wire could be global to the board |
| Door relays | 4 | NO or NC | |
| Exit Request connectors | 4 | NO or NC | |
| Door Sensor inputs | 4 | | |
| DIP Switch | 2 | | For RS485 bus address and network/factory reset triggering |
| Jumpers | ? | To switch from Wiegand to RS485 for instance, except if it can be properly managed software side only | |
| LED Activity | 4 | | Will blink on transmission from reader data inputs |
| MCU connector | 1 | | This should be generic enough to work with several hardware (BeagleBone, OrangePi, Raspberry Pi etc...) We probably don't want it to be manufactured as a HAT board (or it should requires intermediate HAT board as adapter?) |

Could be bumped to 8 doors/readers or 4 doors with 8 readers (for in/out scenarios) but it may be difficult to make it fit.

Connectors should be "removable" to facilitate the wiring, that's also what the industry is used to afaik.
Now, we should also think about the following options:
 - DIN?
 - UPS power support with proper monitoring to Leosac (battery low / switched to battery). Leosac has to be considered as a critical system, will be required at some point as the power is coming from the board (cannot use already existing MCU specific UPS hats).
 - ... ?

The schematics should be under an explicit and permissive open hardware license.

Looking for feedbacks😄.

omnisci3nce, 2022-07-01:
> 

I would suggest having a physical RESET button to power cycle the RPi or equivalent Single-Board-Computer without having to unplug and plug in the power supply from/to the board.

Maxhy, 2022-07-05:
Good point @omnisci3nce. Not sure if this reset button should be accessible without removing the case (if any) as you may not want to facilitate powering off of such unit when going live.

omnisci3nce, 2022-07-05:
Yeah. If the case is easy enough to take off then I see no reason why not. Once you have the button you can allow it to do multiple things too based on how long it has been held down for. This is what various other access control manufacturers do.

OfficialKris, 2024-09-29:
Hi, just stumbled upon this project and wanted to know if this is something people are still interested in.

I have a background in electronics and would be willing to create an initial design. I also do not have access to the previous design files.

Anyone interested?

Maxhy, 2024-09-29:
Yes definitely. I'm more on the software/firmware side with knowledge of the access control industry than the electronics side. Your help would be really helpful, for this project or any Free Software access control related project as we all need this kind of hardware to have something more robust to install.

OfficialKris, 2024-09-29:
Thank you for your response,

I was thinking an initial version could be drawn from the previous design with some minor modifications. I can also say a bill of materials and instructions for assembly would be required for this project. I know most source here is AGPL but I was thinking a more permissive license like the CERN permissive hardware license would be beneficial as companies would be more willing to work with that and could hopefully bring more users to this area.

I can start two repositories, one for the hardware and one for the assembly documentation. Maybe at a later date it could be brought into the main project.

Maxhy, 2024-09-30:
Looks good. I'm fine with the CERN permissive hardware license as this hardware will probably be used on several projects (it's up to you anyway here).
What do you think about the goal (not first version obviously) requirement https://github.com/leosac/access-control/issues/129#issuecomment-1096309701?

OfficialKris, 2024-09-30:
I think the goals are good. That is a system which I would want. I was thinking a server rack assembly (with option to remove flanges) would be a good fit. I like the form factor of the previous design but disagree with the RJ45 jacks as a connector. The Phoenix MC connector series seems like a more industry standard fit. I could see using maybe two or three rows of those on the faceplate would give enough pins for a complete system in a 1U chassis.

Maxhy, 2024-10-02:
Yes Phoenix MC connector series are definitely better than RJ45 jack I agree.
Server rack is fine when controllers are installed on a server room, but when installed on a ceiling having hardware which can be plugged on DIN rail is better. One option doesn't necessary stop the other and we always have to start with something anyway :).