leosac/access-control #133: Having issues getting input

kothmann1, 2021-09-18:
So I have successfully gotten leosac to run, but now I have my reader wired up to it (I'm using a UHPPOTE wiegand RFID reader) with the data wires connected to my GPIO pins through a buck converter for step-down to 3.3V. I connected the reader's ground to the pi's ground as well as the power supply's ground, which is the only thing that made the inputs not go haywire since they were "floating" before. Now when I scan a card, leosac receives input consistently every time, but it is only a few bits of 0s, rather than any meaningful data. It usually says "timeout, buffer size = x" where x varies between 2 and 6 bits. The cards I'm using are standard 26-bit wiegand cards though.

kothmann1, 2021-09-18:
Update to this; I'm testing it with a wiegand library on my arduino and it is working, so I don't think it's anything electrical/hardware, if anyone has any thoughts they'd be greatly appreciated.

Maxhy, 2022-02-24:
@kothmann1 did you resolved your issue?

kothmann1, 2022-02-24:
No, it did not work without piface, I ended up writing a custom solution where I used a microcontroller to convert the RFID output to UART and print to the RPi and took leosac out of the loop.

Maxhy, 2022-02-24:
Mhhh ok, for the record do you remember trying https://leosac.github.io/leosac-doc/doc_output/develop/da/d7f/page_module_sysfs_gpio.html? Thanks.

Maxhy, 2022-03-25:
I have updated the documentation on https://leosac.github.io/leosac-doc/doc_output/develop/d8/d25/page_guide_rpi_wiegand.html.

I would recommend to use a Level Shifter for prototyping.

Btw a coming Leosac update will also add support for proper Wiegand format decoding (not your issue but it will facilitate the format decoding of individual fields like Uid, Facility Code, ...).

kothmann1, 2022-03-25:
I did try the sysfs module at the time, and I was using a level shifter from 5V down to 3V3, wiegand output was reaching the pi pins at 3.3V per my oscilloscope. It might be related to the fact that I was using a Rpi4 and an RPi zero W to test instead of the Rpi1 this was built for, but not sure.