leosac/access-control #135: Hrdware Support

Uzzi, 2022-04-04:
Hi, I've Raspberry Pi 2B and 4, rfid reader https://www.amazon.it/gp/product/B07DK9R7K4/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1.
I've reader PIFACE DIGITAL 2 out of stock.
How can I connect reader with Raspberry? Witch hardware I've to buy?

Thank toy so much for help

Maxhy, 2022-04-04:
Your reader has Wiegand output according to the description on Amazon. If you're using Pifare Digital 2 then you don't need additional hardware.
See https://leosac.github.io/leosac-doc/doc_output/develop/df/d1a/page_guide_raspbian.html and https://leosac.github.io/leosac-doc/doc_output/develop/d5/d97/page_guide_rpi_piface_wiegand.html

Current deb files on Release tab are outdated, you should takes it from nightly builds for now if you don't want to build Leosac yourself from source, like https://github.com/leosac/leosac/actions/runs/2082116174

You can also go with Docker image if you prefer with leosac/leosac:snapshot image (https://hub.docker.com/r/leosac/leosac/tags).
If Docker, you will have to run with privileged permissions (easiest way but may expose too much) or mount the spi device (eg. /dev/spidev0.0) to the container. If not using Piface but direct GPIOs access, it would have been /sys if using sysfsgpio  (for the record, GPIO I/O Memory through /dev/gpiomem is not supported by Leosac as there is no value for such support) and /dev/gpiochip0 if using libgpiod.

Uzzi, 2022-04-04:
Hi, I cannot use Piface Digital 2, it's out of stock! 
I need a substitute to connect my reader, relay,raspberry,door lock

Maxhy, 2022-04-12:
Ah I understood the opposite my bad.

Well there is no much similar hats available on the market which support both digital inputs up to 5v and relay outputs all in one. For testing purpose you may want to go with a voltage divider like a level shifter (eg. https://leosac.github.io/leosac-doc/doc_output/develop/d8/d25/page_guide_rpi_wiegand.html) and use a relay board (like https://www.waveshare.com/product/raspberry-pi/hats/rpi-relay-board.htm) for the relays.
Geekworm created a clone of the Piface digital with their Geekworm DIDO board (https://geekworm.com/products/raspberry-pi-3-model-b-plus-digital-input-output-expansion-board-dido-module) but I have not tested it by myself.

A few Leosac users created their own board and shared the schematics (see #129 for instance). A dedicated and more advanced Leosac board will be required at some point to facilitate adoption of the project and have more robust deployment but this is not even started for now. Hope it helps.

closed 2022-04-22 by Maxhy

Uzzi, 2022-06-13:
Hi, I've bought https://wiki.geekworm.com/Raspberry_Pi_DIDO_Board , could you help me about https://ae01.alicdn.com/kf/H1af968b7b7ec4bce83c5e4152c6be2b2z.jpg connections schema?
Thank you so much

Maxhy, 2022-06-15:
Sure. I don't have the DIDO board myself but from what I have read it should be Piface compatible.
Wiegand D0 (or Wiegand Low) and Wiegand D1 (or Wiegand High) should go to the inputs of DIDO board. Green LED and Beeper could be connected to the outputs of the DIDO board but that's optional and will require additional Leosac configuration. Then you setup Leosac like https://leosac.github.io/leosac-doc/doc_output/develop/d5/d97/page_guide_rpi_piface_wiegand.html