leosac/access-control #142: Module Issues

vanderpunk, 2022-09-13:
If i try to start leosac with any modules other than libauth-file.so the service won't start.

I'm running Leaosac version 1656078238:0.8.0-1 on a 64bit Bullseye version of Rasbian OS
pi@raspberrypi:~ $ cat /etc/os-release                             
   PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
    NAME="Debian GNU/Linux"
    VERSION_ID="11"
    VERSION="11 (bullseye)"
    VERSION_CODENAME=bullseye
    ID=debian
    HOME_URL="https://www.debian.org/"
    SUPPORT_URL="https://www.debian.org/support"
    BUG_REPORT_URL="https://bugs.debian.org/"
pi@raspberrypi:~ $ uname -a                                                                                                                                                                                       
    Linux raspberrypi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux    

Below is the output of "systemctl status leosac"
● leosac.service - Leosac daemon
     Loaded: loaded (/lib/systemd/system/leosac.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2022-09-13 11:16:54 EDT; 4s ago
       Docs: http://www.leosac.com/
    Process: 1700 ExecStart=/usr/bin/leosac -k /etc/leosac.d/kernel.xml (code=exited, status=1/FAILURE)
   Main PID: 1700 (code=exited, status=1/FAILURE)
        CPU: 372ms

Sep 13 11:16:53 raspberrypi leosac[1700]: [2022-09-13 11:16:53.990] [console] [info] [1700] Attempting to load module nammed PIFACEDIGITAL_GPIO (shared lib file = libpifacedigital.so)                          Sep 13 11:16:53 raspberrypi leosac[1700]: [2022-09-13 11:16:53.993] [console] [info] [1700] Loading library at: /usr/lib/leosac/libpifacedigital.so                                                              Sep 13 11:16:54 raspberrypi leosac[1700]: [2022-09-13 11:16:54.000] [console] [error] [1700] FAILURE, full path was:{/usr/lib/leosac/libpifacedigital.so}: DynLib::dlopen(): libwebsock-api.so: cannot open shar>Sep 13 11:16:54 raspberrypi leosac[1700]: [1700] FAILURE, full path was:{/usr/lib/leosac/libpifacedigital.so}: DynLib::dlopen(): libwebsock-api.so: cannot open shared object file: No such file or directory    Sep 13 11:16:54 raspberrypi leosac[1700]: Exception propagated to main(). Will now exit.
Sep 13 11:16:54 raspberrypi leosac[1700]: exception: Cannot load modules. Search path was:
Sep 13 11:16:54 raspberrypi leosac[1700]:          -> /usr/lib/leosac
Sep 13 11:16:54 raspberrypi leosac[1700]:          -> /usr/lib/leosac/auth
Sep 13 11:16:54 raspberrypi systemd[1]: leosac.service: Main process exited, code=exited, status=1/FAILURE                                                                                                       Sep 13 11:16:54 raspberrypi systemd[1]: leosac.service: Failed with result 'exit-code'.

to see if the .so file were in the correct place
pi@raspberrypi:~ $ ls /usr/lib/leosac/
auth                libdoorman.so          libled-buzzer.so         libmonitor.so        libreplication.so  libstdin-controller.so  libtest-and-reset.so  libws-notifier.so
libalarms.so        libevent-publish.so    liblibgpiod.so           libnotifd.so         librpleth.so       libsysfsgpio.so         libwebsock-api.so
libbench-toggle.so  libinstrumentation.so  libmodule_piface_deps.a  libpifacedigital.so  libsmtp.so         libtcp-notifier.so      libwiegand.so

Maxhy, 2022-09-14:
Lets start with stupid questions first.

Are you sure the user `pi` has proper permissions on *.so files located at /usr/lib/leosac ?
Can you run `ldd` on libpifacedigital.so then on libwebsock-api.so?

vanderpunk, 2022-09-14:
Permissions on the *.so files
pi@raspberrypi:~ $ ls -al /usr/lib/leosac/
total 7900
drwxr-xr-x  3 root root    4096 Sep 13 11:05 .
drwxr-xr-x 85 root root    4096 Sep 13 11:05 ..
drwxr-xr-x  2 root root    4096 Sep 13 11:05 auth
-rw-r--r--  1 root root  476920 Jun 24 09:44 libalarms.so
-rw-r--r--  1 root root  159392 Jun 24 09:44 libbench-toggle.so
-rw-r--r--  1 root root  258496 Jun 24 09:44 libdoorman.so
-rw-r--r--  1 root root  188296 Jun 24 09:44 libevent-publish.so
-rw-r--r--  1 root root  175824 Jun 24 09:44 libinstrumentation.so
-rw-r--r--  1 root root  485176 Jun 24 09:44 libled-buzzer.so
-rw-r--r--  1 root root  217248 Jun 24 09:44 liblibgpiod.so
-rw-r--r--  1 root root   16114 Jun 24 09:44 libmodule_piface_deps.a
-rw-r--r--  1 root root  237960 Jun 24 09:44 libmonitor.so
-rw-r--r--  1 root root  188096 Jun 24 09:44 libnotifd.so
-rw-r--r--  1 root root  605552 Jun 24 09:44 libpifacedigital.so
-rw-r--r--  1 root root  192336 Jun 24 09:44 libreplication.so
-rw-r--r--  1 root root  324424 Jun 24 09:44 librpleth.so
-rw-r--r--  1 root root  757656 Jun 24 09:44 libsmtp.so
-rw-r--r--  1 root root  171808 Jun 24 09:44 libstdin-controller.so
-rw-r--r--  1 root root  233808 Jun 24 09:44 libsysfsgpio.so
-rw-r--r--  1 root root  204768 Jun 24 09:44 libtcp-notifier.so
-rw-r--r--  1 root root  188328 Jun 24 09:44 libtest-and-reset.so
-rw-r--r--  1 root root 1908392 Jun 24 09:44 libwebsock-api.so
-rw-r--r--  1 root root  867856 Jun 24 09:44 libwiegand.so
-rw-r--r--  1 root root  192496 Jun 24 09:44 libws-notifier.so

pi@raspberrypi:~ $ ldd /usr/lib/leosac/libpifacedigital.so
        linux-vdso.so.1 (0x0000007f9e0a0000)
        libwebsock-api.so => not found
        libleosac_lib.so => /lib/libleosac_lib.so (0x0000007f9dc7d000)
        libleosac_db.so => /lib/libleosac_db.so (0x0000007f9d814000)
        libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f9d7e3000)
        libodb-2.4.so => /lib/aarch64-linux-gnu/libodb-2.4.so (0x0000007f9d7b2000)
        libodb-pgsql-2.4.so => /lib/aarch64-linux-gnu/libodb-pgsql-2.4.so (0x0000007f9d776000)
        libodb-sqlite-2.4.so => /lib/aarch64-linux-gnu/libodb-sqlite-2.4.so (0x0000007f9d73d000)
        libzmqpp.so => /lib/libzmqpp.so (0x0000007f9d708000)
        libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f9d530000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f9d50c000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f9d397000)
        /lib/ld-linux-aarch64.so.1 (0x0000007f9e070000)
        libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f9d383000)
        libscrypt.so.0 => /lib/libscrypt.so.0 (0x0000007f9d36d000)
        libzmq.so.5 => /lib/aarch64-linux-gnu/libzmq.so.5 (0x0000007f9d2a6000)
        libboost_serialization.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_serialization.so.1.74.0 (0x0000007f9d251000)
        libboost_regex.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_regex.so.1.74.0 (0x0000007f9d154000)
        libodb-boost-2.4.so => /lib/aarch64-linux-gnu/libodb-boost-2.4.so (0x0000007f9d141000)
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f9d096000)
        libpq.so.5 => /lib/aarch64-linux-gnu/libpq.so.5 (0x0000007f9d036000)
        libsqlite3.so.0 => /lib/aarch64-linux-gnu/libsqlite3.so.0 (0x0000007f9ceea000)
        libbsd.so.0 => /lib/aarch64-linux-gnu/libbsd.so.0 (0x0000007f9cec5000)
        libsodium.so.23 => /lib/aarch64-linux-gnu/libsodium.so.23 (0x0000007f9ce7b000)
        libpgm-5.3.so.0 => /lib/aarch64-linux-gnu/libpgm-5.3.so.0 (0x0000007f9ce19000)
        libnorm.so.1 => /lib/aarch64-linux-gnu/libnorm.so.1 (0x0000007f9cce4000)
        libgssapi_krb5.so.2 => /lib/aarch64-linux-gnu/libgssapi_krb5.so.2 (0x0000007f9cc86000)
        libicui18n.so.67 => /lib/aarch64-linux-gnu/libicui18n.so.67 (0x0000007f9c97c000)
        libicuuc.so.67 => /lib/aarch64-linux-gnu/libicuuc.so.67 (0x0000007f9c784000)
        libssl.so.1.1 => /lib/aarch64-linux-gnu/libssl.so.1.1 (0x0000007f9c6e7000)
        libcrypto.so.1.1 => /lib/aarch64-linux-gnu/libcrypto.so.1.1 (0x0000007f9c436000)
        libldap_r-2.4.so.2 => /lib/aarch64-linux-gnu/libldap_r-2.4.so.2 (0x0000007f9c3d1000)
        libmd.so.0 => /lib/aarch64-linux-gnu/libmd.so.0 (0x0000007f9c3b5000)
        libkrb5.so.3 => /lib/aarch64-linux-gnu/libkrb5.so.3 (0x0000007f9c2c9000)
        libk5crypto.so.3 => /lib/aarch64-linux-gnu/libk5crypto.so.3 (0x0000007f9c28b000)
        libcom_err.so.2 => /lib/aarch64-linux-gnu/libcom_err.so.2 (0x0000007f9c277000)
        libkrb5support.so.0 => /lib/aarch64-linux-gnu/libkrb5support.so.0 (0x0000007f9c25a000)
        libicudata.so.67 => /lib/aarch64-linux-gnu/libicudata.so.67 (0x0000007f9a733000)
        liblber-2.4.so.2 => /lib/aarch64-linux-gnu/liblber-2.4.so.2 (0x0000007f9a712000)
        libresolv.so.2 => /lib/aarch64-linux-gnu/libresolv.so.2 (0x0000007f9a6eb000)
        libsasl2.so.2 => /lib/aarch64-linux-gnu/libsasl2.so.2 (0x0000007f9a6bf000)
[... 37 lines cut ...]
        libgssapi_krb5.so.2 => /lib/aarch64-linux-gnu/libgssapi_krb5.so.2 (0x0000007fa201f000)
        libicui18n.so.67 => /lib/aarch64-linux-gnu/libicui18n.so.67 (0x0000007fa1d15000)
        libicuuc.so.67 => /lib/aarch64-linux-gnu/libicuuc.so.67 (0x0000007fa1b1d000)
        libssl.so.1.1 => /lib/aarch64-linux-gnu/libssl.so.1.1 (0x0000007fa1a80000)
        libcrypto.so.1.1 => /lib/aarch64-linux-gnu/libcrypto.so.1.1 (0x0000007fa17cf000)
[... cut ...]
        libunistring.so.2 => /lib/aarch64-linux-gnu/libunistring.so.2 (0x0000007f9f518000)
        libtasn1.so.6 => /lib/aarch64-linux-gnu/libtasn1.so.6 (0x0000007f9f4f4000)
        libnettle.so.8 => /lib/aarch64-linux-gnu/libnettle.so.8 (0x0000007f9f4a1000)
        libhogweed.so.6 => /lib/aarch64-linux-gnu/libhogweed.so.6 (0x0000007f9f449000)
        libgmp.so.10 => /lib/aarch64-linux-gnu/libgmp.so.10 (0x0000007f9f3c1000)
        libffi.so.7 => /lib/aarch64-linux-gnu/libffi.so.7 (0x0000007f9f3a6000)

Maxhy, 2022-09-14:
Are you using a .deb package or did you built it by yourself?
Can you temporary add /usr/lib/leosac to `LD_LIBRARY_PATH` then run `sudo ldconfig` and try again to run leosac?

vanderpunk, 2022-09-14:
I'm using the .deb package from https://github.com/leosac/leosac/actions Release #53

I believe this temporarily adds /usr/lib/leosac to LD_LIBRARY_PATH
    pi@raspberrypi:~ $ export LD_LIBRARY_PATH=/usr/lib/leosac 

Then ran "sudo ldconfig"
pi@raspberrypi:~ $ sudo ldconfig                                                                                                                                                                           

Then tried to run leosac
pi@raspberrypi:~ $ sudo systemctl start leosac                                                                                                                                                             pi@raspberrypi:~ $ systemctl status leosac.service                                                                                                                                                         ● leosac.service - Leosac daemon                                                                                                                                                                                Loaded: loaded (/lib/systemd/system/leosac.service; disabled; vendor preset: enabled)                                                                                                                      Active: failed (Result: exit-code) since Wed 2022-09-14 11:29:55 EDT; 7s ago                                                                                                                                 Docs: http://www.leosac.com/                                                                                                                                                                            Process: 4838 ExecStart=/usr/bin/leosac -k /etc/leosac.d/kernel.xml (code=exited, status=1/FAILURE)                                                                                                       Main PID: 4838 (code=exited, status=1/FAILURE)                                                                                                                                                                  CPU: 183ms                                                                                                                                                                                                                                                                                                                                                                                                   
Sep 14 11:29:55 raspberrypi leosac[4838]: [2022-09-14 11:29:55.947] [console] [info] [4838] Attempting to load module nammed PIFACEDIGITAL_GPIO (shared lib file = libpifacedigital.so)
Sep 14 11:29:55 raspberrypi leosac[4838]: [2022-09-14 11:29:55.947] [console] [info] [4838] Loading library at: /usr/lib/leosac/libpifacedigital.so
Sep 14 11:29:55 raspberrypi leosac[4838]: [2022-09-14 11:29:55.950] [console] [error] [4838] FAILURE, full path was:{/usr/lib/leosac/libpifacedigital.so}: DynLib::dlopen(): libwebsock-api.so: cannot ope>Sep 14 11:29:55 raspberrypi leosac[4838]: [4838] FAILURE, full path was:{/usr/lib/leosac/libpifacedigital.so}: DynLib::dlopen(): libwebsock-api.so: cannot open shared object file: No such file or direct>Sep 14 11:29:55 raspberrypi leosac[4838]: Exception propagated to main(). Will now exit.
Sep 14 11:29:55 raspberrypi leosac[4838]: exception: Cannot load modules. Search path was:
Sep 14 11:29:55 raspberrypi leosac[4838]:          -> /usr/lib/leosac
Sep 14 11:29:55 raspberrypi leosac[4838]:          -> /usr/lib/leosac/auth
Sep 14 11:29:55 raspberrypi systemd[1]: leosac.service: Main process exited, code=exited, status=1/FAILURE
Sep 14 11:29:55 raspberrypi systemd[1]: leosac.service: Failed with result 'exit-code'.

knight-of-ni, 2022-09-14:
hmmm.... I would have expected that to work. 

Maybe try modifying your library path this way:
https://stackoverflow.com/a/13428971

vanderpunk, 2022-09-14:
that seems to have worked

[image]

knight-of-ni, 2022-09-14:
Undefined symbol errors can be caused by using a deb package that was built against a different kernel or distro than the one you are using. Can you verify you are using the Leosac package for Debian Bullseye, rather than the one for Buster?
You are using the arm64 package, right?

I wonder if building the package in Raspbian, rather than Debian, would resolve this issue.

vanderpunk, 2022-09-14:
Not sure how to verify the Leosac package other than showing what i downloaded and installed
[image]

I'll try building it on the pi

knight-of-ni, 2022-09-14:
When you went here:
https://github.com/leosac/leosac/actions/runs/3026551796

which deb package did you select? There are choices for Bullseye and Buster. If you downloaded the package for Buster, you could get the symptoms you have described.

Maxhy, 2022-09-15:
The latest error you have with Undefined Symbol @vanderpunk is related to a missing getter implementation on the codebase. I guess that means the current HEAD is unstable :\
I've fixed this specific error with commit 977556f99bff8c89b801f51951168d5f2076f41b and triggered a snapshot build on https://github.com/leosac/leosac/actions/runs/3060069639 for .deb generation and docker image update. But as I said, consider it unstable for now. I really need to go over a proper release (and related tests). I'm currently focused on another aspect of the access control solution which is credential provisioning (eg. RFID card encoding on a secure way with custom keys). Once that advanced enough to be used as a PoC, I will focus on Leosac proper release.

vanderpunk, 2022-09-15:
Thanks @Maxhy 
Understood it's unstable but it did successfully start with the Doorman Module 

[image]

You all are doing an awesome job here. Looking forward to watching this project progress.

Maxhy, 2025-08-30:
I believe this can be closed :).

closed 2025-08-30 by Maxhy