leosac/access-control #143: Raspberry pi 4 build failed

hinkma, 2023-01-08:
I try build leosec for RPI4 64bit and it fails.

[  9%] Linking CXX shared library ../../../librpleth.so
cd /tmp/tmp.CMRNrYG0Is/leosac_0.8.0/obj-aarch64-linux-gnu/src/modules/rpleth && /usr/bin/cmake -E cmake_link_script CMakeFiles/rpleth.dir/link.txt --verbose=1
/usr/bin/c++ -fPIC -g -O2 -ffile-prefix-map=/tmp/tmp.CMRNrYG0Is/leosac_0.8.0=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -std=c++14 -Wno-undef -Wno-shadow -Wno-unknown-pragmas -O3 -DNDEBUG -Wl,-z,relro -shared -Wl,-soname,librpleth.so -o ../../../librpleth.so CMakeFiles/rpleth.dir/init.cpp.o CMakeFiles/rpleth.dir/RplethModule.cpp.o CMakeFiles/rpleth.dir/rplethpacket.cpp.o CMakeFiles/rpleth.dir/rplethprotocol.cpp.o CMakeFiles/rpleth.dir/network/circularbuffer.cpp.o  -Wl,-rpath,::::::::::::::::::::::::::::
make[3]: Leaving directory '/tmp/tmp.CMRNrYG0Is/leosac_0.8.0/obj-aarch64-linux-gnu'
[  9%] Built target rpleth
make[2]: Leaving directory '/tmp/tmp.CMRNrYG0Is/leosac_0.8.0/obj-aarch64-linux-gnu'
make[1]: *** [Makefile:174: all] Error 2
make[1]: Leaving directory '/tmp/tmp.CMRNrYG0Is/leosac_0.8.0/obj-aarch64-linux-gnu'
dh_auto_build: error: cd obj-aarch64-linux-gnu && make -j2 VERBOSE=1 returned exit code 2
make: *** [debian/rules:4: build] Error 25
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui -b failed

hinkma, 2023-01-10:
I try use before instruction Build and install Debian package, that not work.
I use instruction Building [Leosac] and now building pass.
Next problem is starting.

 pi@raspberrypi:~ $ sudo systemctl status leosac
● leosac.service - Leosac daemon
     Loaded: loaded (/etc/systemd/system/leosac.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2023-01-10 18:34:41 EET; 2min 47s ago
       Docs: http://www.leosac.com/
    Process: 6628 ExecStart=/usr/local/bin/leosac -k /etc/leosac.d/kernel.xml (code=exited, status=127)
   Main PID: 6628 (code=exited, status=127)
        CPU: 3ms

Jan 10 18:34:41 raspberrypi systemd[1]: Started Leosac daemon.
Jan 10 18:34:41 raspberrypi leosac[6628]: /usr/local/bin/leosac: error while loading shared libraries: libleosac_lib.so: cannot open shared object file: No such file or directory
Jan 10 18:34:41 raspberrypi systemd[1]: leosac.service: Main process exited, code=exited, status=127/n/a
Jan 10 18:34:41 raspberrypi systemd[1]: leosac.service: Failed with result 'exit-code'.

What I can do?

Maxhy, 2023-01-19:
@hinkma did you run `make install` after `make`?
If so, be sure module folder is on library path, with appropriate permissions (see recent #142).

closed 2025-08-05 by Maxhy